lumapu/ahoy #1425: HERF Microinverter Subbrand

Eisenhauer1987, 2024-02-12:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

It Is possible to add Subbrands for Hoymilies. I have a HERF inverter which uses almost the same communication protocol. The SN varies am it is not possible to enter the Serial in the config. A01002AP4-1T9 is the serial number.

lumapu, 2024-02-16:
Ich antworte mal in Deutsch:

Verstehe das Problem, aber wie soll die Seriennummer interpretiert werden? Welche SN wird im Protokoll dann verwendet? Aktuell sind die Seriennummern zB. `116122334455` wobei für die Kommunikation nur `22334455` vewendet wird, der Teil `1161` dient zur Erkennung welcher Wechselrichtertyp (z.B. Anzahl der Eingänge) verwendet wird.

Eisenhauer1987, 2024-02-17:
Hi, ich bin bei dem dem Schwesterprojekt OpenDTU schon weiter:
https://github.com/tbnobody/OpenDTU/issues/1751

Ich denke das ließe sich auch hierher übertragen

lumapu, 2024-02-17:
ok, danke für den Hinweis.

stefan123t, 2024-10-29:
@Eisenhauer1987 kannst Du das bitte auch mal mit der AhoyDTU Software testen. Wir haben leider keinen HERF-800 oder anderen Inverter mit den alphanumerischen Seriennummern. Aber wenn die selbe Seriennummer bei der Umrechnung herauskommt, sollte ja eigentlich auch alles schick sein ?

Dann können wir ein weiteres Issue zu machen :grin:

carsten-wolf, 2025-03-06:
Ich habe einen HERF 800 Inverter. Die S/N wird von der AhoyDTU nicht angenommen, diese bleibt rot hinterlegt. Wenn ich die umgewandelte aus der OpenDTU nehme wird die angenommen, nur habe ich bei dem HERF 800 dann nur einen Eingang zur Verfügung. Installier ist die Version 0.8.154