lumapu/ahoy #1430: ESP8266 Nach update kein Lebenszeichen mehr

Flibust, 2024-02-16:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

Ich habe die 2 o.g. Module, vor ca 1 Jahr installiert und lief es relativ ok.
Nun wollte ich gestern das Ganze mal schnell updaten, und habe die Datei  "240213_ahoy_0.8.81_bd53280_esp32-wroom32" runtergeladen und sie über die Web Oberfläche geflasht. Seitdem hatte ich fast wie kein Lebenszeichen von meinem DTU. Das Einzige, was passiert - wenn ich auf RESET drücke, leuchtet die LED kurz, sonst gibts nichts mehr - keine Verbindung über Kabel zum PC, kein WLAN von DTU zu sehen. Ich habe versucht zu reseten - die RESET Taste lange gedruckt, ein/ausstecken, keine Reaktion.
Habe ich die falsche Datei geflasht und ist mein DTU somit unbrauchbar geworden? Wenn ja, soll ich die beiden Module wieder bestellen oder reicht nur ein neuer ESP8266?

rejoe2, 2024-02-16:
Die 0.8.81 hat einen Fehler, der bei aktiviertem MQTT ständige reboots erzeugt. Bitte updaten, das Problem ist afaik mit der 0.8.82 behoben.

Flibust, 2024-02-16:
Hi, danke für die schnelle Antwort. Das Problem besteht darin, dass ich nun absolut keinen Zugriff auf mein DTU habe. Weder per Updater, noch zB über Nademcu. Es gibt nicht mal ein akustisches Feedback vom PC, wenn ich es mit dem PC verbinde.

rejoe2, 2024-02-16:
Na ja, wenn gar nichts passiert, ist ggf. auch der USB-seriell-Wandler hinüber, der da drin ist. Glaube ich aber nicht, wenn du ein OTA-update eingespielt hast, dass das passiert ist.

Hast du eine Linux-Büchse oder ein Windoof?
Unter Linux sollte lsusb was liefern, sonst ist der Wandler nicht ansprechbar. (Nachdenk, lange her mit ESP8266...) Manchmal hilft es, anszustöpseln bei gedrückter reset-Taste und dann nach der flash-Anweisung erst loszulassen.

jimknopf63, 2024-02-16:
> Ich habe Win11, nichts exotisches. Beim Anstöpseln passiert leider nichts, vom PC wird nichts erkannt, somit auch kein Flashen möglich.

Probiere mal bitte ein anderes Kabel, evtl. hast Du ein reines Ladekabel anstatt ein Datenkabel erwischt. Und irgendwann beim Releasewechsel musste man beim 8266 die PIN (glaube D3/D4) softwareseitig ändern. Weiß ja nicht wie alt deine Version ist.

rejoe2, 2024-02-16:
> Ich habe Win11, nichts exotisches.

Das ist Ansichtssache, was "exotisch" ist... 

Wenn das mit dem Kabel nicht die Ursache ist:
Versuche es mal mit einem Linux (ggf. live-System von USB booten) oder suche dir jemanden, der damit Erfahrung hat. Wind.* ist nach meiner Erfahrung nicht für's debuggen von derartigen Problemen gemacht (oder ich bin zu lange raus, kann auch sein). Wenn es überhaupt geht, musst du jedenfalls eher auf der Konsole das originale Espressiv-flash-Tool verwenden als irgendwas an einer "Meta-Oberfläche" rumzuklicken. Just my2ct.

Flibust, 2024-02-16:
3 verschiede Kabel haben nichts neues bewirkt.
Mit Linux würde ich jetzt nicht anfangen können, der Zeitaufwand wäre wohl viel zu groß. 
Vorm Posting habe ich all die farbigen Kabel neu verbunden und das Ganze zur Sicherheit gecheckt. 
Das genaue Checken hätte ich vorm falschen Flashen anwenden sollen :/
Ich muss wohl alles neu bestellen.

Flibust, 2024-02-16:
Kann ich nochmal die gleichen Module bestellen oder wäre so ein ESP32 entscheidend besser?
Ich habe zwei Stück 600er Wechselrichter und damit bleibt es auch.

rejoe2, 2024-02-16:
> Kann ich nochmal die gleichen Module bestellen oder wäre so ein ESP32 entscheidend besser? Ich habe zwei Stück 600er Wechselrichter und damit bleibt es auch.

ESP32 ist einfach "fluffiger", es lohnt nicht, beim ESP8266 zu bleiben.

Es gibt (via discord zu finden) auch zwei shops, die "community-boards" anbieten, dann muss man nicht selber "fummeln" und erhält dann sogar einen ESP32-S3 (und die Option, ggf. später dann mal "mehr" software auf dem ESP zu haben für Zählerablesung, Nulleinspeisung, Kommunikation mit Akku (größtenteils Zukunftsmusik, aber absehbar)...)...

MetaChuh, 2024-02-16:
@rejoe2 

> Es gibt (via discord zu finden) auch zwei shops, die "community-boards" anbieten ...

bitte hier direkt verlinken, in discord findet es keiner, auch ich nicht 😉

rejoe2, 2024-02-16:
https://3printd-solution.eu/
https://shop.allianceapps.io/products/allianceapps-opendtu-fusion (Ankündigung siehe hier: https://discord.com/channels/984173303147155506/1003657762061942854/1122575372396019874, irgendwo gab es da auch einen coupon, aber das finde ich grade nicht).

Flibust, 2024-02-16:
Ok, ich habe noch was rausgefunden. Und zwar habe ich vor einem Jahr folgendes bestellt: 
WeMos D1 Mini Pro V 3,0 NodeMcu und NRF24L01+
Spielt das jetzt eine Rolle oder ist ein WeMos D1 Mini Pro gleich ESP8266? Auf dem Chip steht schon ESP8266MOD

rejoe2, 2024-02-16:
> Ok, ich habe noch was rausgefunden. Und zwar habe ich vor einem Jahr folgendes bestellt: WeMos D1 Mini Pro V 3,0 NodeMcu und NRF24L01+ Spielt das jetzt eine Rolle oder ist ein WeMos D1 Mini Pro gleich ESP8266? Auf dem Chip steht schon ESP8266MOD

...wenn es auf dem Chip steht...

Es gibt afaik schon ein D1-mini-"kompatibles" ESP32-Dev-Board, aber wie das im Detail aussieht? Keine Ahnung... Was du hast, ist jedenfalls kein ESP32.

mroenne2022, 2024-02-16:
Wenn ich das alles richtig gelesen hab hast du eine esp32 wroom Datei genutzt, hast aber einen esp8266! Ich denke das war der erste und grundsätzliche Fehler. Ich denke kpl neu flashen manuell sollte machbar sein.

You69Man, 2024-02-16:
> Wenn ich das alles richtig gelesen hab hast du eine esp32 wroom Datei genutzt, hast aber einen esp8266! Ich denke das war der erste und grundsätzliche Fehler. Ich denke kpl neu flashen manuell sollte machbar sein.

Ja, das war sicherlich das ursächliche Problem, eine ESP32 Datei auf einen ESP8266 zu laden. Das kann nicht funktionieren. 
Es sollte aber auch kein Grund für einen Totalschaden des ESP8266 sein. Mit dem [USB ESP Flashtool](https://ahoydtu.de/web_install/) müsste man ihn normalerweise wieder auf die Beine bekommen.

Hast du nachgesehen, ob sich Ahoy mit einem neuerlichen (korrekten) Updatefile vielleicht wieder im AP Mode meldet?

Ich verwende übrigens auch gerne die Wemos D1 mini boards, sowohl die ESP8266 als auch die ESP32 kompatiblen (https://www.az-delivery.de/products/esp32-d1-mini). Funktionieren bei mir alle sehr gut.

Flibust, 2024-02-16:
> Ja, das war sicherlich das ursächliche Problem, eine ESP32 Datei auf einen ESP8266 zu laden. Das kann nicht funktionieren. Es sollte aber auch kein Grund für einen Totalschaden des ESP8266 sein. Mit [...]

Wenn ich auf die "[USB ESP Flashtool](https://ahoydtu.de/web_install/)" Seite gehe und auf Connect klicke, kriege ich nur die Meldung, dass keine kompatiblen Geräte gefunden wurden, die Driver habe ich alle installiert, neugestartet, nix.
Die blaue LED an dem ESP82 hat sich früher entweder permanent oder blinkend gemeldet. Seit dem og Update meldet sie sich kurz nur wenn ich auf RESET drucke, sonst nicht.

Flibust, 2024-02-16:
> > 
> 
> Hast du nachgesehen, ob sich Ahoy mit einem neuerlichen (korrekten) Updatefile vielleicht wieder im AP Mode meldet?
> 
Das habe ich nicht verstanden, was und wo nachgeschaut werden könnte

Flibust, 2024-02-16:
Ach, und den Kondensator habe ich auch nie gehabt, ist das vielleicht mit dem Update zwingend erforderlich geworden und funktioniert deswegen nicht?

rmayergfx, 2024-02-16:
Mach mal folgendes. Installiere dir den aktuellen Google Chrome und öffne damit diese Seite:
https://esphome.github.io/esp-web-tools/
Am WEMOS machst du dir ein Bild von der Verkabelung und ziehst alle Kabel ab!
Dann steckst du mit einem wirklich voll belegten Kabel das Teil an den PC und versuchts mit Connect den mal mit ESPHome zu flashen. Keine Panik, damit konnte ich schon einige DTUs wieder retten. Dort kannst du auch gleich in der Config dein WLAN zuerst mal hinterlegen und über ESP Home dann die neue AhoyDTU bin passend für den Wemos installieren.

You69Man, 2024-02-16:
> > Am WEMOS machst du dir ein Bild von der Verkabelung und ziehst alle Kabel ab! Dann steckst du mit einem wirklich voll belegten Kabel das Teil an den PC und versuchst mit Connect den mal mit ESPHome zu flashen. 
> 
> Damit ich das richtig verstehe - du meinst ich soll die Kabel abziehen und alles neu verbinden damit ich sicher bin, dass sie richtig verbunden sind oder meinst du etwas anderes?
> 
> 
> 

Ich glaube er meint, zunächst die Drähte von Pins abzuziehen, damit von hier nichts stört, und dann ein vollbelegtes USB Kabel, also ein echtes Datenkabel für den Download verwenden.

Flibust, 2024-02-16:
Ok, es lag doch am Kabel! Ich hätte nie gedacht, dass von all meinen alten Micro-USB Kabeln nur eins vollbelegt ist.
Danke euch! Ihr war sehr geduldig und seid bis zum Schluss dran geblieben, tolle Leistung, vielen Dank! :)

rmayergfx, 2024-02-16:
Kleiner Tipp dazu, das Kabel sofort Kennzeichnen und an einem sicheren Ort aufbewahren und dies auf keinen Fall für das tägliche Aufladen benutzen!

MetaChuh, 2024-02-16:
> das Kabel sofort Kennzeichnen und an einem sicheren Ort aufbewahren

ooooh wie wahr 👍

hab' auch brav letztes jahr alle kabel mit edding lackstift beschriftet 😉
p (power) und p+d (power+data)

gab nichts schlimmeres als im stress ein passendes datenkabel zu finden, wir sind fast alle davon gebrandmarkt 🙈😄

[image]

MetaChuh, 2024-02-16:
ps @Flibust 
bitte dieses issue schließen, wenn deine erstfrage erfüllt ist.

Flibust, 2024-02-16:
Fall gelöst

closed 2024-02-16 by lumapu