lumapu/ahoy #1432: Homeassistant Discovery funktioniert nicht

mathiaslehmann, 2024-02-17:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Mqtt ist aktiviert mit einen Wechselrichter HM-1200. Dieser hat 3 angeschlossene PV Module

### Error description

Wenn ich in der DTU ein Homeassistant Discovery betätige, so erhält man in Homeassistant folgenden Fehler. Das neue MqTT Gerät wird nicht hinzugefügt. Hinweis, ich habe meine Wechselrichter Seriennummer gegen "123456789012" in unten befindlicher Meldungen ausgetauscht. 

Logger: homeassistant.components.mqtt.client
Source: components/mqtt/client.py:846
Integration: MQTT (documentation, issues)
First occurred: 14:31:59 (166 occurrences)
Last logged: 14:43:16

Can't decode payload b'{"name":"YieldDay\\u0000_ID\\u0000on\\u0000\xf4\xaf\\u0000\\u0000\x9b:\x0e\x80\x10!\xfb?\x01\\u0000\\u0000\\u0000","stat_t":"inverter/WR1/ch0/YieldDay","unit_of_meas":"Wh","uniq_id":"123456789012_ch0_YieldDay","dev":{"name":"WR1","ids":"123456789012","mdl":"WR1","cu":"http://192.168.10.12","mf":"Hoymiles"},"dev_cla":"energy","stat_cla":"total_increasing"}' on homeassistant/sensor/WR1/ch0_YieldDay/config with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<function _callback_wrapper at 0x7f9229045e40>, <function async_start.<locals>.async_discovery_message_received at 0x7f91d80c2fc0>, <function async_subscribe.<locals>.<lambda> at 0x7f91dc06e2a0>)>)
Can't decode payload b'{"name":"YieldTotal\\u0000D\\u0000on\\u0000\xf4\xaf\\u0000\\u0000\x9b:\x0e\x80\x10!\xfb?\x01\\u0000\\u0000\\u0000","stat_t":"inverter/WR1/ch0/YieldTotal","unit_of_meas":"kWh","uniq_id":"123456789012_ch0_YieldTotal","dev":{"name":"WR1","ids":"123456789012","mdl":"WR1","cu":"http://192.168.10.12","mf":"Hoymiles"},"dev_cla":"energy","stat_cla":"total_increasing"}' on homeassistant/sensor/WR1/ch0_YieldTotal/config with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<function _callback_wrapper at 0x7f9229045e40>, <function async_start.<locals>.async_discovery_message_received at 0x7f91d80c2fc0>, <function async_subscribe.<locals>.<lambda> at 0x7f91dc06e2a0>)>)
Can't decode payload b'{"name":"P_DC\\u0000Total\\u0000D\\u0000on\\u0000\xf4\xaf\\u0000\\u0000\x9b:\x0e\x80\x10!\xfb?\x01\\u0000\\u0000\\u0000","stat_t":"inverter/WR1/ch0/P_DC","unit_of_meas":"W","uniq_id":"123456789012_ch0_P_DC","dev":{"name":"WR1","ids":"123456789012","mdl":"WR1","cu":"http://192.168.10.12","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"power","stat_cla":"measurement"}' on homeassistant/sensor/WR1/ch0_P_DC/config with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<function _callback_wrapper at 0x7f9229045e40>, <function async_start.<locals>.async_discovery_message_received at 0x7f91d80c2fc0>, <function async_subscribe.<locals>.<lambda> at 0x7f91dc06e2a0>)>)
Can't decode payload b'{"name":"Efficiency\\u0000D\\u0000on\\u0000\xf4\xaf\\u0000\\u0000\x9b:\x0e\x80\x10!\xfb?\x01\\u0000\\u0000\\u0000","stat_t":"inverter/WR1/ch0/Efficiency","unit_of_meas":"%","uniq_id":"123456789012_ch0_Efficiency","dev":{"name":"WR1","ids":"123456789012","mdl":"WR1","cu":"http://192.168.10.12","mf":"Hoymiles"},"exp_aft":95}' on homeassistant/sensor/WR1/ch0_Efficiency/config with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<function _callback_wrapper at 0x7f9229045e40>, <function async_start.<locals>.async_discovery_message_received at 0x7f91d80c2fc0>, <function async_subscribe.<locals>.<lambda> at 0x7f91dc06e2a0>)>)
Can't decode payload b'{"name":"MaxPower\\u0000y\\u0000D\\u0000on\\u0000\xf4\xaf\\u0000\\u0000\x9b:\x0e\x80\x10!\xfb?\x01\\u0000\\u0000\\u0000","stat_t":"inverter/WR1/ch0/MaxPower","unit_of_meas":"W","uniq_id":"123456789012_ch0_MaxPower","dev":{"name":"WR1","ids":"123456789012","mdl":"WR1","cu":"http://192.168.10.12","mf":"Hoymiles"},"exp_aft":95}' on homeassistant/sensor/WR1/ch0_MaxPower/config with encoding utf-8 (for <Job None HassJobType.Callback functools.partial(<function _callback_wrapper at 0x7f9229045e40>, <function async_start.<locals>.async_discovery_message_received at 0x7f91d80c2fc0>, <function async_subscribe.<locals>.<lambda> at 0x7f91dc06e2a0>)>)

mathiaslehmann, 2024-02-17:
Ich habe vergessen zu schreiben welche Version von Home Assistant bei mir läuft. Ist Home Assistant in Version 2024.2.2

MetaChuh, 2024-02-17:
@mathiaslehmann 

bitte mache folgendes:
1) downgrade deine ahoy dtu auf die vorherige, bei dir funktionierende version
2) beobachte das HomeAssistant projekt, ob sie neuere ahoy funktionen bereits unterstützen.
3) öffne ein issue bei [home-assistant/operating-system/issues](https://github.com/home-assistant/operating-system/issues) wenn neuere ahoy versionen noch nicht unterstützt sind
4) schließe das issue hier, da es nicht den funktionsumfang von ahoydtu betrifft.

thx

mathiaslehmann, 2024-02-17:
Habe jetzt mal vorherige Release Version 0.8.36 eingespielt. Hier funktioniert es mit dem Home Assistant Discovery noch. Danach Update auf Version 0.8.83. Solange ich kein neues Discovery starte, läuft alles in Home Assistant. Bekomme also Werte geliefert. Einziges Topic was nicht mehr funktioniert ist dann "ALARM_MES_ID", hier erhalte ich Status "Nicht verfügbar".

rmayergfx, 2024-02-17:
Dito ALARM_MES_ID ist bei mir ab Version 0,8,x auch nicht mehr verfügbar in HA Version 2024.2.1
@MetaChuh 
Die Changes sind in der developer und warum sollte HA daran etwas ändern. btw, schon mal ein Feedback im HA Forum bekommen? Reaktionen dort sind je nach Thema sehr gering. Ich denke eher wir sollten hier an der Developer Release arbeiten, damit es wieder geht!

MetaChuh, 2024-02-17:
@rmayergfx 

> Die Changes sind in der developer und warum sollte HA daran etwas ändern.

ahoy dtu v0.8.83 ist kein developer sondern release build.
imho ist dieses issue irrelevant für das ahoy dtu projekt, und die ha devs haben einfach nicht in der ahoy dev phase partizipiert, darum ihre inkompatibilität.

> @lumapu macht hier einen sehr guten Job und das aus meiner Sicht wesentlich schneller und effizienter.

das finde ich auch, darum unterstütze ihn auch du mit verweisen mancher issues auf die jeweiligen anlaufstellen, statt zu verursachen, dass @lumapu evtl etwas einbaut, was für sein projekt nicht notwendig ist.
wenn er sachen aus good will für andere einbaut, und das dann, mit vielen die ihn dann traktieren, nach hinten los geht, nimmt es auch ihm, so wie bei vielen coolen projekten, die freude, seine sachen mit anderen zu teilen.

rmayergfx, 2024-02-17:
@MetaChuh 
die 0.8.8x vor 0.8.83 waren jedoch dev und da ist das Problem auch schon aufgetaucht.
Und warum bzw. was wurde da geändert? Ich verstehe es nicht wirklich. Ich stelle es nur fest das es mit 0.8.79 noch ging und danach nicht mehr.
Daher ist es für mich immer die Frage Bug oder Absicht und wenn Absicht, warum?

MetaChuh, 2024-02-17:
@rmayergfx meine empfehlung: wenn du ha mit ahoy verwendest, dann bitte auf die letzte funktionierende ahoy version zurückgehen, bis ha ihren code zur aktuellen ahoy release angepasst hat.
bei mqtt gab es änderungen, die du nachlesen kannst.
bei api auch, aber bei den api basierten open source steuerungen die ich kenne, wurde dies bereits vor ahoy release eingebaut.

MetaChuh, 2024-02-17:
hier noch mal die empfehlung für den original issue post, bevor dies auch in diesem issue untergeht:

@mathiaslehmann 

bitte mache folgendes:
1) downgrade deine ahoy dtu auf die vorherige, bei dir funktionierende version
2) beobachte das HomeAssistant projekt, ob sie neuere ahoy funktionen bereits unterstützen.
3) öffne ein issue bei [home-assistant/operating-system/issues](https://github.com/home-assistant/operating-system/issues) wenn neuere ahoy versionen noch nicht unterstützt sind
4) schließe das issue hier, da es nicht den funktionsumfang von ahoydtu betrifft.

thx

rmayergfx, 2024-02-17:
@MetaChuh 
Wer hat Änderungen imBereich MQTT gemacht und warum? Wo ist das nachzulesen. Sorry, ich habe wirklich viel Verständnis, konnte aber in den Changelog nichts dazu finden. Wenn es in einer DEV funktioniert und in der nächsten nicht mehr und ich sehe im Changelog nichts von MQTT gehe ich davon aus, das es ein Fehelr ist.
Hier die Dev Changelog Seite: https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md
Sei doch so nett und poste die Links zu den MQTT Änderungen, damit wir alle schlauer sind.

MetaChuh, 2024-02-17:
added loss rate to /visualization in the statistics window and MqTT
added minimal version (without: MqTT, Display, History), WebUI is not changed! (not compiled automatically)
added max_power to MqTT total values
improved MqTT by marking sent data and improved last_success resends
fix MqTT night communication
combined firmware and hardware version to JSON topics (MqTT)
removed add to total (MqTT) inverter setting
fixed crash once firmware version was read and sent via MqTT #1428
added new setting for future function to send log via MqTT
combined firmware and hardware version to JSON topics (MqTT) #1212

sarcastic side joke: cool dass es immer wieder menschen gibt, die schreiben können, bevor sie lesen können ... no pun intented 😉

lumapu, 2024-02-17:
ich habe was an MqTT geändert um die topics zu reduzieren.
HA ist mir dabei wohl durchgerutscht, ich kann nicht alles testen und in den letzten Tagen keinen bug diesbezüglich gesehen.
Was meckert HA genau? Das oben ist nur zu viel Log, evtl. kann der Fehler nochmal gesondert gepostet werden .

mathiaslehmann, 2024-02-17:
Hallo @lumapu , danke für deine Rückmeldung. Ich werde morgen mal versuchen noch andere Informationen aus Home Assistant heraus zu bekommen und würde die Erkenntnisse hier teilen. Wünsche noch einen schönen Abend. Und nochmals vielen Dank für deine Mühe und dieses tolle Projekt.

mathiaslehmann, 2024-02-18:
Hallo @lumapu ich habe mir HA noch mal angeschaut. Die Meldung ist "Can't decode payload ........ with encoding utf-8". Dazwischen stehen jeweils die Topic Informationen. Es gibt diese Meldung für die Topics YieldDay, YieldTotal, P_DC, Efficiency und MaxPower. Ich hoffe das hilft Dir etwas weiter. Vielen Dank für die Mühe.

mathiaslehmann, 2024-02-18:
Hallo @lumapu ich habe noch einige Versuche gemacht und bin die einzelnen dev Versionen zurück gegangen. Ab Version 0.8.74 bekomme ich die Fehler in HA.

andred, 2024-02-18:
Das problem ist, dass Ahoy seitdem Muell in der discovery sendet:
```
Feb 18 08:44:09 board-b99f4b hass[510]: 2024-02-18 08:44:09.591 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"name":"CH1_U_DC\\u0000\xa6\xfc?\\u0000\\u0000\xff\\u0000\\u0000\\u0000\\u0000\xff\x9f:\x0e\x80\x10!\xfb?\xac\xa1\xfc?","stat_t":"ahoy/hm-1500/ch1/U_DC","unit_of_meas":"V","uniq_id":"116183305178_ch1_U_DC","dev":{"name":"hm-1500","ids":"116183305178","mdl":"hm-1500","cu":"http://10.1.1.240","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"voltage","stat_cla":"measurement"}' on homeassistant/sensor/hm-1500/ch1_U_DC/config with encoding utf-8 (for <Job None HassJobType.Callback <function async_start.<locals>.async_discovery_message_received at 0xffff8f870ea0>>)
Feb 18 08:44:09 board-b99f4b hass[510]: 2024-02-18 08:44:09.597 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"name":"CH1_I_DC\\u0000\xa6\xfc?\\u0000\\u0000\xff\\u0000\\u0000\\u0000\\u0000\xff\x9f:\x0e\x80\x10!\xfb?\xac\xa1\xfc?","stat_t":"ahoy/hm-1500/ch1/I_DC","unit_of_meas":"A","uniq_id":"116183305178_ch1_I_DC","dev":{"name":"hm-1500","ids":"116183305178","mdl":"hm-1500","cu":"http://10.1.1.240","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"current","stat_cla":"measurement"}' on homeassistant/sensor/hm-1500/ch1_I_DC/config with encoding utf-8 (for <Job None HassJobType.Callback <function async_start.<locals>.async_discovery_message_received at 0xffff8f870ea0>>)
...
```

rmayergfx, 2024-02-18:
@lumapu 
Danke für das prüfen

@MetaChuh 
Finde es wirklich nicht mehr Lustig mit welcher Arroganz sich manche Leute hier über andere Lustig machen. Das ganze hier wurde sachlich diskutiert und nie persönlich.
Ich habe speziell danach gefragt wann und wo diese Änderungen dokumentiert wurden. Im Changelog stehen diese nicht. Es sind links auf 2 Issues und diese werden auch immer wieder "closed" und sind dann in einem anderen Bereich zu finden.
Es gibt hier genügend User die aktiv dazu Beitrage das Produkt zu verbessern, es steckt jedoch nicht jeder so tief in der Materie das er alle Problemstellungen und Bugs liest!

lumapu, 2024-02-18:
Ich verstehe beide Seiten, aber du hast Recht durch die stabile Version sind es wieder mehrere issues geworden. Das liegt einfach an dem viel größeren Kreis, der die Firmware jetzt nutzt.
Das ganze ist nicht zu persönlich zu nehmen, es geht darum, dass wir schon seit vielen Wochen die einzelnen Funktionen testen und vorerst davon ausgehen dass keine Fehler mehr enthalten sind.
Wir hatten in der Vergangenheit schon öfter das Phänomen, dass ich zu schnell war gefixt habe was in Endeffekt kein Fehler von Ahoy war l.
Es gibt neben dem offiziellen Changelog noch das Git commit Log, aber kein Problem wenn du zweites nicht kennst.

Zurück zum Thema: danke für den Punkt ab dem es nicht mehr geht, dann schaue ich mir die Änderungen dort an.

rmayergfx, 2024-02-18:
@lumapu 
Du machst einen super Job, Dev sind ja dazu da das getestet werden kann und da es leider so ist das hier auf verschiedenen Plattformen und Bereichen leider unterschiedliche Informationen vorhanden sind ( Discord, git, Git commit, changelog) kann man nie alles unter einen Hut bekommen. Ich teste gerne und gebe auch gerne Feedback an die Entwickler weiter, empfinde es aber also absolut respektlos gegenüber Dritten diese persönlich in Foren anzugreifen, ein Problem des anonymen Internets.

lumapu, 2024-02-18:
ich werde eher eine stabile stable nachreichen, ich musste den Schritt jetzt mal wieder gehen und bin auch weiterhin der Überzeugung, dass das richtig war

andred, 2024-02-18:
> Das problem ist, dass Ahoy seitdem Muell in der discovery sendet:
>
> ```
> Feb 18 08:44:09 board-b99f4b hass[510]: 2024-02-18 08:44:09.591 WARNING (MainThread) [homeassistant.components.mqtt.client] [...]

So wie ich das sehe, sendet Ahoy den kompletten Buffer, und nicht mehr den Null-terminierten string fuer 'name'

lumapu, 2024-02-19:
git diff zwischen `0.8.73` und `0.8.74`: https://github.com/lumapu/ahoy/compare/6b5435a2467bb1ba745d40213f1b7e4d0db359e6...3740a09d2a04e46da340d98cb3479a54a55b0f61

mathiaslehmann, 2024-02-20:
Guten Morgen @lumapu ich habe die dev Version eingespielt. HA Discovery funktioniert wieder. Habe dazu das Gerät in HA gelöscht und per Discovery wieder hinzugefügt. Vielen Dank für die super schnelle Behebung. 
Eine Sache fällt noch auf, die Topic "ALARM_MES_ID" wird per Discovery angelegt, aber es kommen keine Werte von der DTU. Ich persönlich benötige die Topic nicht, vielleicht sollte die Topic dann aus dem Discovery raus? Nochmals vielen Dank für Deine tolle Arbeit.

closed 2024-02-22 by mathiaslehmann

boboG, 2024-03-16:
Habe gerade 1,5 Stunden mit diesem Problem verbracht, bis ich diesen Thread gefunden habe. Danke für den Fix und danke an die HA-Entwickler dafür, das HA die Daten der Ahyo-DTU nahtlos fortschreibt, auch wenn man wie ich das MQTT-Gerät gelöscht hat, weil man nicht mehr weiter wusste.

derlm, 2026-03-02:
Danke Mathias, den Gedanken hatte ich auch schon. 
Problem ist, dass ich die HW nicht einfach so tauschen kann, da mein Vater 2h entfernt wohnt. Mein Support läuft überwiegend über Teams und Webzugriff auf Homeassistant. 
Zudem funktioniert die Kommunikation zwischen HM-600 und DTU sowohl mit der alten, als auch der neuen FW einwandfrei - nur die Discovery seitens DTU zum Homeassistant klappt "nur halb".

derlm, 2026-03-06:
Hier nochmal ein Update: nachdem wir sowohl HA als auch AHOY heruntergefahren, vom Netz getrennt, neu gestartet und alle Einstellungen Zeile für Zeile mit meiner laufenden Box verglichen haben, funktionierte es, als ob nichts gewesen wäre. 
Hatte vorher noch die MQTT Integration komplett vom HA deinstalliert, nach dem Neustart neuinstalliert und dann unmittelbar das Discovery gestartet. HW Stand jetzt ist nach wie vor ESP8266 mit 250812_ahoy_0.8.156_c4fe30f_esp8266-all-de.bin. Trotzdem vielen Dank für Deine Antwort!