lumapu/ahoy #1434: [Bug] MI Inverters not working with startFastWrite usage

joerocklin, 2024-02-17:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a8

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

I have MQTT set up, pin settings for my device (a feather huzzah32), and updates for region/timezone. I don't know what else has been changed.

### Error description

The commit in 0.8.77 (15349520) caused all communication with my MI inverters to cease. I've tracked it down to the change in `hmRadio.h` from `mNrf24->startWrite` to `mNrf24->startFastWrite`: https://github.com/lumapu/ahoy/blob/5ebfe5a8fb5bcc833b64a0dd083686031808cec8/src/hm/hmRadio.h#L391

Changing this back to the following restores communication with the inverters: https://github.com/lumapu/ahoy/blob/7c532ca1cc63bfda9a68cfb9b7baf23d1d17f94b/src/hm/hmRadio.h#L418

The [documentation for the `startFastWrite` call](https://nrf24.github.io/RF24/classRF24.html#acd19843064cb70ec23507412e519e4ef) mentions needing to call [`txStandBy`](https://nrf24.github.io/RF24/classRF24.html#a12cc453453c94969d4d3f0edb3778c83), and I don't see that anywhere. I'm not very familiar with the ahoy code, so I'm not sure where it would be appropriate to put that call.

lumapu, 2024-02-17:
@rejoe2 @tictrick did you hear from txStandby command? 
I can't imagine that MI inverters aren't accessible and more because rejoe2 has at least 4 of them.

knickohr, 2024-02-17:
Please specify your MI-Inverters.

joerocklin, 2024-02-18:
I have MI-1200s and possibly a couple of MI-1000s. 

Im not 100% sure though. Part of my interest in this project was because I don't think the people who installed my system used what they told me they were going to.

rejoe2, 2024-02-18:
Do the serial numbers start with 1061 (2nd gen MI) or 1062(3rd gen)?

Nevertheless, issueing a txStandby() when entering the rx mode in hmRadio might be a good idea.
We did some tests with 2nd gen 4ch MI as well, so basically the startFastWrite() command imo isn't the root cause. But there is at least one known bug with MI (state machine) breaking MQTT coms in some cases after new day has started.

joerocklin, 2024-02-18:
All of them start with 1061. I know I have 4 inverters each with 4 panels connected and 2 more that each have 3 panels. The serial numbers on them make me think they are all 1200's. 

I was wondering about the MQTT pieces as well. I'm happy to help figure things out any way I can. I'm comfortable with C/C++, just don't know a lot about Ahoy codebase.

rejoe2, 2024-02-22:
@joerocklin - any news on this?
Note: there's been an update on the MI state issue as well, but unfortunately this is more or less just to get the code more readable and more logical structured. To be honest: I did no longer understand what I had coded before...

The problem itself seems to be the Alarm array to be somehow/somewhen reorganized, so the single values stored there are no longer at the same place as before. (Or there's another mistake I wasn't able to find until now).

joerocklin, 2024-02-22:
I haven't been able to get back into the code too much yet. I did pull in the changes and tested again though. A quick check just showed that it still doesn't seem to get anything from the inverters with the `StartFastWrite` call (I waited about 5 minutes. Putting the `StartWrite` call back  started receiving after the 3rd comm loop.

rejoe2, 2024-02-22:
Additionally: please provide some more info on the nRF module you are using. As you mentionned an external antenna, it's pa+lna version, right? Shielded, unshielded? Sure it's genuine nordic transceiver in it?

joerocklin, 2024-02-22:
I am using a pa+lna version. It is not shielded. I thought it was an authentic nordic transciever, but it looks like there's a round dot on the IC, so I would call that suspect.

Here are some logs, pulled from a direct serial connection and not the webserial output (I missed the boot part of the startFastWrite, but it starts before wifi is connected):
[MI-StartWrite.log](https://github.com/lumapu/ahoy/files/14377768/MI-StartWrite.log)
[MI-StartFastWrite.log](https://github.com/lumapu/ahoy/files/14377769/MI-StartFastWrite.log)

rejoe2, 2024-02-22:
OK, so a "blob" nRF module is really suspicious... In the good old MySensors days these had often been described as "rubbish", esp. the range is really small with these afai remember.

But: First make sure, Ahoy can reach a NTP server (or set one with your browser). Just yesterday we had a (regular HM) user here with a rather similar log without any reception and no time info in it (see https://github.com/lumapu/ahoy/issues/1440#issuecomment-1953649431).

Additionally: Event when it's working with "startWrite()", the rf performance is rather poor. Giving a slightly increased power level a try might be a good idea (no guarantee, esp. with the "blob" "nRFs").

rejoe2, 2024-02-22:
@lumapu - any idea why keeping CE high seems to conflict with a non set time in the mcu? Should we actively put it to low after sending is over? 
Additionally: In the oscilloscope pictures you showed on discord to evaluate the CE state with both of the funcions (and before as well!) the CE state was HIGH long (?) before the Write command has been transferred to the nRF. This imo is suspicious. CE should be low as long as we do neither send nor wait for frames from the inverters. Or did I miss something?

lumapu, 2024-02-22:
I have no idea and I see no correlation between both. The time should be synced directly after booting the ESP.
Are there more logs as the both on a previous post here where you refer to?

rejoe2, 2024-02-22:
> I have no idea and I see no correlation between both. The time should be synced directly after booting the ESP. Are there more logs as the both on a previous post here where you refer to?

No, and there's no confirmation from @joerocklin until now on startFastWrite() working as expected as soon as time is synced. Nevertheless this coincidence imo is remarkable, so let's see what @joerocklin reports on that...

I had a look in the code wrt. to why there's the tx log entry at all - even as it seemed to be "night time" in the other case. @joerocklin - do you have "no communitation at night" activated as well? Does deactivating this feature make a difference also when using startFastWrite()?
@lumapu - not sure, but perhaps we have the commands send out in the startup phase but do no longer listen if it's "night"? Strange...

joerocklin, 2024-02-23:
@rejoe2 I do have 'Pause communicate at night' activated for all inverters. 

I tried pushing the power up to 'High', but that didn't show any change. So I tried the 'Max' setting, and I started getting data. But then it stopped communicating with all inverters. I reverted back to the `startWrite` method and lower power and it started working. I'll see if I can move the board closer to the inverters this weekend and do some more testing.

rejoe2, 2024-02-24:
> @rejoe2 I do have 'Pause communicate at night' activated for all inverters.
 
Do you have a synced time? Or is the missing timestamps just caused by using USB serial debugging? (You may simply use the Webserial debug. At least here, it takes around 11 seconds to log in there, the output starting from then on is sufficient imo.. 

> I tried pushing the power up to 'High', but that didn't show any change. So I tried the 'Max' setting, and I started getting data. But then it stopped communicating with all inverters. I reverted [...]

Sounds to some extend like a problem with the power consumption of the longer activated pa+lna? (This is the effective difference of the two "write" methods). Could you please try a different power supply (and/or just a different USB cable) for the DTU?

rejoe2, 2024-03-01:
@joerocklin - any news?

joerocklin, 2024-03-05:
Work and life got really busy and I haven't had a chance to sit down and investigate.

rejoe2, 2024-03-05:
OK, thanks for the short update.

stefan123t, 2024-10-29:
> The [documentation for the `startFastWrite` call](https://nrf24.github.io/RF24/classRF24.html#acd19843064cb70ec23507412e519e4ef) mentions needing to call [...]

@2bndy5 could you have a quick look at this and give us a short recap of the right way to use the RF24 lib, please :blush: ?
`startWrite` vs. `startFastWrite` and what the `txStandBy` may be used for.

2bndy5, 2024-10-30:
We had a brief discussion about the various `write*()` confusion (see https://github.com/nRF24/RF24/issues/816). We ended up augmenting the docs to try and relieve the confusion (see ["which write should I use?"](https://nrf24.github.io/RF24/md_COMMON__ISSUES.html#autotoc_md181)).

<details><summary>Some opinionated ranting</summary>
<p>

Personally, I never liked how manicBug originally wrote the multitude of `write*()` methods (over 10 years ago). They all focus on what I call "babysitting" as if all RF24 users are infants that won't understand async concepts.

I personally have written a few different implementations for the nRF24L01. And each implementation only has a blocking `send()` and a non-blocking `write()` (which would be used in conjunction with other API about processing the Status/IRQ flags). I'm currently writing an embedded rust implementation (with python & node.js bindings) using this approach (again).

</p>
</details>

If you are using interrupts, then the code doesn't need to explicitly call `txStandby()`. Instead, you would just handle the interrupt to see if the transmission succeeded or not (using [`RF24::whatHappened()`](https://nrf24.github.io/RF24/classRF24.html#afb97dc4bdf4d2d84ea44060ac5b4ed89)).

If `startFastWrite()` is failing but `startWrite()` is working, then it is likely that the radio's CE pin is not being held HIGH long enough (mandatory minimum 10 microseconds) to initiate the transmission. This is the only real difference between the 2 functions.

## Looking at the [source link in OP](https://github.com/lumapu/ahoy/blob/5ebfe5a8fb5bcc833b64a0dd083686031808cec8/src/hm/hmRadio.h#L391)
I see the code flushes the RX FIFO (👍🏼) and enters TX mode (`stopListening()` also calls `flush_tx()` when ACK payloads are enabled 👍🏼) immediately before calling `startFastWrite()`. This makes me think that the mandatory 10 us pulse on CE pin is not satisfied.

rejoe2, 2024-10-30:
Sorry for beeing out of all that stuff for quite some time now. Atm. doing some tests would be a little complicated as well..

> This makes me think that the mandatory 10 us pulse on CE pin is not (always) satisfied.

On the one hand, that sounds quite logic, but on the other side, afair the cause for using startFastWrite() had been startWrite() putting CE to low to soon.  Our intention had been to keep CE high for much longer than the hardcoded 10us in startWrite(), as it had turned out to be contraproductive for receiving ACKs from the inverters for those using PA+LNA modules. 

So where's the point in the code to put CE to low to early? Short look points to stopListening(), but that's called when a (much longer) timeout is reached or we got all fragments expected. Confused.

On the other hand, adding a short (10ms) delay after the mMillis = millis(); line should not do any big harm and should make sure, the requirements are met. 
Right, or did I miss something essential?

@lumapu - Your opinion on that?

2bndy5, 2024-10-30:
To clarify, I said mandatory 10 microseconds, not 10 milliseconds. This is important when interrupts are involved. I'm not familiar with all the code here, but typically you shouldn't `delay()` during a ISR.