lumapu/ahoy #1435: ESP8266 - Nach Update von 0.8.36 auf 0.8.83 leuchtet die blaue LED dauerhaft !?

mroenne2022, 2024-02-17:
## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ X] circuit board
* [ X] external antenna (SMA)

Bei meinen beiden ESP8266 leuchtet nach dem Update von 0.8.36 auf 0.8.83 die blaue LED dauerhaft. Zuvor war das nicht der Fall , lediglich immer kurz zum Zeitpunkt des Sendens hat sie kurz "aufgeblinkt" ! Kann man das ggf. abstellen ? Auf den ersten Blick hab ich da nichts gefunden !

lumapu, 2024-02-17:
Das liegt daran, dass die blaue LED beim ESP8266 am Pin D4 hängt.
In unserer Standardkonfiguration ist dieser mit dem IRQ des NRF24 verbunden und dürfte daher nur kurz blinken.
Geändert haben wir das Verhalten von dem Pin CE des NRF24, dort liegt nun die meiste Zeit ein high Pegel an.
Ich nehme an, dass in deiner Hardware an Pin D4 des ESP der CE Pin des NRF24 hängt. Ist das der Fall, dann benötigst du nur einen Lötkolben um die Pins D3 und D4 zu tauschen. Danach nicht vergessen gleiches auch in der Konfiguration von Ahoy zu tauschen.

mroenne2022, 2024-02-17:
@lumapu 
Das war offensichtlich die Lösung. Eine DTU war mit Steckverbindung, deshalb konnte ich mal eben tauschen (in der Software/Konfig ebenfalls getauscht). LED nun aus 👍 Ich geh davon aus das die DTU morgen weiterhin Daten liefert.  Vielen Dank für Info und Erklärung 😀

lumapu, 2024-02-17:
ja hoffen wir dass, morgen wissen wir mehr. Danke für's Feedback

mroenne2022, 2024-02-18:
@lumapu 
DTU läuft prima wie zuvor, blaue LED bleibt aus. 
Alles bestens, nochmals vielen Dank 👌
Dann können wir hier schließen denk ich.

closed 2024-02-18 by mroenne2022

DB1BMN-collection, 2024-02-18:
Guten Morgen,

 ja, das dauerhafte Leuchten der blauen LED auf dem 8266 habe ich auch bemerkt in der letzten Beta 0.8.82 und beim Release 0.8.83. Hatte aber die Vermutung, dass es an den häufigen Abstürzen wegen der Diagramme liegt und habe wieder auf die 0.8.36 downgegradet, siehe #1436 

Beim ESP8266 ist es ja so, dass die blaue LED an IO02/D4 gegen GND gezogen wird, damit sie leuchtet.

In den vorherigen Versionen bis 0.8.36 war sie ja dauerhaft aus, d.h. D4 und somit Chip-Enable lagen dauerhaft auf High.

Gibt es einen triftigen Grund (außer Stromsparen?), dass man jetzt D4/CE dauerhaft auf Low legt und nur kurz zum Kommunizieren den nRF enabled?

Sorry, ich habe das Projekt längere Zeit nicht verfolgt, bin aber beeindruckt von den neuen Funktionen *thumbsup*, weiter so!

lumapu, 2024-02-18:
Das hat nur technische Gründe, bzgl. Strom sparen haben wir Module die Richtung Sonne schauen 😉.

fila612, 2024-02-19:
@lumapu: was würde passieren wenn die Pins nicht so einfach geändert werden können, weil z.B. die Platine das nicht hergibt? Hat dies technische Auswirkungen oder leuchtet einfach nur die LED dauerhaft? ich habe aktuell den CE an D4 und kann nur den IRQ zwischen D1/D2/D3 wählen, da es hier in der Vergangenheit immer Probleme mit diversen ESPs gab, habe ich einen Jumper eingebaut - bei CE aber nicht :(

You69Man, 2024-02-19:
Ein schwarzer Edding Stift könnte Abhilfe schaffen ;-)

DB1BMN-collection, 2024-02-19:
Ja, das mit dem Edding steht sogar schon in meiner Anleitung drinne ;-)

Hm, könnte man denn nicht den Idle-Zustand des CE-Pinis in Software konfigurierbar machen?
Und erst wenn eine Kommunikation stattfindet dann den gewünschten Zustand annehmen und ggf. eine Seutup-Time abwarten, bevor man den /CS runterzieht? - Mit nem Radio-Button oder so ließe sich das doch leicht umschalten, oder?

Oder nimmt es das nRF-Modul übel, wenn man ständig an seinem CE wackelt?

mroenne2022, 2024-02-23:
@LeSpocky 
Nach dem Tausch blinkt die blaue LED nur immer 1x ganz kurz auf wenn Daten abgefragt/gesendet werden, ansonsten ist sie aus. So war es vor dem Tausch bei den vorherigen Versionen auch.
Allerdings hab ich D3 und D4 getauscht,  nicht D2 !!