lumapu/ahoy #1436: [Bug] 0.8.83 (@ESP8266) still Resetting when Plotting Diagrams enabled

DB1BMN-collection, 2024-02-18:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Upgraded from 0.8.36 -> 0.8.82 -> 0.8.83 by uploading *.bin file thro Ahoy´s Web-UI.
Downgraded then back to 0.8.36

### Error description

When enabled plotting Diagrams the firmware restartes every few Minutes.
Blue LED continuously on (CE = D4, IRQ = D3), see [https://github.com/lumapu/ahoy/issues/1435#issue-2140501941](url)

You69Man, 2024-02-18:
Hast du einmal versucht, die Einstellungen zurückzusetzen ("Erase Settings"), und alles manuell zu konfigurieren und dann schauen, ob das Phänomen dann immer noch auftritt?

DB1BMN-collection, 2024-02-18:
Guter Punkt, danke!
Hab jetzt mal ein "Factory Reset" gemacht (so hieß es noch in der 0.8.36) und alle Daten neu eingegeben.
Werde das mal beobachten.

DB1BMN-collection, 2024-02-18:
OK, hab das jetzt mal 45 Minuten beobachtet und es scheint keinen Absturz zu geben und das Diagramm in der Webansicht wird geplottet in einem Intervall von 5 Sekunden.

Auf dem Display sehe ich jedoch nur ein leeres Diagramm. (Show Ratio 50 %, Line 1-3).
Habe jetzt mal auf Line 2-3 gestellt und gucke weiter.

Dirk-Ahoy, 2024-02-18:
Du musst die Koordinaten beim Sunset eintragen sonst geht der Display-Graph nicht.

DB1BMN-collection, 2024-02-18:
> Du musst die Koordinaten beim Sunset eintragen sonst geht der Display-Graph nicht.

Cool, danke!
Habs jetzt mal eingetragen, testen kann ich ja erst morgen.

Ist das Absicht oder ein Bug? - Die Web-Darstellung funktioniert ja auch ohne Koordinaten.

You69Man, 2024-02-18:
Ist kein Bug, sondern wird für die korrekte Skalierung auf den Tagesverlauf benötigt.

DB1BMN-collection, 2024-02-19:
So, jetzt habe ich das mal den ganzen Tag durchlaufen lassen.
Als ich die Koordinaten eingetragen habe, funktionierte das Plotting auch anfangs.

Es muss aber dennoch mindestens einen Absturz gegeben haben, denn als ich gegen 18:20h heim kam, war kein Diagramm sichtbar weder im Display noch in der Web-Ansicht.
Als "Uptime" zeigte es ca. 45 Minuten an, der Absturz muss als sogar noch vor dem Sonnenuntergang (ca. 17:45) passiert sein.

lumapu, 2024-02-19:
wäre es möglich über USB ein Log mitlaufen zu lassen?

DB1BMN-collection, 2024-02-19:
Ähm, den ESP per USB an einen Rechner anschließen und per COM-Port die Daten in eine Text-Datei schreiben?
Sollte ich hinbekommen, muss ich nur den alten Hammer-Terminal ausgraben.
Ich setz das morgen früh auf, bevor ich losfahre.

Danke schon mal im Voraus!

DB1BMN-collection, 2024-02-20:
So, heute morgen gegen 7:30h gestartet, es mit Br@y-Terminal [1] aufzuzeichnen.
Gegen 7:46h erfolgte der erste Reset durch den Watchdog.
Um 7:53:20h hab es dann eine "User exception (panic/abort/assert)" / "Unhandled C++ exception: OOM" mit anschließendem Stack-Dump.
Um 8:06:12h hat der Watchdog wieder zugebissen.

Gegen 9:04h stürzte das Laptop aus unbekannter Ursache mit  einem Bluescreen ab.

In Angang das Logfile, hoffentlich ohne kompromittierende Daten.
[20240220073232.log](https://github.com/lumapu/ahoy/files/14350828/20240220073232.log)

[1] https://sites.google.com/site/terminalbpp/

lumapu, 2024-02-20:
maybe that's too much for the small ESP. History (can't be disabled) and display.
is the ESP stable if you disable the display in the settings?

DB1BMN-collection, 2024-02-20:
Will try tomorrow

Dirk-Ahoy, 2024-02-22:
du solltest mal im Log die Daten vom WR abstellen und Privatemodus dann kommt da viel weniger - kann man dann besser auswerten.

lumapu, 2024-02-22:
what happens if you completely disable the display?

DB1BMN-collection, 2024-02-22:
OK, set Display type to "None". Will try tomorrow.
After sunset, when the inverter was not delivering it was fine even when the "Night behaviour" was active: 
[image]

DB1BMN-collection, 2024-02-23:
So, ich hoffe, es ist OK, wenn ich auf deutsch schreibe?

Also habe, gestern Abend das Display komplett deaktiviert (Display "None"). Erstaunlicherweise bleibt die letzte Anzeige dabei erhalten.
Nach einem Neustart kurz vor Mitternacht lief die DTU dann bis zum Sonnenaufgang durch (Uptime ca. 9 Stunden).

Habe dann das Logging im Terminal gestartet und hänge es hier an: 
[20240223084816.zip](https://github.com/lumapu/ahoy/files/14390106/20240223084816.zip)
Während des Tages gab es dann 8 "wdt resets".
Nachmittags erfolgte der letzte Neustart um 15:17h und hielt sich dann bis zum Sonnenuntergang, wo ich das Logging wieder gestoppt habe.

lumapu, 2024-02-25:
ok, jetzt wissen wir, dass das Display nicht schuld ist. Ist eine Abfrage der RestApi von irgendwo extern aktiv?

DB1BMN-collection, 2024-02-25:
> Ist eine Abfrage der RestApi von irgendwo extern aktiv?

Soweit ich weiß, nicht.
Es wird nur per MQTT an HomeAssistant weitergefunkt, aber nichts aktiv abgefragt.
Ich habe jetzt mal MQTT deaktiviert. Schauen wir mal weiter.

DB1BMN-collection, 2024-02-27:
So, wie erwaret lief die DTU nun die ganze Nacht und den ganzen Tag durch ganz ohne Absturz und hat ein schönes Diagramm gezeichnet. MQTT war dabei aus, also scheint es daran zu liegen.
Wundert mich, dass es sonst noch niemandem aufgefallen ist. - Oder nutzt ihr alle den großen ESP32?

[image]

DB1BMN-collection, 2024-02-27:
Was ich noch testen will:
Ich hatte ja [eingangs ](https://github.com/lumapu/ahoy/issues/1436#issue-2140865325)geschrieben, dass ich von 0.8.36 auf 0.8.83 upgedatet habe über das Web-UI.
Es wurde daraufhin [empfhohlen ](https://github.com/lumapu/ahoy/issues/1436#issuecomment-1951075642), einen "Factory Reset" zu machen und alle Einstellungen neu einzugeben.
Ich werde jetzt den ESP neu per USB-Kabel löschen und mit dem Webinstaller neu flashen, das Display und MQTT wieder aktivieren und gucken was es morgen macht.

DB1BMN-collection, 2024-02-28:
Ja, wie erwaret gab es heute jede Menge Neustarts, wenn MQTT akttiviert ist.
Hate einen frischen Wemos D1 mini per USB-Kabel und Webserial auf die 0.8.83 programmiert und die Koordinaten wie oben eingegeben.
Der erste Abstur war ca. 20 Minuten nach Sonnenaufgang, der letzte rund eine Stunde vor der letzten Übertragung.
Spiele jetzt die 0.8.86 auf und teste morgen wieder weiter.

Dirk-Ahoy, 2024-02-28:
Ich denke du solltest auf ESP32 oder besser noch Fusion Board upgraden - der 8266 is ja schon recht schwach und ziemlich an der Grenze  - mein Tip :)

DB1BMN-collection, 2024-02-28:
Das ist schade, denn mit MQTT alleine lief es ja den ganzen Sommer durch.
Erst mit dem Diagramm kam dieses Problem.
Auch die Diagramme alleine funktionieren ja. Ich denke, da lässt sich noch was optimieren
Ich werde morgen mal den Heap und die Fragmentierung beobachten.

Das Problem ist auch, dass ich auf das Format des Wemos D1 mini festgelegt bin durch die Leiterplatte.

Dirk-Ahoy, 2024-02-28:
Ja neue Features - dann ist der "alte" am Ende - steht ja gaub auch irgendwo das mqtt nicht empfohlen ist beim 8266 - muss man abwägen alte Version oder neue mit stärkerer Hardware

lumapu, 2024-02-28:
evtl. lohnt ein weiterer Test morgen, ich habe in der `0.8.88` einen MqTT Fehler gefixt: schaue mal bei #1458

DB1BMN-collection, 2024-02-29:
Danke, hab noch in der Nacht die 0.8.88 (warum muss ich nur an Lego denken?) installiert.
Leider keine abhilfe. Zahlreiche Neustarts im laufe des Tages :-(

Fassen wir aber zusammen:
* Es passiert nur, wenn MQTT aktiviert ist;
* es ist egal, ob das Display an oder aus ist;
* es ist egal, ob das Diagramm im Display aktiviert ist;
* es passiert nur tagsüber, wenn der der Wechselrichter ordentlich produziert, also viel Traffic kommt.
** Der letzte Absturz passiert ca. 1h vor Sonnenuntergang, aber danach "bekomme" ich noch die letzten paar Wh mit als Zacken auf dem Display. Das bleibt so bis in den Abend hinein.

Ich werde die nächsten Versionen weiter beobachten und antesten

lumapu, 2024-02-29:
Danke für's Feedback. Könntest du auf deinem MqTT Broker prüfen, ob bei den JSON Payloads die schließende Klammer `}` da ist? Vor allem bei den Statistikdaten ist das interessant.

MetaChuh, 2024-03-01:
@DB1BMN-collection 

> Das Problem ist auch, dass ich auf das Format des Wemos D1 mini festgelegt bin durch die Leiterplatte.

kein problem, den esp32 gibt's pinkompatibel in der d1 mini version von zb. az-delivery (direkt oder amazon).
die hab' ich genau aus den gleichen gründen für die upgrades genommen.

letzte stabile version auf esp8266 mit heavy api usage war bei mir v0.6.9, darum musste ich auch den esp upgraden, oder auf features verzichten.
mit dem esp32 d1 mini als drop in replacement auf meiner gesockelten eigenbauplatine, rennt es seit v0.8.83 für mich nahezu fehlerfrei.

[image]

MetaChuh, 2024-03-01:
ps: hab's gleiche gehäuse wie deins, der esp32 ist bei mir, zum platz sparen sandwich bauweise unter dem oled display

[image]

DB1BMN-collection, 2024-03-03:
Hehe, die Lösung mit den LEDs in den Display-Löchern gefällt mir!
Ich hab jetzt mal den "S2 mini" bestellt zum Testen: https://de.aliexpress.com/item/1005006160675425.html
Für die HMS-Wechselrichter brauchen wir ja eh den ESP32.
Ein Träum wäre mit USB-C und Antennenbuchse...

Ist natürlich schade für all die Leute, die schon einen D1 mini haben und sich irgendwann wundern, warum ihre Kiste regelmäßig abschmiert. - Naja, ich beobachte das mal weiter.

MetaChuh, 2024-03-03:
> Hehe, die Lösung mit den LEDs in den Display-Löchern gefällt mir!

hahaha, ja, ist definitiv klassich old school entstanden ... fertiges ding ohne leds fertig im garten ... bier aufmachen und d'ran erfreuen ... lange anstarren bis man merkt "was macht man mit den löchern" ... dann 2 leds dazubauen ... nochmal bier aufmachen und nochmal d'ran erfreuen 😂

(danach war ich schon zu betrunken, um @lumapu zu fragen, ob man noch zwei weitere gpio's für leds anreißen kann, die man auch plausibel nutzen könnte 🙈😂)

> Ich hab jetzt mal den "S2 mini" bestellt zum Testen

werden identisch funktionieren wie die esp32 d1 mini von az-delivery.
einziger unterschied ist die qualitätskontrolle und garantie, also wie immer 3er pack kaufen.

> Ist natürlich schade für all die Leute, die schon einen D1 mini haben und sich irgendwann wundern, warum ihre Kiste regelmäßig abschmiert

auch die versuchen wir zu supporten ... wird aber immer schwieriger, weil brauchen tun' sie v0.6.9, jedoch wollen alles neu, was es am commodore amiga gibt, und nicht mehr am vorhandenen c64 läuft 😉

greetings und glg,
metachuh

You69Man, 2024-03-03:
> Ist natürlich schade für all die Leute, die schon einen D1 mini haben und sich irgendwann wundern, warum ihre Kiste regelmäßig abschmiert. - Naja, ich beobachte das mal weiter.

Es ja immer noch nicht klar, woher diese Resets kommen. Ich betreibe auch einen D1 mini, Esp8266, mit immer neuesten FW Versionen, ohne jegliche Reboot Probleme.

DB1BMN-collection, 2024-03-04:
@MetaChuh Magst du bitte hier mal gucken: https://github.com/lumapu/ahoy/issues/965#issuecomment-1977502784
Vielleicht hast du ne Idee, wie ich den S2 programmieren könnte?

MetaChuh, 2024-03-05:
@DB1BMN-collection 
😂 ok 😉

MetaChuh, 2024-03-07:
ich bin generell immer für schließen 😉

You69Man, 2024-03-07:
> ich bin generell immer für schließen 😉

Ja, wäre auch eine Möglichkeit, aber ich fürchte, davon wird das eigentlich Reboot-Problem (welches immer das ist) nicht behoben.
Also ich denke schon, dass da noch was ist, was gefunden werden muss. 
Zwei verdächtige Kandidaten: Irgendwas im MQTT, und/oder Zeitsynchronisation/NTP

Prinzipiell glaube ich nicht, dass man die Antwort auf "naja, der ESP 8266 ist halt zu schwach" vereinfachen sollte, denn zumindest mit meinen HM-800, langweilt sich auch mein 8266 immer noch mit jeder Menge Idle-Time. Nur spontan kann es ihm vielleicht hie und da einmal kurz zu viel werden, aber so etwas ließe sich normalerweise gut fixen,  wüsste man einmal wo der Hund begraben ist (wir haben es ja nicht mit einem Echtzeitsystem zu tun). 

D.h. jede weitere Rückmeldung, jedes weitere Testszenario (egal ob gut, oder schlecht), kann weitere wertvolle Erkenntnisse bringen. 
Aber wenn die Headline schon einmal Vorurteile schürt, ist man von vorn herein am falschen Dampfer.

DB1BMN-collection, 2024-03-07:
Danke,
ja ich logge ja jetzt die Uptime und die RSSI in Home-Assistant und bekomme eine schöne Sägezahn-Kurve mit vielen Restarts am Tag. - Nachts läufts durch.
[image]
Ich hoffe, dass ich am WE auch noch den Heap plotten kann.
Wie ich schon [schrieb](1481#issue-2170241140 ), wäre es cool (zumindest fürs Debuggen), wenn auch die Uptime, RSSI und Heap sich als Entitäten (was für ein Wort!?) bei Home Assistant melden würden, so wie sich ja auch jeder Wechselrichter mit rund ca. 40 Entitäten anmeldet.

Dass der 8266 zu schwach ist glaube ich auch nicht, denn z.B. Tasmota kann ja auch MQTT und Diagramme plotten.

Ich hab mir zwar den S2-mini geholt, aber der scheint wohl [verfused ](965#issuecomment-1984627816)zu sein.

DB1BMN-collection, 2024-03-10:
So,
kleines Update mit kleinen Erkenntnissen.
Da ich offensichtlich einen zu hohen Ableitstrom im Kabel zum Wechselrichter habe, ließ ich dieses abgezogen, so dass keine Einspeisung erfolgte.
Der Wechselrichter kommunizierte aber weiterhin mit der DTU und meldete z.B. seine Temperatur.
Ergebnis: Die Laufzeit der DTU verlängerte sich signifikant.
Heute Nachmittag kam dann der erste Restart nach über 159.000 Sekunden. - Für mein Verständnis keine "Magic Number" im Binärsystem.
Keine Auffälligkeiten bei den anderen Werten. RSSI und Heap.
Version ist 0.8.91

[image]

N.b.: Am 2024-03-08 gegen 17h habe ich einen manuellen Neustart durchführt, diesen also bitte nicht weiter beachten.

Vielleicht sollten wir uns auch mal den Stackpointer anschauen? - Weiß jemand, wie man den plotten kann?

lumapu, 2024-03-10:
auffällig ist für mich die Heap-Fragmentation. Das ist auch der größte unterschied zum ESP32, der deutlich mehr Heap hat. Ich werde in Zukunft wieder mehr auf den ESP8266 schauen, da er irgendwie etwas ins Hintertreffen gelangt ist.
Danke für deine Geduld und die guten Reports

DB1BMN-collection, 2024-03-10:
Bitte gern geschehen :-)

Ja, die Heap-Fragmentierung habe ich auch versucht zu verstehen.
Auffällig ist, dass in der Nacht vom 9. zum 10. März der Wert zwar relativ ruhig ist (klar, gibt ja auch keine Kommunikation zum Inverter), aber mit "13" im Durchschnitt sogar höher als am Tag.

Heute Nachmittag *vor* dem Absturz war der Wert wieder so um 10 mit ein paar kurzen Zacken.
Nach dem Reset war die Fragmentierung bei 33.
Der nächste Restart kam rasch nach ca. 15 Minuten, wobei die Fragmentierung weiter so hoch blieb.

Jetzt hat es um 15:17h wieder einen Restart gegeben, wobei die Fragmentierung nun auf 5-8 runterging.
Mal sehen, ob es dann wieder länger durchhält.

[image]

Solange ich kein Ersatzkabel habe, lasse ich das mal so weiter durchlaufen.

Zordandi, 2024-03-20:
Ich kann die Probleme mit MQTT auf einem ESP8266 bestätigen, welcher den Hardware Watchdog auslöst.
Ich habe vor ein paar Tagen mit der Verwendung von Mosquitto auf Home-Assistant begonnen.
Dabei habe ich zuerst die v0.8.83 verwendet und musste wegen dem Bug bei MQTT auf v0.8.84 gehen.
Seit der Aktivierung von MQTT sind mir mehrmalige Neustarts wegen Hardware Watchdog aufgefallen.
-> kann ich irgendwie bei der Problemfindung helfen?!?

lumapu, 2024-05-30:
bitte die ganz neue Version `0.8.123` mal testen.

DB1BMN, 2024-05-31:
Oh, so viele Neuerungen! Danke!

Habs heute morgen um 5:00 h gleich mal installiert auf dem 8266.
Folgende Beobachtung: Im Browser werden die seitlichen Menüs nicht anzeigt und die Startseite ist "blind". Strg + F5 habe ich mehrfach gedrückt. NoScript ausgeschaltet.
Reporting per MQTT läuft aber weiter. Ich lass das mal weiter laufen
[image]

lumapu, 2024-05-31:
ich habe keine Ahnung woran das liegt, scheinbar wurde irgendwie die `api.js` nicht aktualisiert - kanns mir nicht erklären. Ich hatten den gleichen Fehler auch hier bei mir - jetzt mit der `-all` geht es. Hoffentlich komme ich da noch dahinter.

stefan123t, 2024-10-30:
@DB1BMN-collection / @DB1BMN und @Zordandi bitte mal mit der aktuellen Version 0.8.152 testen und zurück melden ob das Problem noch besteht ? Danke!

Was die Heapauslastung angeht scheint der ESP8266 mit 14x 1024 Bytes in der Regel stabil zu laufen.
Zeiten mit nur 8x 1024 Bytes scheint der ESP8266 z.B. mittags auch durchaus auszuhalten.

Zur Heap Fragementierung könnte prinzipiell schon die MQTT Bibliothek beitragen.
Ich meine die JSON Daten werden zuerst in lokalen Datenstrukturen abgelegt und dann versendet.

@lumapu macht es evtl. Sinn dabei die Größe des JSON Buffers jeweils mit zu loggen ?