lumapu/ahoy #1437: [Bug]

jrnas, 2024-02-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a8fb5bcc833b64a0dd083686031808cec8

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nothing changed

### Error description

since 0.8.38 update randomly (every 1,2 minutes) inverters were on status "available and is not producing"

juepi, 2024-02-19:
Hi jrnas,

Go to the Settings of your inverter and verify if the checkbox `reset values when inverter status is 'not available'` is active. If it is, deactivate the setting and reboot AhoyDTU and verify if the problem still persists.

yours,
Juergen

jrnas, 2024-02-19:
It helps yes, but i would like to know if the inverter is available or not, it would be great if there is a timeout that can be configured in the UI

juepi, 2024-02-19:
Ok, then it may be related to #1035 @lumapu ?

Jrnas, this was just for verification ;-)

jrnas, 2024-02-19:
Yes but before the update it works, but since update not correctly, it seems that the timeout has been lowered or something

lumapu, 2024-02-19:
Which version did you use before?
Do you have an idea how much your inverter produced once this happened? Was it in the early morning?

jrnas, 2024-02-19:
The old release one 0.8.36. no it doesn’t matter whitch time

knickohr, 2024-02-19:
Please check your power settings in the inverter Settings tab.

tl-photography, 2024-02-23:
I have similar issues. 

I recently updated to 0.8.83 from the old release and I have the second day no connection to the inverter around 11am CET. 

After a reboot the inverter is found immediately.

I'll check if this is the same tomorrow. 

PS: I use a ESP32-D0WDQ6 on a nRF24 with a HM-300.

stefan123t, 2024-10-30:
@jrnas is this problem still ocurring with the current v0.8.152 maybe we can close the issue in case it has been fixed ?

jrnas, 2024-10-30:
nope you can close that issue, thanks

closed 2024-10-30 by jrnas