lumapu/ahoy #1439: [Bug] ePaper Display shows Wifi not connected if using a static IP.

derFrickler, 2024-02-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

ahoy_v0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Wemos D32 with ePaper Display

### Error description

If using a static IP configuration on the DTU, the ePaper Display shows Wifi not connected instead of the configured IP, it works fine when using DHCP.

lumapu, 2024-02-19:
Thank you for this error report

stefan123t, 2024-10-30:
@derFrickler can you test this with the current v0.8.152 from development03 branch ?
According to lumapu this should be fixed with that version.

derFrickler, 2024-10-30:
Hi, i had to update Ahoy last week (as i broke the display config and ended up in a reboot loop of the ESP32) and i can confirm that its working now, IP is displayed.

Thanks a lot.