lumapu/ahoy #1442: Schreibfehler in der deutschen Übersetzung

QuickStopGroceries, 2024-02-20:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.8.3

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

..

### Error description

Bitte dem Wort "deaktiviert" noch das zweite i spendieren. :)
[image]

Gubi2023, 2024-02-20:
Webserial auf deutsch: "Days"
[image]
 mit Absicht nicht übersetzt?

lumapu, 2024-02-22:
wir übersetzen alles @Gubi2023 😅. Danke für die Hinweise

Gubi2023, 2024-02-23:
8.85: gefixt "Tage"

Gubi2023, 2024-02-23:
> wir übersetzen alles @Gubi2023 😅. Danke für die Hinweise

[image]
 Lässt sich das auch übersetzen, oder ist das ne andere Baustelle??

MetaChuh, 2024-02-23:
@Gubi2023 

richtige baustelle.
ist derzeit hardgecoded in `web/html/serial.html`
hat wahrscheinlich niemand bemerkt, weil man's erst nach action button klicken sieht.

```js
 80                   try {
 81                       return document.execCommand("copy");  // Security exception may be thrown by some browsers.
 82                   } catch (ex) {
 83                       alert("Copy to clipboard failed" + ex);
 84                   } finally {
 85                       document.body.removeChild(ta);
 86                       alert("Copied to clipboard");
 87                   }
```

Gubi2023, 2024-02-24:
[image]
danke!

Gubi2023, 2024-02-29:
ausserdem ist der Text auch in der englischen Version auf deutsch  
[image]

MetaChuh, 2024-02-29:
hi @Gubi2023 

danke für's video, das erleichtert vieles 👍

dies kommt nicht von ahoydtu.

`Weitere Aufforderungen von [URL] verbieten` ist keine funktion in ahoydtu, sondern entweder eine browserfunktion eines browsers den ich nicht kenne, oder ein plugin (js oder werbeblocker plugin) in einem browser.
diese generieren den text selbst, und wählen die localization sprache anhand deiner browser spracheinstellung.

standard ansicht in chromium/chrome, egal wie oft man auf kopieren klickt:

[image]

standard ansicht in safari, egal wie oft man auf kopieren klickt:

[image]

probier mal einen anderen browser den du hast, der keine plugins installiert hat, und klicke dann öfters auf kopieren.

MetaChuh, 2024-02-29:
@QuickStopGroceries 

dein anliegen wurde bereits in v0.8.85 mit dem fehlenden i in `Wechselrichterabfrage deaktiviert (Nacht)` gefixed.
darum bitte dieses issue selbständig schließen.
danke dir.

closed 2024-03-01 by QuickStopGroceries