lumapu/ahoy #1443: Anordnung der Solarmodule in der Web-Anwendung

Fan73, 2024-02-20:
Es wäre schön wenn man die Anordnung der Fenster für die Solarmodule ändern könnte.
Bei mir z.B. wird durch den Aufbau und Verkabelung zum Inverter das linke Solarmodul auf der rechten Seite angezeigt.
Ändern könnte man das nur durch umstecken am Inverter. Dafür sind aber die Kabel zu kurz.

Vielen Dank 
Jürgen

kiu77, 2024-02-20:
Eindeutige Bezeichnungen vergeben reicht nicht
?

Fan73, 2024-02-20:
Nunja, von meinem Arbeitsplatz zum Beispiel schaue ich in Richtung Module, die sind nebeneinander, also eins links und das andere rechts. So habe ich sie auch bezeichnet, Modul-L und Modul-R.
Auf dem Monitor in der Webanwendung befindet sich nun Modul-R links und das andere eben rechts.
Zugegeben, ein kleiner Schönheitsfehler, aber wäre doch schön, wenn man die Anordnung auf irgendeine Art und Weise wählen könnte...

Gruß Jürgen

MetaChuh, 2024-02-21:
lässt sich dein monitor um 180 grad auf den kopf drehen ?
(so ein monitor den man beliebig hochkant oder querformat wie ein lenkrad drehen kann)

Fan73, 2024-02-21:
Für diese Funktion braucht man nicht den Monitor drehen, das erledigt Windows per Tasteneingabe!
Was meinst Du, passiert dann mir dem Rest der Webseite, mit der Schrift usw.?

lumapu, 2024-02-21:
das Thema hatten wir doch erst kürzlich, bei Gelegenheit suche ich den issue raus

Joh49, 2024-02-21:
Es ist doch eigentlich eine Frage wie die Module an den Wechselrichter angeschlossen werden. Bei mehreren Wechselrichtern ist der zuerst einzugeben, welcher zuerst erscheinen soll, bei den Modulen kommt das zuerst welches an den 1. Port angeschlossen ist.

MetaChuh, 2024-02-21:
> Für diese Funktion braucht man nicht den Monitor drehen, das erledigt Windows per Tasteneingabe!

ah, sehr cool, danke.
kannte ich noch nicht 👍

> Was meinst Du, passiert dann mir dem Rest der Webseite, mit der Schrift usw.?

die ist dann glaub‘ ich auch auf dem kopf gedreht.
links und rechts der panels sollten dann aber wahrscheinlich zueinander passen.

evtl geht auch den kopf nach unten drehen (kopfstand oder yoga)
dann müssten alle parameter (links rechts und lesbar) wieder zusammenpassen.

technics42, 2024-02-21:
Now, if you are standing on your balcony left/right modules are shown in wrong order? Just take a look from your garden to your modules, then order is correct, but naming is wrong 😉

Just don't call them left/right, but e.g. star-/larboard - it will always depend on your point of view.

kiu77, 2024-02-21:
Vorschlag: Benenne die Panels anders.
Statt "Panel links" und "Panel rechts"
z.B.: "Panel Ost" und "Panel West"

Dann hast Du nicht mehr den scheinbaren Widerspruch zwischen Panelposition und Anzeige.

AUSSERDEM: Wenn Du die Webseite auf einem Smartphone anschaust, ist "links" und "rechts" völlig nichtintuitiv und eigentlich meistens falsch ;-)

Fan73, 2024-02-21:
ja, das mag ja alles mehr oder weniger stimmen aber,
Bei mir ist das linke Paneel östlich und das rechte Paneel westlich und das linke Paneel wird auf der Webanzeige rechts, oder halt westlich dargestellt und das linke auf der anderen Seite... usw...
Zugegeben, für die wenigsten von uns relevant, ein kleiner Schönheitsfehler, sollte leicht zu beheben sein falls mal nichts anderes zu tun ist, ich kann aber auch mit dieser Sache leben....
Vielleicht klappt es irgendwann mal mit dieser Funktion bzw. Möglichkeit...

Grüß Jürgen

MetaChuh, 2024-02-21:
ich habe eine idee:

was wäre, wenn du den laptop um 180 grad drehst, und das fenster hinter dir ist ?
wären dann links und rechts der settings identisch mit der position wo du sitzt ?
falls du dann die panels gleichzeitig sehen willst, könnte man theoretisch einen spiegel neben dem monitor aufstellen, was die problematik reversiert.

Fan73, 2024-02-21:
Du bist so unendlich klug und allwissend.... 
Hut ab, 
das ist die Lösung....
nur,  
....leider habe ich keinen laptop....

Fan73, 2024-02-21:
und was meinst du nun erreicht zu haben?

Fan73, 2024-02-21:
dafür werden sich sicher einige Nutzer der Software es sich 3 mal überlegen sich mit Idenn und Wünschen hier einzubringen...
gute Nacht....

kiu77, 2024-02-21:
Sorry, Fan73,
es ist ganz sicher nicht adäquat, auf beleidigte Leberwurst zu machen, weil Dir jemand erklärt, dass das, was für Dich ein Problem ist, für andere nebensächlich bis zu vernachlässigen ist.
Es wurde mehrfach zwischen ernst und belustigt versucht, Dir zu zeigen, dass es Wichtigeres gibt, und es hätte durchaus eine Gelegenheit gegeben, das einzusehen. Muss aber nicht.
Wenn es eine Lehre daraus gibt, dann sicher so etwas wie, dass es auch Nebensächlichkeiten gibt, von deren Wünschen sich Entwickler abgrenzen - und das berechtigt, sogar freundlich und lustig.
Zu Ideen und Wünschen: Wünsche sollten halt eine gewisse Relevanz haben. Klar sehen alle das "Problem", aber es ist so winzig, dass man eben den Begriff in Anführungszeichen schreiben muss. Es ist auch kein großes Problem, solche "Probleme" zu haben oder zu äußern, aber man sollte besser damit leben können, dass die Welt da eine andere Vorstellung hat.

technics42, 2024-02-22:
Maybe no need for buying extra cables, maybe you could also turn your inverter so you could switch existing cables...

closed 2024-03-10 by lumapu