lumapu/ahoy #1444: Bei API Leistungssteuerung erhalte ich seit 0.8.80 Werte über 100% in der Anzeige

jimknopf63, 2024-02-20:
## Hardware
  * [ ] ESP8266
  * [ X] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ X] external antenna (SMA)

### Power Stabilization:
* [ X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.80
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich arbeite mit der Nulleinspeisung von reserve85  (V1.66). Ab der Version 0.8.80 zeigt mir Ahoy teilweise utopische Leistungswerte in der Übersicht (u.a. zwischen 100-180% entsprechende Wattzahl) Im Log von der Nulleinspeisung von reverse85
stehen aber die realistischen Leistungswerte die übertrage werden. Ist das jetzt ein Ahoy Problem oder eins der Nulleinspeisung?

tester277, 2024-02-20:
Hallo jimknopf,
das ist ein Konfigurationsproblem Deiner Nulleinspeisung.
Vergergleiche mal die Reihenfolge und die Max Werte (HOY_MAX_WATT =  )in Deiner Konfiguration mit der Reihenfolge in Ahoy.

reserve85, 2024-02-21:
So ist es, @jimknopf63 du musst sowohl AHOY als auch das Script aktualisieren, dann gehts wieeder. 
Es sind die letzten Wochen mehrere API-Changes aufgekommen (u.A. wurde der Faktor entfernt was bei dir das Problem ist).

closed 2024-02-21 by jimknopf63

MetaChuh, 2024-02-21:
@jimknopf63 

freut uns, dass du's mit den gegebenen informationen deine anfragen lösen konntest 👍
und danke für's selbständig, unaufgeforderte schließen der issue 🙏👍

(für semi off topic leser: warum das danke ?:
manche user informieren in privaten open source projekten leider nicht, wenn ihre probleme gelöst sind, da sie für sie selbst gelöst sind, und deshalb nicht mehr privat wichtig, was massiv offene community issues in kleineren projekten hinterlässt, obwohl die grundfragen und funktionalität des original issues abgedeckt wurden)

sharing is caring as you did, it's much appreciated at all open source communities 🙏❤️

hugs and greetings,
MetaChuh