lumapu/ahoy #1445: mangelde Browser kompatibiltät [Bug]

robt24, 2024-02-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

lumapu

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

only network

### Error description

Ein großes Problem für erstmalige Nutzer ist die mangelnde Kompatibiltät mit Browsern. ich habe  zuerst Chrome 81 verwendet und habe nicht gefunden, wo ich die WR-ID eintragen kann. Dann mit chrome 85 -  dasselbe. 
ahoy sollte den User-Agent-String auswerten und eine Warnung ausgeben, daß mit dieser Version das GUI unvollständig erscheint! Besser noch:  im changelog jeweils die minimale Browserversion angeben, die erforderlich ist!

robt24, 2024-02-21:
120 wird für Win7 gar nicht angeboten. Außedem fressen die neuesten Versionen viel zu viel RAM
 

Gesendet: Mittwoch, 21. Februar 2024 um 15:42 Uhr
Von: "Ollipop030 - ***@***.***" ***@***.***>
An: "lumapu/ahoy" ***@***.***>
Cc: "robt24" ***@***.***>, "Author" ***@***.***>
Betreff: Re: [lumapu/ahoy] mangelnde Browser kompatibiltät [Bug] (Issue #1445) 

 

Du verwendest einen 4 Jahre alten Browser und wunderst dich, das irgendwelche Seiten nicht richtig angezeigt werden können? Bitte mal updaten (aktuell sollte 120 irgendwas sein) und nochmal probieren.

—Message ID: ***@***.***>

robt24, 2024-02-21:
> Der Support für Windows 7 endete am 14. Januar 2020!
> Ich hoffe doch das du mit diesem alten System nicht mehr im Internet unterwegs bist.

Du kommst irgendwie komplett vom Thema weg!

 
 

Gesendet: Mittwoch, 21. Februar 2024 um 17:35 Uhr

rmayergfx, 2024-02-21:
Falls es dir hilft, hole dir Firefox 115ESR in der 32bit Variante, damit geht es.
Ich komme nicht vom Thema weg, das ist nur eine Info das solch alte Systeme nicht mehr ins Internet sollten.
Die meisten alten WIN7 Geräte können zudem ohne Probleme auf WIN10 aktualisiert werden, da läuft der Support wenigstens noch bis zum 14. Oktober 2025.

robt24, 2024-02-21:
Doch Du kommst vom Thema weg ahoy-dtu gibt keine Meldung aus, wenn man mit einem Browser zugreift, auf dem nicht alles dargestellt wird. Das sehe ich als Bug.

Warum man für so eine einfache ANwendung überhaupt mechanismen benutzen muß, die nur noch neuere Browser können, ist mir unverständlich

 
 

Gesendet: Mittwoch, 21. Februar 2024 um 17:49 Uhr

HorstyS, 2024-02-21:
Uhu.... Ich hol schon mal Popcorn.

robt24, 2024-02-21:
und ich wechsle zu opendtu

HorstyS, 2024-02-21:
> und ich wechsle zu opendtu

Dann mach vorher noch diesen threat zu. Braucht's ja dann nicht mehr.

Peertje-62, 2024-02-21:
Deutsche Bahn: Administrator für  Windows 3.11 gesucht
https://www.t-online.de/digital/aktuelles/id_100331850/deutsche-bahn-administrator-fuer-windows-3-11-gesucht.html

Peertje-62, 2024-02-21:
Sorry

robt24, 2024-02-21:
ich  habe diesen Thread nicht für mich persönlich aufgemacht, auch wenn Du Dir das nicht vorstellen kannst ! Der Bug ist ja weiterhin da

hrolofs, 2024-02-21:
Hi,

last doch im support forum darüber diskutieren. Das passt dort besser hin als hier.
https://discord.com/channels/984173303147155506/1029761098381017098

Poste dort doch auchmal ein Bild wie die Settings Einstellungen bei dir aussehen. Dann fällt es leichter zu schaun obs wirklich am Browser liegt oder nicht.

HorstyS, 2024-02-21:
> ich habe diesen Thread nicht für mich persönlich aufgemacht, auch wenn Du Dir das nicht vorstellen kannst ! Der Bug ist ja weiterhin da

Doch, das hast du.

dknubben, 2024-02-21:
Ich muss mich hier mal "einklinken". Danke an @robt24 für dieses Issue.
Ich habe Windows 95 mit IE 2.0 und auch Netscape 4 produktiv im Einsatz. Leider läuft Ahoy mit beiden Browsern überhaupt nicht gut! Das ist ein Bug, der dringend gefixt werden muss!

hrolofs, 2024-02-21:
> Ich habe Windows 95 mit IE 2.0 und auch Netscape 4 produktiv im Einsatz

Ich bin raus aus der Diskussion. Das war vor meiner aktiven IT-ler Zeit.

Peertje-62, 2024-02-21:
Robt24,
Sorry

lumapu, 2024-02-21:
Ich weiß echt nicht mehr weiter - bei einem Fernseher verstehe ich das Problem ja noch aber bei einem PC mit Windows 7 werde ich ganz bestimmt nicht irgendwelche Meldungen einbauen.
Bin gespannt was das Team OpenDTU davon hält 😃.

Kein Bug, kein Feature Request, das bringt uns nicht weiter ...

Peertje-62, 2024-02-21:
Hi,
FYI
Tried it with my neighbour on Amazon fire tv with Amazon Silk browser:
works fine
regards

MetaChuh, 2024-02-22:
@robt24 

> und ich wechsle zu opendtu

neeeeeeeeiiiiiin bitte nicht ... dann müssten ja alle das issue, das du dort aufmachst, nochmals beantworten 😂

HorstyS, 2024-02-22:
> vielleicht legen die dort mehr Wert darauf, daß Leute ihre Software nutzen, als hier, wo die Leute offenbar einiges geraucht haben.

ich glaube du verwechselst hier etwas. aber egal. viel glück.
machst du nun deinen post zu? das problem hat sich ja dann für dich hier erledigt.

Ollipop030, 2024-02-22:
> vielleicht legen die dort mehr Wert darauf, daß Leute ihre Software nutzen, als hier, wo die Leute offenbar einiges geraucht haben.

Ich weiß ehrlich gesagt nicht, was du hier erwartest. Dein Betriebssystem bekommt keinen Support mehr, dein Browser in der alten Version offensichtlich auch nicht, warum sollten die Entwickler von Ahoy dir Support geben?
Auf Ahoy läuft nun mal ein Webserver (wenn auch sehr minimalistisch). Dieser Webbrowser ist auf dem aktuellen Stand der Technik, diesen jetzt so hinzubiegen nur damit ein alter Browser damit funktioniert ist erstmal den Aufwand nicht Wert, und vielleicht werden dabei sogar Sicherheitslücken wieder geöffnet, die längt gefixt wurden.
Es wird wahrscheinlich noch tausende Webseiten geben, die bei dir nicht richtig dargestellt werden können, dort meldest du dich doch auch nicht und erwartest, das die ihre Seiten für dich ändern.

Du hast einen Bug gemeldet, ein Bugfix wäre ein Update deines Browsers. Jeder Browser bekommt monatliche Updates, und das aus gutem Grund: Die Technik entwickelt sich nunmal weiter. 

@lumapu: Ich kenne das noch vom Internet Explorer (habe ich sehr lange nach Supportende noch benutzt): Auf einigen Seiten wurde der Zugriff einfach komplett gesperrt, mit dem Hinweis "Ihr Browser wird nicht mehr unterstützt". Vielleicht kann man sowas evtl. einbauen.

rmayergfx, 2024-02-22:
@Ollipop030 @lumapu 
Das mit dem Browserhinweis finde ich gut, allerdings nicht per Abfrage sondern einfach als Hinweis. Eine Abfrage bedeutet wieder CPU Load und eine mögliche Fehlerquelle.

Stell mir das als einfachen Hinweistext vor, so wie auch auf einigen Webseites zu finden ist. Denn leider verhält sich auch jeder Browser anders in der Anzeige der Inhalte, gibt nicht ohne Grund auf einigen Systemen die Abfrage nach dem User Agent um eine Browserweiche zu aktivieren, das kann man auf performanten Systemen machen aber bitte nicht auf den kleinen Entwicklerboards.

"Um alle Funktionen nutzen zu können empfehlen wir Chrome ab Verssion xxx oder einen anderen auf Chrome basierenden Webbroswer"

rmayergfx, 2024-02-22:
> vielleicht legen die dort mehr Wert darauf, daß Leute ihre Software nutzen, als hier, wo die Leute offenbar einiges geraucht haben.

Bitte sachlich bleiben und Beleidungungen unterlassen.

dknubben, 2024-02-22:
> @lumapu: Ich kenne das noch vom Internet Explorer (habe ich sehr lange nach Supportende noch benutzt): Auf einigen Seiten wurde der Zugriff einfach komplett gesperrt, mit dem Hinweis "Ihr Browser wird nicht mehr unterstützt". Vielleicht kann man sowas evtl. einbauen.

Das wäre natürlich eine gute Idee und wird vielfach in Webprojekten umgesetzt. Allerdings nicht auf IoT devices wie dem ESP. Es ist schlicht unmöglich, alle (alten) User Agents abzufangen. Das ist eine riesige Liste, die nach bestimmten Kriterien abgefragt werden müsste. Wo könnte diese überhaupt hinterlegt werden?

robt24, 2024-02-22:
OpenDTU läuft problemlos mit  Chrome 81, und warum auch nicht, Chrome 81 kann ja alles, was man braucht. Ist auch aufgeräumer und übersichtlicher.

dknubben, 2024-02-22:
> OpenDTU läuft problemlos mit Chrome 81, und warum auch nicht, Chrome 81 kann ja alles, was man braucht. Ist auch aufgeräumer und übersichtlicher.

Das ist ja schön! Willkommen in der modernen IT-Welt mit aktueller Software. Dann mach nun bitte dieses Issue zu und bitte behandle das OpenDTU-Team mit mehr Respekt.

lumapu, 2024-02-22:
> OpenDTU läuft problemlos mit  Chrome 81, und warum auch nicht, Chrome 81 kann ja alles, was man braucht. Ist auch aufgeräumer und übersichtlicher.

Ja dann haben wir doch eine Lösung! Schade dass noch eine Beleidigung kam.
Verstehe nicht was dann Ahoy noch ausrichten kann, wenn die Entscheidung schon gefallen ist. Falls du Werbung machen willst, gerne aber bleib sachlich. Jeder hier kennt beide Systeme und jedem steht frei seine bevorzugte Software einzusetzen oder auch beide.

Hier in GitHub gibt es auch ein Feature, heißt Pull request. Das wäre eine Möglichkeit anderen Gleichgesinnten die Möglichkeit zu geben, die Verbesserungen deinerseits mit allen zu teilen, mit großer Wahrscheinlichkeit geht es sogar früher oder später in den Hauptentwicklungszweig über.

closed 2024-02-22 by lumapu