lumapu/ahoy #1446: PowerLimit kann nicht gesetzt werden wenn Passwort gesetzt ist

scheisserchen65, 2024-02-22:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

ESP Tools (flash)

### Setup

MQTT

### Debug Serial Log output

```shell
INFO     Inverterlimit 600 Watt was previously not accepted by at least one inverter, trying again...                                               
INFO     setting new limit to 600 Watt                                                                                                              
INFO     Ahoy: Inverter "HM-600": setting new limit from 354 Watt to 354 Watt                                                                         
INFO     Ahoy: Inverter "HM-600": Limit timeout!                                                                                                    
INFO     Ahoy: Inverter "HM-300": setting new limit from 245 Watt to 245 Watt                                                                       
INFO     Ahoy: Inverter "HM-300": Limit timeout!                                                                                                    
INFO     Ahoy: Inverter "HM-600" reachable: True                                                                                                    
INFO     Ahoy: Inverter "HM-300" reachable: True
```

### Error description

1) wenn ich ein Passwort in der AHOY-DTU gesetzt habe wird das PowerLimit nicht übertragen, es erscheint im Logfile (Linux, (var(log/sys) "Limit timeout!"
2) nachdem ich das Passwort aus der AHOY-DTU testweise entfernt hatte funktionierte alles
3) Erneuter Versuch: Passwort gesetzt: Fehler, Passwort raus, alles funktioninert

Ollipop030, 2024-02-22:
Es gab Änderungen an der API, du musst das Script für die Nulleinspeisung updaten.

MetaChuh, 2024-02-22:
hallo @scheisserchen65 😂

hier ein script beispiel für curl:

```bash
# syntax:
# curl --header "Content-Type:application/json" --request POST --data '{"auth": "<dtu_password>", "cmd": "limit_nonpersistent_relative", "id": <inverter_nummer>, "val": <prozent>}' "http://<dtu_ip>/api/ctrl"

# beispiel:
curl --header "Content-Type:application/json" --request POST --data '{"auth": "passiwortilein", "cmd": "limit_nonpersistent_relative", "id": 0, "val": 50}' "http://192.168.0.4/api/ctrl"
```

falls du's übersetzt in php, python, etc brauchst, sag' bescheid

greetings
(ich geh jetzt aufs klo und spiel nintendo 😂)

scheisserchen65, 2024-02-22:
Besten Dank, hat nach dem Update funktioniert :)

MetaChuh, 2024-02-22:
perfekt👍

wollte dir grad' den update link für reserve85 zeroexport v1.79 schicken [`download link`] aber du warst schneller 👍

issue bitte schließen nicht vergessen 🙏

closed 2024-02-22 by scheisserchen65