lumapu/ahoy #1447: Radio statistics for Inverter [Bug]

technics42, 2024-02-22:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.84

### Github Hash

f500999

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter intervall set to 15 sec.

### Error description

Radio statistics for inverter are showing something like this:
Inverter loss rate	lost 65526 of 11 	595690.91 %
DTU loss rate	lost 30 of 61 	49.18 %

Was is the meaning of those values?
The first one surely is wrong, regardless of the meaning.

technics42, 2024-02-22:
Ok, this will be the reason.
I always update new version to a second HW and keep it running some time in parallel to the prior version.

So, first value is loss-rate reported by inverter and second one is loss-rate reported by DTU itself.

rejoe2, 2024-02-22:
> So, first value is loss-rate reported by inverter and second one is loss-rate reported by DTU itself.

Not exactly. The inverter sends two values, (received and send), which then are compared with what AhoyDTU has counted as send and received... So it's always both sides compared in the loss rates.

technics42, 2024-02-22:
So how exactly are "inverter loss-rate" and "DTU loss-rate" calculated? If the first one is using values from inverter and DTU this will also be wrong on any restart of DTU, or?

rejoe2, 2024-02-22:
For details see parseGetLossRate() in Communication.h. Ahoy requests last values from inverter first and uses that as base. So no real problem with restarts...

technics42, 2024-02-22:
Ah, all those losses are calculated by using inverter values and not only getting the values from only the named system.

So, it's always the "receiver's" loss-rate:
inv. loss-rate = not rec. msg by inv. / sent msg. by DTU

DTU loss-rate = not rec. msg by DTU / sent msg. by inv.

rejoe2, 2024-02-22:
You are welcome to contribute to the docu...

closed 2024-08-16 by lumapu