lumapu/ahoy #1448: [Bug]Webserial: Text ändert sich von deutsch auf englisch

Gubi2023, 2024-02-23:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

8.85

### Github Hash

b6afec1

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

deutsch intalliert

### Error description

kleiner Schönheitsfehler: wenn ich im Weblog auf  automatisch  scrollen klicke, ändert sich der text ins englische. Siehe Video

https://github.com/lumapu/ahoy/assets/142897856/f9e2f0da-90c8-48c7-abb9-90e62f54a126

knickohr, 2024-02-23:
Das ist kein Fehler von Ahoy, sonder allein dem Cache von Deinem Browser geschuldet. Leere ihn mal und der Spuk hat ein Ende.

Gubi2023, 2024-02-23:
hab ich gemacht. Fehler bleibt!!

Gubi2023, 2024-02-24:
ok! danke für das schnelle Bugfix!

closed 2024-02-24 by Gubi2023