lumapu/ahoy #1449: [Bug] POST to /api/ctrl leads to reboot of ahoy-dtu

Kernelhacker, 2024-02-23:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

GIT SHA: 5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

2 inverters (0: hm-800, 1: hm-1500)

Werte und Gesamtertrag um Mitternacht zurücksetzen ('Kommunikation während der Nacht pausieren' muss gesetzt sein) [X]
Werte bei Sonnenuntergang zurücksetzen [ ]
Werte zurücksetzen, sobald der Wechselrichter nicht erreichbar ist [ ]
Maximalwerte mitternachts zurücksetzen [X]
Kommunikation starten ohne gültige Zeit (sinnvoll im AP Modus) [ ]
Grid-Profil auslesen [X]
Ertragseffizienz (Standard 1.0) 0,959999979

No MQTT, NRF24 enabled and working fine.

### Debug Serial Log output

```shell
I: barn-plus/ch0/Efficiency: 95.004 %
I: barn-plus/ch0/MaxPower: 110.300 W
I: 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008aa0f  PS      : 0x00060230  A0      : 0x800ea484  A1      : 0x3ffdcb20  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x3ffdfcdc  
A6      : 0x3ffdfbdc  A7      : 0x3ffdccac  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x0000006c  A12     : 0x3f402240  A13     : 0x00000001  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008a9d1  LEND    : 0x4008a9e1  LCOUNT  : 0xfffffffd  

Backtrace: 0x4008aa0c:0x3ffdcb20 0x400ea481:0x3ffdcb40 0x400ed7eb:0x3ffdcc10 0x40197cd2:0x3ffdcce0 0x400fd2ad:0x3ffdcd00 0x400fb3ba:0x3ffdcd50 0x400fb455:0x3ffdcda0 0x400f3541:0x3ffdcdc0 0x400f35d1:0x3ffdcdf0 0x400f3db6:0x3ffdce10

ELF file SHA256: 93f179b8c34d791c

E (5051) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57
```

### Error description

Trying to POST a REST API command (e.g. power on inverter) leads to reboot of ahoy-dtu:

curl --location 'http://192.168.3.195/api/ctrl' \
--header 'Content-Type: application/json' \
--data '[
    {
        "id": "0",
        "cmd": "power",
        "val": 1
    }
]'

rejoe2, 2024-02-23:
Your log doesn't show any time info in the "I" lines. 

Has the ESP a time set when this happens or not?
(For more background see https://github.com/lumapu/ahoy/issues/1434#issuecomment-1960019949 and other linked issue there).

MetaChuh, 2024-02-23:
@Kernelhacker 

hast du's schon mit dem neuen auth probiert, falls du ein passwort gesetzt hast ?
```bash
curl --header "Content-Type:application/json" --request POST --data '{"auth": "dein_passwort", "cmd": "power", "id": 0, "val": 1}' "http://192.168.3.195/api/ctrl"
```

(leichter lesbar in multi line):
`curl --header "Content-Type:application/json" --request POST --data '{"auth": "dein_passwort", "cmd": "power", "id": 0, "val": 1}' "http://192.168.3.195/api/ctrl"`

ps for nerds: --location braucht man bei ahoydtu nicht, der curl parameter dient nur, damit 302 redirects mancher webseiten verfolgt werden

Kernelhacker, 2024-02-23:
Die Ursache ist gefunden: Der Body war falsch: Das JSON Objects war in einem Array gekapselt. Ohne das genau debugged zu haben, gehe ich davon aus, dass hier ein wenig Fehlerbehandlung gut wäre, um solche Fehler in Zukunft abzufangen:

In RestApi.h in "onApiPostBody": 
```
DeserializationError err = deserializeJson(json, reinterpret_cast<const char*>(mTmpBuf), mTmpSize);
JsonObject obj = json.as<JsonObject>();
```
An dieser Stelle sollte erst geprüft werden, ob json wirklich ein JsonObject ist und falls nicht sollte eine entsprechende Fehlermeldung ausgegeben werden. 
Wenn ich das Projekt endlich übersetzt bekomme, kann ich ja einen Patch bauen.

MetaChuh, 2024-02-23:
@Kernelhacker 

sehr cool, dass es wieder funktioniert, und danke dir für die zusatzinfo 👍

> An dieser Stelle sollte erst geprüft werden, ob json wirklich ein JsonObject ist und falls nicht sollte eine entsprechende Fehlermeldung ausgegeben werden.

leider ist der sketch/flash von nutzbar ca 1.3mb auf esp32 schon bis auf wenige bytes ausgeschöpft, und wir kämpfen gerade, wie man alles auf einem 2mb total sketch unterbringen kann.

derzeit ist der dev task, den bestehenden code zu optimieren, damit man nicht die standard esp32 droppen muss, weil die neuen geplanten funktionen nicht mehr rein passen, und nicht allen sagen muss, sie sollen eine neue esp hardware mit mehr speicher kaufen.

ps: vielleicht kann es @lumapu in weniger worten besser erklären

thx & greetings,
metachuh

Kernelhacker, 2024-02-25:
> @Kernelhacker
> 
> sehr cool, dass es wieder funktioniert, und danke dir für die zusatzinfo 👍
[..]
> leider ist der sketch/flash von nutzbar ca 1.3mb auf esp32 schon bis auf wenige bytes ausgeschöpft, und wir kämpfen gerade, wie man alles auf einem 2mb total sketch unterbringen kann.

Ja, das ist alles sehr knapp - ich habe im Moment zu wenig Zeit um wirklich viel beizusteuern und mein Pull-Request für diesen Fehler ist mit @lumapu's Fix auch schon wieder überholt ;)
Habe früher mal ein paar Projekte auf PIC16 Basis mit Assembler gemacht. Im Vergleich zu diesen Prozessoren ist der Platz auf den ESPs schon extremer Luxus!

lumapu, 2024-02-25:
ja Luxus stimmt, trotzdem sehr schnell voll 😉

closed 2024-08-16 by lumapu