lumapu/ahoy #1453: [Bug] ESP LED permanent on, but set to off

jolo1581, 2024-02-24:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8,83

### Github Hash

5ebfe5a

### Build & Flash Method

was already installed

### Setup

[image]

### Error description

Since new version it is not possible to power off the blue LED of ESP.
Also it seems to me, it is permanent on, if Ahoy is connected to inverter

MetaChuh, 2024-02-24:
@jolo1581 

the blue led is connected to d4 (gpio2)
you have ce on d4, instead of irq

the correct wiring should be:
```
cs  d8
ce  d3
irq d4
```

this way the blue led only flashes briefly if irq is triggered

please see the [`ahoydtu schematics`](https://ahoydtu.de/getting_started/):

[image]

[image]

jolo1581, 2024-02-24:
This should Not be the problem.
It worked fine in older versions

MetaChuh, 2024-02-24:
believe me,
just do it, or downgrade to a previous version that worked for you regarding the led.
it's all documented, all changes, and all why the changes ... your choice.

@lumapu imho this issue can be closed too

mroenne2022, 2024-02-24:
@jolo1581 
If you can , change the cable d3 and d4 and  in Setup too  - then it works fine with 0.8.83 and the blue led is off. I have already changed my dtu's last week,  now the problem is solved.

DB1BMN-collection, 2024-02-25:
Hi, just a stupid (?) Question: How do you change the D3 and D4?

mroenne2022, 2024-02-25:
@DB1BMN-collection 
Change the Cable CE and IRQ on NRF Modul and change it in the Setup Ahoy too and the problem is solved.
[image]

DB1BMN-collection, 2024-02-25:
Ah, I understand, you have no PCB 😅
OK, in my Case I will have tu cut PCB traces and do some crosses with bodge wires.
Will try it tomorrow. My Eyes are tired for tonight.
Thanks!

mroenne2022, 2024-02-25:
@DB1BMN-collection 
Yes,right. I have only cable between NRF and esp8266,  so it was easy to change.

DanielR92, 2024-06-29:
Problem solved, closed now. Please reopen when something is not working properly.

closed 2024-06-29 by DanielR92