lumapu/ahoy #1454: Neustart wird verlangt, obwohl keine Änderungen durchgeführt[Bug]

Gubi2023, 2024-02-25:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.86

### Github Hash

e7d9a8e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

deutsche Installation

### Error description

Guten Morgen! heute morgen fordert mich die Index-Seite auf, die DTU neu zu starten, obwohl ich wissentlich keine Änderungen vorgenommen habe! Kann das sein?
[image]

Dirk-Ahoy, 2024-02-25:
irgendwas musst du geändert haben die Info kommt nicht von selbst.

Gubi2023, 2024-02-25:
Korrektur: ich habe heute nacht die  .json exportiert. Gilt das schon als "Änderung durchgeführt"?? (Ich ändere doch damit nichts...)

knickohr, 2024-02-25:
Wäre ich mir gar nicht so sicher !

Ich habe auch die Meldung, kann aber nicht mit Sicherheit aussagen ob ich nicht doch etwas geändert habe bei den vielen Tests 🤔

Also mal abwarten.

hrolofs, 2024-02-25:
Glaube das ist mit der 0.8.71 reingekommen.

## 0.8.71 - 2024-02-03
* fix heuristics reset
* fix CMT missing frames problem
* removed inverter gap setting
* removed add to total (MqTT) inverter setting
* fixed sending commands to inverters which are soft turned off
* save settings before they are exported #1395

Vermute das hängt mit dem #1395 zusammen den ich erstellt hatte.

Dirk-Ahoy, 2024-02-25:
Okay - dann Feature :) - hab ich gerade hier nachgestellt.

knickohr, 2024-02-25:
Ahhh jaaa, das könnte gut sein mit #1395 

habe gestern mehrfach Exportiert, wenn auch nur anzeigen lassen.

mroenne2022, 2024-02-25:
@Gubi2023 
Die json exportieren gilt offensichtlich tatsächlich als Änderung, hatte ich bei meinen DTU's auch.

lumapu, 2024-02-25:
ja ihr hab es gefunden 😊. Werde ich erst mal nicht ändern, wollte an den Einstellungen sowie einiges umbauen, sodasss nur noch selten ein Neustart erforderlich ist, bis dahin bleibt es so wie es ist.

Gubi2023, 2024-02-25:
ok. soll ichs zumachen oder erst mal offen lassen?

lumapu, 2024-02-25:
lass ruhig offen

closed 2024-03-23 by Gubi2023