lumapu/ahoy #1456: [Bug] turn inverter on and off via mqtt

pixelcatcher, 2024-02-25:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.3.6

### Github Hash

6fb0535

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nothing

### Error description

please add support to turn inverter on and off via mqtt. as it seems this is not implemented and can only be done with the web interface as for now.

knickohr, 2024-02-25:
Already implementiert :

<TOPIC>/ctrl/restart/<INVERTER_ID>
<TOPIC>/ctrl/power/<INVERTER_ID> payload 0/1

pixelcatcher, 2024-02-25:
thanks! was unable to find it ;)

DanielR92, 2024-06-29:
Solved, thanks Knickohr.

closed 2024-06-29 by DanielR92