lumapu/ahoy #1457: MQTT Topics Change and Missing since 0.8.79

rmayergfx, 2024-02-26:
Aktuell bin ich immer noch auf der Version 0.8.79 weil in den Versionen 0.8.8x mir bei der Überprüfung mit dem MQTT Explorer Topics fehlen. So. z.B. im ch0 ALARM_MES_ID:
[image]

Zusätzlich sehe ich noch weitere Topics, die nicht mehr verfügbar sind:
[image]

Warum ist das so? Wurden diese durch andere Topics ersetzt? Finde dazu keine Info und auch beim erneuten Einlesen bzw. prüfen mit dem MQTT Explorer sehe ich nicht wirklich neue Topics.

knickohr, 2024-02-26:
https://github.com/lumapu/ahoy/blob/development03/manual/User_Manual.md

Aber auch das ist schon kalter Kaffee. An dieser Bsustelle wird gerade einiges getan und in Zukunft wohl komplett umstrukturiert.

Besser bist Du wohl hier aufgehoben :
https://github.com/lumapu/ahoy/issues/1212

Vor allem mit der Excel-Liste in dem Issue.

rmayergfx, 2024-02-26:
Danke für den Hinweis. Das hilft schon einmal weiter. Leider versteckt sich das halt mal wieder in einem anderen Issue und daher findet man es nicht so leicht.

lumapu, 2024-02-28:
brauchst du die `ALARM_MES_ID`. Ich würde die gerne in Zukunft komplett aus MqTT streichen

rmayergfx, 2024-02-29:
@lumapu 
ich denke ich benötige die nicht wirklich, frage mich nur gerade warum diese ursprünglich mit aufgenommen wurde, welche Werte sich dahinter verbergen und was nun als Alternative kommt oder einfach ersatzlos gestrichen wird. Muss dann ja dementsprechend die Übersicht im HA umbauen damit das optisch nicht mehr stört. Möchte ungerne AhoyDTU in HA komplett neu einbinden, habe die befürchtung das dann die Historie durcheinanderkommt.

stefan123t, 2024-10-30:
@rmayergfx die `ALARM_MES_ID` ist die letzte Event ID oder der Alarm Code aus dem Event Log des Inverters.
Ich weiß jetzt auch nicht mehr welches von beiden, also entweder the "Event Counter" oder der letzte "Alarm Code" (0-9999).

Mein Lieblingscode ist nach wie vor 9999 inverter probably stolen, ich frage mich immer noch welcher Inverter diesen Wert zurückliefert, wenn er tatsächlich gestohlen wurde ist er i.d.R. auch nicht mehr erreichbar :rofl: