lumapu/ahoy #1458: MQTT radio_stst

zechnerhubert, 2024-02-26:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: D1 Mini Node Mcu mit ESP 8266-12F WLAN
Retailer URL: az-delivery.com

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode (NRF24L01+PA+LNA https://www.amazon.de/dp/B07SN49PGT/ref=pe_27091401_487027711_TE_SCE_dp_1
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [X] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.83_
Github Hash: 5ebfe5a

### Build & Flash Method:
* [X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I used Ahoy with iobroker MQTT adapter.

Since the last Ahoy update in iobroker I always get the error, "Client [AHOY-DTU-16117F] Cannot parse "inverter/Privat/radio_stat": {"tx":11439,"success":8038,"fail":292,"no_answer":3109,"retransmits":3529,"lossIvRx":58495,"lossIvTx":1008,"lossDtuRx":14831,"lossDtuTx":17".
If I delete the data point "inverter/Privat/radio_stat", I get the error message "State "mqtt.1.inverter.Privat.radio_stat" has no existing object, this might lead to an error in future versions".

Is the transfer from Ahoy to MQTT not correct?

knickohr, 2024-02-26:
This is a JSON?String. Did you parse this String correct ?

zechnerhubert, 2024-02-27:
Dieser String kommt ja vom ahoy!
Und es sieht aus als ob MQTT ein Problem damit hat.

zechnerhubert, 2024-02-27:
ich bekomme sonst auch minütlich Werte übertragen, heute hat History jedoch nur ein paar Daten aufgezeichnet:
27.02.2024  10:26:23 140,9W
27.02.2024  11:21:53 457,9W
27.02.2024  12:23:09 260,6W
27.02.2024  12:50:24 236,5W
27.02.2024  13:19:54 171,9W
27.02.2024  13:22:54 140,7W

Das ist doch auch nicht normal, oder?

knickohr, 2024-02-27:
Das sind aber keine Radiodaten, das sind Produktivdaten. Die sind noch nicht als JSON verfügbar. Das funktioniert definitiv sauber, ansonsten hätte ich schon längst einen Issue dazu aufgemacht. Bei mir kamen die Werte von 16 Invertern minütlich sauber rein.

zechnerhubert, 2024-02-27:
Aber was sagt den diese Fehlermeldung aus?

knickohr, 2024-02-27:
Vorausgesetzt das ist richtig raus kopiert, dann fehlt am Ende schon mal diese spitze Klammer :
{"tx":11439,"success":8038,"fail":292,"no_answer":3109,"retransmits":3529,"lossIvRx":58495,"lossIvTx":1008,"lossDtuRx":14831,"lossDtuTx":17"**}**.

Und ich denke genau da parset es nicht mehr richtig.

zechnerhubert, 2024-02-27:
Hier eine aktuelle Fehlermeldung von heute:
2024-02-27 11:21:53.853 - error: mqtt.1 (4049898) Client [AHOY-DTU-16117F] Cannot parse "inverter/Privat/radio_stat": {"tx":15678,"success":8046,"fail":293,"no_answer":7339,"retransmits":3611,"lossIvRx":58495,"lossIvTx":1008,"lossDtuRx":14831,"lossDtuTx":17

zechnerhubert, 2024-02-27:
liegt es am Ahoy?

knickohr, 2024-02-27:
Das kann ich erst sagen wenn ich diese Werte Morgen wieder bekomme. Kann schon möglich sein. Allerdings mein Broker, bzw. Node-Red meckert da nichts an, bekam heute durchgängig diese Werte angezeigt.

zechnerhubert, 2024-02-27:
Meiner meckert auch erst seit dem Update auf 0.8.83

knickohr, 2024-02-28:
Eben überprüft :

Der Topic ist korrekt und vollständig, kommt auch sauber im Broker an

[image]

zechnerhubert, 2024-02-28:
Also ich habe auch heute wieder solche Meldungen:
2024-02-28 16:48:33.068 - error: mqtt.1 (4049898) Client [AHOY-DTU-16117F] Cannot parse "inverter/Privat/radio_stat": {"tx":22744,"success":8070,"fail":304,"no_answer":14370,"retransmits":4805,"lossIvRx":58495,"lossIvTx":1008,"lossDtuRx":14831,"lossDtuTx":1
Habe auch heute wieder ganzen Tag nur 19 Werte aufgezeichnet.
Gibt es im Ahoy, die Möglichkeit ein log zu sehen, was gesendet wird? Wenn ja, muss ich mich am WE dazusetzen.

zechnerhubert, 2024-02-28:
Ich habe jetzt den MQTT Adapter einmal auf Debug gesetzt, um alle Nachrichten vom MQTT Adapter mitzuloggen. Vielleicht sehe ich dann mehr!

lumapu, 2024-02-28:
Fehler identifiziert, der JSON-String ist zu lang für den reservierten Bereich (war 140 Zeichen). Wird in der nächsten Version gefixt (kommt heute Nacht) 😊

knickohr, 2024-02-29:
Häh ?!  Kann aber nur ein Problem vom 8266 sein, beim Fusion (ESP32) funktioniert es.

lumapu, 2024-02-29:
liegt auch an der Konfiguration, und wie lange die Statistik schon zählt

closed 2024-03-02 by zechnerhubert