lumapu/ahoy #1459: AhoyDTU did not run after update from Ver 0.8.60 to 0.8.83/86. No life-data. No conection to nRF24L01+.

jotroj, 2024-02-28:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.60 is running, update to 0.8.66 -  no conection to nRF

### Github Hash

https://github.com/lumapu/ahoy/commits/94f5a2a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

keine Änderungen im Setup durch mich. Nach update zurück auf 0.8.60 läuft wieder alles. Normal werden 3 Inverter eingelesen.

### Error description

AhoyDTU did not run after update from Ver 0.8.60 to 0.8.83/86. No life-data. No conection to nRF24L01+. keine Änderungen im Setup durch mich. Nach update zurück auf 0.8.60 läuft wieder alles. Normal werden 3 Inverter eingelesen.

lumapu, 2024-02-28:
can you try to do the update during night time (no communication). We improved the communication a lot. If you directly start into the day with the 0.8.83 version you get better results.

jotroj, 2024-03-02:
Mit Version 0.8.63 funktioniert die Kommunikation mit Nrf. Nach [NTP]: getNtpTime failed und webSerial, buffer overflow!
Mit Version 0.8.64 funktioniert die Kommunikation mit Nrf nicht mehr.getNtpTime failed
Mit Version 0.8.86 funktioniert die Kommunikation mit Nrf nicht. Auch nicht bei nächtlichem reboot.
0.8.63 - 77b1f96
---------------
10:15:25.775 I: [NTP]: getNtpTime failed
webSerial, buffer overflow!
10:15:32.896 RadhausDach/ch2/Irradiation: 35.400 %
10:15:32.897 I: RadhausDach/ch2/MaxPower: 142.500 W
10:15:32.900 I: RadhausDach/ch0/U_AC: 233.800 V.........

tschoenfelder, 2024-03-02:
Hallo,

hatte ich bei einem von 2 Ahoy's auch. Auf der System-seite stand Interrupt-Pin not working. Scheinbar waren die PIN-Zuordnung beim Upgrade verrutscht. Ich habe dann mit Tasmotizer mit Erase .86 neu installiert, PINs und Inverter neu eingegeben und es ging (und läuft seit 3 Tagen ....).

lumapu, 2024-03-03:
@jotroj kannst du bei dir auch die von @tschoenfelder beschriebenen Punkte prüfen und Feedback geben?

jotroj, 2024-03-14:
Weiterer Test mit 0.8.93: 1.) Internet Zugang für [NTP] gesperrt. Sync für System auf "From Browser", NTP Server / IP wie "pool.ntp.org" --> keine Kommunikation mit Nrf. Meldung [NTP]: getNtpTime failed
2.)Internet Zugang für [NTP] gesperrt. Sync für System auf "From Browser", NTP Server / IP wie "**pool.ntp.orgi**" (keine gültige Adresse)  --> Kommunikation mit Nrf funktioniert. 
3.)Keine History

Gubi2023, 2024-03-14:
zu 3.)Keine History:  FW  ...esp8266.bin  enthält keine History mehr, nur noch  ..esp8266-all.bin

lumapu, 2024-08-16:
sollte aktuell nicht mehr so sein, wenn ich mich nicht irre.

stefan123t, 2024-10-30:
@jotroj kannst Du bitte nochmal mit der aktuellen Version 0.8.152 testen ? 
Laut lumapu sollte das Problem behoben sein und wir könnten dann das Issue schließen.

jotroj, 2024-10-30:
Keine Probleme mit Version 0.8.152 gesehen.

closed 2024-10-30 by jotroj