lumapu/ahoy #1460: [Bug] Support HMS

Altblechkutscher, 2024-02-28:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard config, 2 Inverters added 
HMS-1800-4T
HMS-400-1T
Module is CMT2300A
Connection is working to HMS-1800-4T

### Error description

When adding the HMS-400-1T, the 4th digit of the serial number is a "5", which means it is not detected as HMS-400 with the CMT2300A-Radio, it shows the menu that is known from the HM-Series. 
[image]
[image]

If possible, there shoudl be an option to select the radio that should be used to overwrite the default recognition.

lumapu, 2024-02-28:
ok that's a new version - never saw it before

knickohr, 2024-02-28:
Eigentlich kein Bug, eher eine Erweiterung. Aber OK.

Noch ein Tipp :  der rechte Aufkleber mit der SN läßt sich abziehen und ist für Deine Unterlagen bestimmt. Nicht das der nächste Regen ihn abwäscht und Du keine SN mehr hast 😉

lumapu, 2024-02-28:
den bekommen wir morgen zum laufen 😊. Dann brauchen wir noch die HW-IDS und er ist voll integriert.

Altblechkutscher, 2024-02-28:
Labels sind abfotografiert, auf dem Karton ist auch noch ein Label.

Schlau mich noch kurz auf - HardwareID bekomme ich wie ausgelesen?

lumapu, 2024-02-28:
[image]

lumapu, 2024-02-28:
das war zu schnell, in Zeile 54 muss lauten:
`if(((iv->config->serial.b[4] & 0x0f) == 0x04) || ((iv->config->serial.b[4] & 0x0f) == 0x05)) {`

Altblechkutscher, 2024-02-28:
Bist Du Dir sicher mit Zeile 54, ich tippe auf 55 :-D

`if((iv->config->serial.b[4] & 0x0f) == 0x04) { ` zu `if(((iv->config->serial.b[4] & 0x0f) == 0x04) || ((iv->config->serial.b[4] & 0x0f) == 0x05)) {`

So, habe mal schnell VSCODE installiert mit platformio, konnte für ESP32 kompilieren. Im Live sieht es schon mal halbwegs passend aus:

[image]

lumapu, 2024-02-28:
ja bin mir sicher, du hast oben schon eine Zeile eingefügt, ich hab mich aufs Original bezogen 😊. 
Sieht doch schon gut aus, evtl. das Bild nochmal ersetzten, jetzt sieht man deine Seriennummer ... (mir macht das nichts)

Altblechkutscher, 2024-02-28:
Haste zweimal wahr. Danke für den schnellen Support.

lumapu, 2024-02-28:
ist mir eine Freude 

die Radio Settings sind aber noch nicht auf HMS / HMT denke ich, da braucht man noch eine Anpassung in der html Seite `src/web/html/setup.html` in Zeile 789: `case 0x0005:` vor `case 0x0004: ...`

Altblechkutscher, 2024-02-29:
Guten Morgen,
habe das aktuelle devrelease mal gezogen und geflasht.
Ich würde sagen Teilerfolg:
[image]

Kurzer Blick ins Log ergibt noch einen check failed:

07:15:37.000 I: (#0) Radio infos: t: 4, s: 2, f: 0, n: 2 | p: 20
07:15:37.002 I: (#0) TX 863.00Mhz | 15 00 80
07:15:37.136 I: (#0) RX  82ms | 27 -42dBm | 95 81
07:15:37.137 I: (#0) Payload (14)
07:15:37.139 -----
07:15:37.140 I: (#0) Radio infos: t: 5, s: 3, f: 0, n: 2 | p: 20
07:15:37.141 I: (#0) TX 863.00Mhz | 15 0b 80
07:15:37.372 I: (#0) RX  83ms | 27 -42dBm | 95 01
07:15:37.373 I: (#0) RX 133ms | 27 -43dBm | 95 02
07:15:37.374 I: (#0) RX 179ms | 23 -42dBm | 95 83
07:15:37.375 I: (#0) Payload (42)
07:15:37.375 E: plausibility check failed, expected 30 bytes
07:15:37.386 -----
07:15:37.387 I: com loop duration: 387ms
07:15:37.387 -----

lumapu, 2024-02-29:
sehr cool, das bekommen wir auch noch hin 😎

lumapu, 2024-02-29:
ist das ein zwei kanalinger? Ich dachte der hatte nur einen DC Kanal, daher der Plausibilität Check Fehler

MetaChuh, 2024-02-29:
@lumapu 
ist ein einkanalinger.

[image]

lumapu, 2024-02-29:
von den Daten her kommen aber wohl zwei Kanäle

MetaChuh, 2024-02-29:
> von den Daten her kommen aber wohl zwei Kanäle

is hoid' lei a schizo' inverter 😉

ernst: keine ahnung, evtl haben sie in den hms-1t die elektronik für die dual inverter eingebaut, und nur nicht beschaltet ?
wer meldet sich freiwillig seinen hms-1t während der garantiezeit aufzuschrauben und die fotos davon zu posten ?

knickohr, 2024-02-29:
> is hoid' lei a schizo' inverter 😉
>
> ernst: keine ahnung, evtl haben sie in den hms-1t die elektronik für die dual inverter eingebaut, und nur nicht beschaltet ? wer meldet sich freiwillig seinen [...]

Du ! 😁

Altblechkutscher, 2024-02-29:
Wie kann ich denn die 42 Byte sichtbar machen, wo der Plausicheck fail ist? Evtl gibt das mehr Aufschluss.

lumapu, 2024-02-29:
[image]

lumapu, 2024-02-29:
[image]

Altblechkutscher, 2024-03-01:
Moin,
hier mal n Log

Jetzt nur der 400er, da sieht man das Plausi Problem 

0.8.88 - 2bc6610
---------------
07:24:03.000 I: (#0) Radio infos: t: 2, s: 1, f: 0, n: 1 | p: 20
07:24:03.003 I: (#0) TX 863.00Mhz | 15  *  *  *  * 87 66 36 94 80 01 00 65 e1 74 83 00 00 00 00 00 00 00 00 bb 42 fa 
07:24:03.139 I: (#0) RX  82ms | 27 -40dBm | 95  *  *  *  *  *  *  *  * 81 27 12 07 e6 02 74 07 fa 00 64 00 00 00 00 9d 95 c3 
07:24:03.142 I: (#0) Payload (14): 27 12 07 e6 02 74 07 fa 00 64 00 00 00 00 
07:24:03.145 -----
07:24:03.146 I: (#0) Radio infos: t: 3, s: 2, f: 0, n: 1 | p: 20
07:24:03.157 I: (#0) TX 863.00Mhz | 15  *  *  *  * 87 66 36 94 80 0b 00 65 e1 74 83 00 00 00 00 00 00 00 00 b1 48 f0 
07:24:03.391 I: (#0) RX  84ms | 27 -40dBm | 95  *  *  *  *  *  *  *  * 01 00 01 01 3f 00 00 00 02 00 00 00 07 00 00 00 00 4a 
07:24:03.394 I: (#0) RX 133ms | 27 -40dBm | 95  *  *  *  *  *  *  *  * 02 08 47 00 00 00 00 00 00 00 00 08 a7 13 87 00 00 07 
07:24:03.408 I: (#0) RX 180ms | 23 -40dBm | 95  *  *  *  *  *  *  *  * 83 00 b3 00 00 00 00 00 27 00 01 4f 75 5d 
07:24:03.410 I: (#0) Payload (42): 00 01 01 3f 00 00 00 02 00 00 00 07 00 00 00 00 08 47 00 00 00 00 00 00 00 00 08 a7 13 87 00 00 00 b3 00 00 00 00 00 27 00 01 
07:24:03.425 E: plausibility check failed, expected 30 bytes
07:24:03.436 -----
07:24:03.437 I: com loop duration: 437ms
07:24:03.437 -----
07:24:18.000 I: (#0) Radio infos: t: 4, s: 2, f: 0, n: 2 | p: 20
07:24:18.002 I: (#0) TX 863.00Mhz | 15  *  *  *  * 87 66 36 94 80 00 00 65 e1 74 92 00 00 00 00 00 00 00 00 2a 82 bb 
07:24:18.140 I: (#0) RX  84ms | 27 -41dBm | 95  *  *  *  *  *  *  *  * 81 27 12 10 20 41 04 01 00 0a 00 20 01 00 00 7c 25 c3 
07:24:18.143 I: (#0) Payload (14): 27 12 10 20 41 04 01 00 0a 00 20 01 00 00 
07:24:18.146 -----
07:24:18.147 I: (#0) Radio infos: t: 5, s: 3, f: 0, n: 2 | p: 20
07:24:18.159 I: (#0) TX 863.00Mhz | 15  *  *  *  * 87 66 36 94 80 0b 00 65 e1 74 92 00 00 00 00 00 00 00 00 e1 88 71 
07:24:18.391 I: (#0) RX  82ms | 27 -41dBm | 95  *  *  *  *  *  *  *  * 01 00 01 01 3f 00 00 00 02 00 00 00 07 00 00 00 00 4a 
07:24:18.394 I: (#0) RX 132ms | 27 -41dBm | 95  *  *  *  *  *  *  *  * 02 08 47 00 00 00 00 00 00 00 00 08 ab 13 86 00 00 0a 
07:24:18.408 I: (#0) RX 179ms | 23 -41dBm | 95  *  *  *  *  *  *  *  * 83 00 b4 00 00 00 00 00 27 00 01 fb 5d c6 
07:24:18.410 I: (#0) Payload (42): 00 01 01 3f 00 00 00 02 00 00 00 07 00 00 00 00 08 47 00 00 00 00 00 00 00 00 08 ab 13 86 00 00 00 b4 00 00 00 00 00 27 00 01 
07:24:18.425 E: plausibility check failed, expected 30 bytes
07:24:18.436 -----
07:24:18.437 I: com loop duration: 437ms
07:24:18.437 -----

Altblechkutscher, 2024-03-01:
Guten Abend,
helft mir mal bitte auf die Sprünge, gibt es eine Protokolldefinition für die HMS irgendwo?

lumapu, 2024-03-01:
nicht so einfach, könntest du unter Tags nochmal eine Aufnahme machen? Es sind zu viele Werte 0. Für mich sieht es so aus, als würden die jetzt 2ch Wechselrichter mit nur einem Anschluss bauen. Die Payload, soweit ich sie entschlüsseln kann entspricht genau dem Format, wie wir sie von den 2-kanal Wechselrichtern kennen:

[image]

Probiere bitte mal die `0.8.89` aus, die sollte sich mit diesem Wechselrichter jetzt richtig verhalten. (sofern meine Annahmen stimmen)

Altblechkutscher, 2024-03-02:
Na, das sieht doch gut aus! Mal noch abwarten ob die Daten ggü. dem Hms1800 plausibel sind wenn etwas die Sonne rauskommt. Danke schonmal!

[image]

Altblechkutscher, 2024-03-02:
Noch ein Update, ich würde sagen passt, haken dran!
[image]

closed 2024-08-16 by lumapu