lumapu/ahoy #1461: Mehr Zellenleistung als der WR kann = Bootloop[Bug]

sandboxgangster, 2024-02-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Für die Berechnung der Irradiation muß man die angeschlossene Peak Leistung angeben wen die aber hör ist als der Wechselrichter kann, landest die Ahoy SW in einem Bootloop.

Ich hab 430Wp an einem HM-300 und kann da max. 299Wp angeben ab 300Wp bootet der ESP8266 nicht mehr.

MetaChuh, 2024-02-29:
hi @sandboxgangster 

> Für die Berechnung der Irradiation muß man die angeschlossene Peak Leistung angeben wen die aber hör ist als der Wechselrichter kann, landest die Ahoy SW in einem Bootloop.

hast du den bootloop auch, wenn du alle api queries bzw. mqtt deaktivierst, und danach dein hm-300 limit wieder auf 300w (+1w) setzt ?

bitte dies testen, und den eventuellen bootloop am besten mit dauerping beobachten, statt auf das web interface zuzugreifen.

dauerping für windows:
`ping /t deine_ahoydtu_ip`

dauerping für linux, dsm, mac, etc:
`ping deine_ahoydtu_ip`

zusatzinfo für beginner die dies evtl. lesen, nicht für dich:
bitte alle `deine_ahoydtu_ip` werte gegen die ip deiner ahoydtu austauschen, z.b.: `192.168.0.10`

sandboxgangster, 2024-02-29:
Ich nutze MQTT nicht, lese die Daten über die REST Api aus, aber die ist dann auch nicht erreichbar bis jetzt klappt eine Wiederbelebung nur wenn ich die FW neu flashe und den ESP und die Config Wipe.

OB der ESP wiederkommt wenn ich die REST API nicht mehr abfrage kann ich mal Testen, schaffe ich allerdinks erst heute Abend.

Danke

knickohr, 2024-02-29:
Du hast die Modulleistung schon hier eingegeben ?

[image]

Ich kann das Problrm nicht nachstellen, habe bei allen Invertern mehr Wp als der Inverter kann.

MetaChuh, 2024-02-29:
> Maybe this would be the perfect time for removing "irradiation" completely. As already discussed, this value cannot be calculated correctly at all.

yes maybe.
but as this value is present at all dtu projects, removing it might make some newer users believe, that one dtu processes less data as another dtu, because one value, albeit meaningless, is missing.

technics42, 2024-02-29:
Make this DTU-project better than others and remove errors 😁
Someone is always the first one!

sandboxgangster, 2024-02-29:
> Du hast die Modulleistung schon hier eingegeben ?
> 
> Ich kann das Problem nicht nachstellen, habe bei allen Invertern mehr Wp als der Inverter kann.

Genau da hapert es bei mir, also sobald ich >299 angebe ist der ESP nicht mehr erreichbar, d.h. ich muss ihn den Flash komplett löschen und die FW neu Flaschen.

Ich kann es jetzt etwas eingrenzen, hab mal eine größere zahl angegeben zum Testen 800Wp da startet der ESP Normal nässter versuch war dann 360Wp ESP wieder tot.

Oben war angemerkt mal Ping zu nutzen da kommt absolut nix zurück.

Hab noch mal geschaut, und per Minicom geschaut was die Console dann sagt
`scanning APs with SSID YOUR_WIFI_SSID                                                                                                                       
scanning APs with SSID YOUR_WIFI_SSID                                        
scanning APs with SSID YOUR_WIFI_SSID                                        
...`

Also macht er einen AP auf das war mir vorher nicht aufgefallen, komme auch drauf aber die komplette Config ist weg.

stefan123t, 2024-10-30:
@sandboxgangster kannst Du das mit der aktuellen Version 0.8.152 nochmal prüfen, ob der Fehler noch auftritt ?

Scheint wohl ein OOM Problem beim ESP8266 zu sein, das dann beim Speichern der Einstellungen zu Problemen (bootloop) führt.

sandboxgangster, 2024-10-30:
Leider habe ich vor einiger zeit auf den ESP32 gewechselt, kann das nicht mehr testen.

stefan123t, 2024-10-30:
@sandboxgangster Du bist leider der Einzige der das Problem bisher berichtet / bestätigt hat :slightly_smiling_face: 

lumapu hat zwar an anderer Stelle auch eine Bootloop behoben / beseitigt, aber ob damit das issue als gelöst betrachtet werden kann, wissen wir nicht.

Kannst Du evtl. Deine alte ESP8266 AhoyDTU noch einmal reaktivieren, oder sollen wir das Issue als nicht reproduzierbar zu machen ?

sandboxgangster, 2024-10-30:
Ich kann es nochmal testen, hab den aufbau noch liegen, schaffe ich aber erst am Wochenenden, dann krame ich es noch mal raus.

stefan123t, 2024-11-02:
@sandboxgangster prima dann warten wir hier einfach noch ein paar Tage oder eine Woche auf Deine Ergebnisse. 
Danke schon mal fürs Nachsehen!

sandboxgangster, 2024-11-03:
Moinsen, hab es gerade mit der "241007_ahoy_0.8.152_78cbd77_esp8266.bin" getestet ich kann den Fehler nicht mehr nachstellen, scheint also behoben zu sein, hab mehrere Werte getestet keine Probleme mehr.

knickohr, 2024-11-03:
Dann mach zu das Ding 😉

closed 2024-11-03 by sandboxgangster