lumapu/ahoy #1462: Verlaufsanzeige ohne Login im Browser sichtbar [Bug]

hhammerl, 2024-02-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

round dot

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.86

### Github Hash

D94E2765F8A2F13496F041CB465C6030 

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Update FW

### Error description

When I open the webpage, no login is required to see Verlauf

knickohr, 2024-02-29:
Please check this Setting :

[image]

hhammerl, 2024-02-29:
Hi knickohr,

thanks allot, I did not know that up to now

closed 2024-02-29 by hhammerl