lumapu/ahoy #1463: Board wird auf History Seite nicht angezeigt [Bug]

mellow5, 2024-02-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.88

### Github Hash

2bc6610

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Update von 0.8.83 auf 0.8.88.

### Error description

Der Name des verwendeten Boards wird auf der Seite "History" nicht angezeigt. Ist mir zufällig aufgefallen. Kann nicht mit Sicherheit sagen, ob es in früheren Versionen auch schon so war.

MetaChuh, 2024-03-03:
@mellow5 
thx for pointing this out, testing the fixes, and closing this issue 👍🙏

this helps @lumapu, as well as the whole open source community, to spend less additional time, to review abandoned or deprecated issues.
much appreciated gesture, which deserves a big thanks to you 👍🙏

greetings,
metachuh

mellow5, 2024-03-14:
In 0.8.93 fehlt das Board wieder auf der Historyseite.

Gubi2023, 2024-03-14:
Ebenso fehlt die Feldstärkeanzeige links unten
[image]

mellow5, 2024-03-16:
Geht wieder

closed 2024-03-16 by mellow5