lumapu/ahoy #1464: Peinliche Frage: Wo ist jetzt die Leistungserstellung ab 0.8.36

kiu77, 2024-03-01:
Bei den neueren Versionen finde ich peiplicherweise die Einstellung der Leistung nicht mehr.
Ich stehe wahrscheinlich mit beiden Beinen auf der Leitung.
Schubst mich jemand runter?

knickohr, 2024-03-01:
Guckst Du da 😉

[image]

kiu77, 2024-03-01:
Vielen herzlichen Dank!
Ich bin einfach nicht drauf gekommen, dass sich an der Stelle eine Schaltfläche verbirgt.
Das sollte man vielleicht farblich akzentuieren, damit es auffällt.

mroenne2022, 2024-03-01:
@kiu77 
Das sind tatsächlich 3 Schaltflächen von links , ausser Temp. nicht. Das war mir auch noch nicht aufgefallen , von daher hat dein Issue mir auch geholfen :-) Vielen Dank für die Info @knickohr ;-)

kiu77, 2024-03-01:
Vorschlag, wie es aussehen könnte:

[image]

kiu77, 2024-03-01:
Oder so:

[image]

MetaChuh, 2024-03-01:
@kiu77 

> Das sollte man vielleicht farblich akzentuieren, damit es auffällt.

idealerweise neon pink, und je mehr % eingestellt sind, desto schneller stroboskop-blinkend ... disco disco 😃

kiu77, 2024-03-01:
;-)
Jaja...
Aber zugeben würdest Du, dass das in der aktuellen Form etwas subliminal für die Wahrnehmung ist.
Bei den Einstellungsseiten ist ja auch klar, wo ne Schaltfläche ist.
Ist fast wie hier: [https://www.youtube.com/watch?v=6DZ4x75MPNA](url)

knickohr, 2024-03-01:
Das ist die Dau-Sicherung 🤪🫣

Aua, nicht hauen 🙉🙈🙊

kiu77, 2024-03-01:
Der war gut ☺
Aber diesen DAU hier hat es ja nicht wirklich abschrecken können...

lumapu, 2024-03-02:
Ich finde die Frage nicht peinlich sondern berechtigt, hier müssen wir uns was schönes einfallen lassen.

firedp, 2024-03-02:
Ich hatte vor Wochen ja die gleiche Frage 
😂

MetaChuh, 2024-03-02:
beispiel mit regler schaltfläche (mock-up):
etwa so ?

[image]

> @kiu77 : Ist fast wie hier: [https://www.youtube.com/watch?v=6DZ4x75MPNA](https://github.com/lumapu/ahoy/issues/url)

hahaha, so ein psychoterror ist die schaltfläche sicher nicht 🙈😂

ralfberger, 2024-03-08:
würde mir bitte jemand sagen was ich unter 
Control : Restart  /  Turn off  /  Turn on
verstehen soll?
Bei Leistungsüberschreitung  Restart oder  Aus- bzw. Einschalten??

kiu77, 2024-03-08:
@ralfberger 
???
Was, wer, wie, wo?

knickohr, 2024-03-08:
Man kann den WR hier ein- und ausschalten, oder rebooten.

ralfberger, 2024-03-08:
Sorry wenn ich mich unklar ausgedrückt habe.
Das war gemeint.

[image]

DanielR92, 2024-08-30:
Kann man dieses Issue schließen?

ralfberger, 2024-08-30:
ich denke schon!

stefan123t, 2024-10-30:
@DanielR92 nein, denn die Buttons / Funktionen sind immer noch nicht als solche erkenntlich:

Das ist m.E. schlechtes UX Design und auf Arbeit bekäme ich beim Barrierefreiheits-Test Probleme wenn die Elemente nicht klar erkennbar und fokusierbar sind. Die anderen Buttons z.B. unter Settings sind ja auch alle Blau mit runden Ecken.

**Success Criterion 3.2.7: Visible Controls**
https://w3c.github.io/wcag/understanding/visible-controls.html

> Oder so:
> 
> ![Bildschirmfoto 2024-03-01 um 14 24 26](https://private-user-images.githubusercontent.com/43971413/309309161-b80f9632-145d-4b66-825c-7767085370c1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyNDgxNjUsIm5iZiI6MTczMDI0Nzg2NSwicGF0aCI6Ii80Mzk3MTQxMy8zMDkzMDkxNjEtYjgwZjk2MzItMTQ1ZC00YjY2LTgyNWMtNzc2NzA4NTM3MGMxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDAwMjQyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhMTVmY2NkYmYzNmViYzA3NTQzMDk0NjNjYjU2ZjgwNmY3NmFlZmJlYTI0OTc2ZDg4YWFjYWNiNGE0N2ZjZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.m8p7R-4CV7RM71zxDj1N2V_rDDt93LruIDMqXHbOhxI)