lumapu/ahoy #1466: Funkmodul CMT2300A: Kann man damit HMS- und HM-Inverter auslesen?

kiu77, 2024-03-01:
Die eigentliche Frage ist:
Kann das Modul CMT2300A nicht nur die neuen HMS-Typen, sondern auch noch die alter HM-Typen auslesen?
Und geht das auch gemischt? Also 1 DTU mit CMT2300A für z.B. 1x HM-800 und 1x HMS-800 gleichzeitig?
Reicht technisch kann das Modul  ja einen enormen Frequenzbereich abdecken. Aber hat das schon mal jemand ausprobiert?

kiu77, 2024-03-01:
Meine Suche führt zu folgenden Resultaten:
1. Für HM-Inverter braucht es ein NRF24L+ und für HMS-Inverter ein CMT2300A (und zwar eins fürs 868-MHz-Band).
2. Man kann beide Funkmodule an einen ESP anschließen, wobei dann klar ein ESP32 genutzt werden muss.
3. Bei einer DTU aus ESP32 und beiden Funkmodulen kann man mehrere Inverter der HM- und HMS-Serien gemischt betreiben und auslesen.

Richtig?

Peertje-62, 2024-03-01:
Yes, 
https://github.com/tbnobody/OpenDTU
regards

Peertje-62, 2024-03-01:
look at:
https://3printd-solution.eu/Ahoy-openDTU-Fusion-fuer-HM-HMS-HMT-mit-Display

lumapu, 2024-03-02:
@kiu77 gute Idee nur ein Funktmodul zu nutzen, aber:
* HM / MI benötigt 2.4 GHz
* HMS / HMT benötigen 868MHz
das CMT Funkmodul kann nur den Frequenzbereich von 140 - 1020MHz abdecken und daher nicht für beide einsetzbar. Beim NRF24 bin ich mir nicht sicher, aber vermute ähliches nur im höheren Frequenzbereich.
Unabhängig davon muss man auch alles was außenrum noch gesendet wird (das seht ihr nicht) richtig konfiguriert werden. Die Rede ist hier von Präamble, Sync, CRC, ...

Wie du schon richtig sagst unterstüzt Ahoy beim ESP32 bereits beide Module parallel, das wird wohl auch so bleiben.

@Peertje-62 Please delete your first comment because it's simply nonsens to have a link to OpenDTU without having any additional info! The second one is better.

Peertje-62, 2024-03-02:
Dear Lumapu,
Did that but I don't agree, in the link was an list with inverters and radio modules
[image]

kiu77, 2024-03-02:
Thank you both. All clear now to me.

lumapu, 2024-03-02:
Thank you. Next time point that out, I didn't get the message to look for a table of inverters. In general I think it doesn't fit to the question. He asked if the radios can be combined to only use one of them for all generations of Hoymiles inverters.

Peertje-62, 2024-03-02:
Hi, 
You are right and I appreciate the work and energy you put into this project and I am still learning every day.
regards

lumapu, 2024-03-02:
Thank you. I appreciate the team work here and in Discord. It's so great that mostly every question has an answer within the first hour. (even in night time CET) 😊

Peertje-62, 2024-03-02:
Hi kiu77,
If your problems are solved you could /should close this issue
have a nice day
regards

kiu77, 2024-03-02:
It's partially solved.
lumapu did read it (thanks) so we can hope ;-)

lumapu, 2024-03-03:
will be closed with next release

closed 2024-08-16 by lumapu