lumapu/ahoy #1467: [Bug]Link funktioniert bei mir nicht (?)

Gubi2023, 2024-03-02:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.86

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update--Download .....

### Error description

seit neuestem ist wieder ein Link auf der Install-Seite vorhanden. 

https://fw.ahoydtu.de/fw/dev%2Fahoy_v0.8.86/

Leider funktioniert der bei mir nicht

Gubi2023, 2024-03-02:
ebenso passiert bei der CHANGES.md nichts...(leeres Fenster öffnet sich)

MetaChuh, 2024-03-02:
hi @Gubi2023 

ich hab' keine bugs der links in v0.8.89 gefunden.
(ps: sind auch in v0.8.86 identisch und unverändert).
hier ein kleines testvideo.

https://github.com/lumapu/ahoy/assets/45773305/016a64a1-375e-455c-9542-83d42885d5cb

Gubi2023, 2024-03-02:
Ok. Anbei mein Video zur Klarstellung:

https://github.com/lumapu/ahoy/assets/142897856/fabc7624-6c1b-4107-9356-e19c5f9c3d34

MetaChuh, 2024-03-02:
du hast recht, stimmt.

die beiden links auf der website liefern auch beim herunterladen mit curl nur 0 byte files.
ich hatte vorher nur die links innerhalb der ahoydtu getestet.

soweit ich sehe betrifft es nur diese beiden links auf fw.ahoydtu.de richtig ?

https://fw.ahoydtu.de/fw/download/dev%2Fahoy_v0.8.89%2FCHANGES.md

und

https://fw.ahoydtu.de/fw/download/dev%2Fahoy_v0.8.89%2Finstall.html

Gubi2023, 2024-03-02:
scheint so, auf die schnelle habe ich keine weiteren gefunden, die nicht gehen.

MetaChuh, 2024-03-02:
@lumapu 
ich glaube die beiden files werden am nginx wegen ihrer dateiendung zum intern parsen abgefangen.

ps: falls viele user den web installer nutzen sollten:
für die web installer nutzung ist's ideal, gleich ein zip bereitzustellen, weil sie manifest.json, index.html, sowie den ordner für ihre plattform in der richtigen ordnerstruktur brauchen.
(sonst müssen sie die ordnerstruktur lokal selbst und korrekt nachbilden)

was meinst du, verwenden das viele, oder ist's was seltenes, was man vorerst so lassen kann ?

Gubi2023, 2024-03-02:
> was meinst du, verwenden das viele, oder ist's was seltenes, was man vorerst so lassen kann ?
ich finde, wenn was angeboten wird, sollte es auch funktionieren!

lumapu, 2024-03-03:
jetzt geht das, wie @MetaChuh richtig vermutet hat, wurden diese beiden Extensions geblockt, habe sie jetzt freigegeben.

Gubi2023, 2024-03-03:
danke für die schnelle Abhilfe !!