lumapu/ahoy #1468: Spontane Reboots bei fehlendem NTP

knickohr, 2024-03-02:
In letzter seit fällt es immer wieder auf das bei fehlerhafter oder fehlgeschlagener NTP-Abfrage die DTU sich rebootet. Der Bug muß schon sehr alt sein, tritt aber kaum auf.

Testzenario von heute :
Ich betreibe einen eigenen NTP-Server der die Daten von GPS-Satelliten holt. Ich brauche eine sehr genaue Zeit, im Mikrosekundenbereich, um damit eine Blitzortungstation zu betreiben. Aber das tut nichts zur Sache, Tatsache ist jedoch, fällt dieser NTP aus, dann erfolgt nach einer fehlerhaften oder fehlgeschlagenen NTP-Abfrage der DTU ein Reboot. Noch schlimmer, fehlt die Zeit über einen längeren Zeitraum, so kommt es sogar zu Rebootschleifen. Solange bis die Zeit wieder erfolgreich abgefragt werden kann.

Würde auch zu dem passen das User hin und wieder über unerklärliche spontane Reboots berichten.

You69Man, 2024-03-02:
Ich hatte die NTP Abfrage auch schon öfter mal in Verdacht, evt. für sporadische Roboots, die manchmal gemeldet werden, verantwortlich zu sein. Insbesonderen, weil die Funktion getNtpTime() ein NTP Paket über UDP empfängt (welches ja im Gegensatz zu TCP selbst überhaupt keine Sicherungsschicht enthält), und dann die empfangenen Daten dennoch ohne irgendeine weitere Plausibilisierung als aktuelle Zeit übernimmt. Ich kenne mich mit NTP zu wenig aus, wie man hier typischerweise einen sinnvollen Check macht, aber vermutlich könnte man zumindest einmal prüfen, ob das Jahr einigermaßen stimmt, bzw. wie weit die empfangene Zeit von der zuletzt empfangenen Zeit abweicht, bevor man die neue Zeit akzeptiert. Da solche NTP- (bzw. UDP-Übertragungs-) Fehler vermutlich nur sehr selten passieren, ist es halt auch nicht ganz einfach so etwas zu debuggen.

knickohr, 2024-03-02:
Ich glaube nicht das es am Inhalt liegt, sondern eher daran das nichts kommt und der Prozeß ewig warten. Und dann kommt Wadi, unser Wachhund und beißt zu 😵

You69Man, 2024-03-02:
Die Implementierung für die NTP Abfrage ist dzt. so:
Zunächst wird die Anfrage an den NTP Server geschickt. Danach wird 5*1500ms auf die Antwort gewartet. Innerhalb der Warteschleife(n) wird in parsePacket ein optimistic_yield() aufgerufen, welches bei Bedarf den WDT wartet. Ist nach den 7500ms keine Antwort gekommen gibt die NTP Abfrage mit einem "getNtpTime failed " auf. Es folgt dann aber kurze Zeit darauf (5s) ein neuer NTP Ticker mit einer neuerlichen Anfrage.

Ich habe gerade einmal einen kleinen Versuch gemacht, indem ich in ahoywifi.cpp Zeile 295 mit einem +1 manipuliert habe, sodass ein vollständiges angekommenes Paket nicht mehr erkannt wird, und daher weiter gewartet wird:
if(NTP_PACKET_SIZE **+1** <= mUdp.parsePacket()) {

Das Ergebnis sieht dann so aus (es ist schon nach Sonnenuntergang):
--------------------------------
Welcome to AHOY!

point your browser to http://192.168.0.9 (Station)
to configure your device
--------------------------------

Wifi event: 11
Wifi event: 10
[WiFi] AP disabled
Wifi event: 11
[WiFi] mDNS established: AHOY-DTU_ESP32.local
I: [NTP]: getNtpTime failed
I: [NTP]: getNtpTime failed
I: (#0) Radio infos: 0 0 0 0 0 | t: 0, s: 0, f: 0, n: 0 | p: 1
I: [NTP]: getNtpTime failed
I: (#0) TX 27 CH23, 15 ret. | 15 0B 80
I: [NTP]: getNtpTime failed
I: [NTP]: getNtpTime failed
I: (#0) request timeout: 14952ms
-----
I: [NTP]: getNtpTime failed
I: (#0) Radio infos: 0 -2 0 0 0 | t: 1, s: 0, f: 0, n: 1 | p: 1
I: [NTP]: getNtpTime failed
I: (#0) TX 27 CH40, 15 ret. | 15 0B 80

Es kommt also kein Watchdog reset, sondern immer neue NTP Anfrageversuche.

lumapu, 2024-05-30:
tritt nicht mehr auf, denke ich .. was denkt ihr?

knickohr, 2024-06-02:
Bisher nicht mehr aufgetreten. Machen wir zu 👍

closed 2024-06-02 by knickohr