lumapu/ahoy #1469: Wechselrichter nicht erreichbar nach neuem Wlan Setup und Update

Jakmanny, 2024-03-02:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

keine Änderungen zum Standard -Setup:

### Debug Serial Log output

```shell
{"inverter":[{"enabled":true,"id":0,"name":"TSun","serial":"11418299xxxx","channels":2,"freq":0,"disnightcom":false,"pa":0,"ch_yield_cor":[0,0],"ch_name":["",""],"ch_max_pwr":[435,435]}],"interval":"30","max_num_inverters":4,"rstMid":false,"rstNotAvail":false,"rstComStop":false,"strtWthtTm":false,"rdGrid":true,"rstMaxMid":false,"yldEff":1}

{"id":0,"enabled":true,"name":"TSun","serial":"11418299xxxx","version":"0","power_limit_read":65535,"power_limit_ack":false,"max_pwr":0,"ts_last_success":0,"generation":1,"status":0,"alarm_cnt":0,"rssi":0,"ts_max_ac_pwr":0,"ch":[[0,0,0,0,0,0,0,0,0,0,0,0],[0,0,0,0,0,0,0],[0,0,0,0,0,0,0]],"ch_name":["AC","",""],"ch_max_pwr":[null,435,435]}

{"id":0,"name":"TSun","serial":"11418299xxxx","generation":1,"max_pwr":0,"part_num":0,"hw_ver":0,"prod_cw":29,"prod_year":2022,"fw_date":"0-00-00","fw_time":"00:00","fw_ver":"0.00.00","boot_ver":0}

{"wifi_rssi":-59,"ts_uptime":375,"ts_now":1709379702,"version":"0.8.83","modules":"MDH-de","build":"5ebfe5a","env":"esp8266-de","menu_prot":false,"menu_mask":61,"menu_protEn":false,"cst_lnk":"","cst_lnk_txt":"","region":0,"timezone":1,"esp_type":"ESP8266"}

{"generic":{"wifi_rssi":-57,"ts_uptime":393,"ts_now":1709379720,"version":"0.8.83","modules":"MDH-de","build":"5ebfe5a","env":"esp8266-de","menu_prot":false,"menu_mask":61,"menu_protEn":false,"cst_lnk":"","cst_lnk_txt":"","region":0,"timezone":1,"esp_type":"ESP8266"},"ts_now":1709379720,"ts_sunrise":1709359488,"ts_sunset":1709398719,"ts_offsSr":600,"ts_offsSs":600,"inverter":[{"enabled":true,"id":0,"name":"TSun","cur_pwr":0,"is_avail":false,"is_producing":false,"ts_last_success":0}],"disNightComm":false,"warnings":[]}

{"generic":{"wifi_rssi":-57,"ts_uptime":411,"ts_now":1709379738,"version":"0.8.83","modules":"MDH-de","build":"5ebfe5a","env":"esp8266-de","menu_prot":false,"menu_mask":61,"menu_protEn":false,"cst_lnk":"","cst_lnk_txt":"","region":0,"timezone":1,"esp_type":"ESP8266"},"system":{"ssid":"Friekaysee5","ap_pwd":"esp_8266","hidd":false,"device_name":"AHOY-DTU","dark_mode":false,"sched_reboot":false,"mac":"08:3A:8D:D3:98:A7","hostname":"AHOY-DTU","pwd_set":false,"prot_mask":61,"sdk":"2.2.2-dev(38a443e)","cpu_freq":80,"heap_free":13512,"sketch_total":2510848,"sketch_used":616,"wifi_rssi":-57,"ts_uptime":411,"ts_now":1709379738,"version":"0.8.83","modules":"MDH-de","build":"5ebfe5a","env":"esp8266-de","menu_prot":false,"menu_mask":61,"menu_protEn":false,"cst_lnk":"","cst_lnk_txt":"","region":0,"timezone":1,"esp_type":"ESP8266","radioNrf":{"en":true,"isconnected":true,"dataRate":2,"sn":"80739847","irqOk":2},"mqtt":{"enabled":false,"connected":false,"tx_cnt":0,"rx_cnt":0,"interval":0},"heap_frag":2,"max_free_blk":13256,"core_version":"3.1.2","flash_size":4096,"reboot_reason":"Software/System restart","schMax":10},"mqtt":{"broker":"","clientId":"","port":"1883","user":"","pwd":"","topic":"inverter","interval":"0"},"ntp":{"addr":"pool.ntp.org","port":"123","interval":"720"},"sun":{"lat":"53.61197","lon":"10.13615","offsSr":600,"offsSs":600},"pinout":{"cs":15,"ce":0,"irq":2,"sclk":0,"mosi":0,"miso":0,"led0":255,"led1":255,"led2":255,"led_high_active":false,"led_lum":255},"radioNrf":{"en":true,"isconnected":true,"dataRate":2,"sn":"80739847","irqOk":2},"serial":{"show_live_data":true,"debug":true,"priv":false,"wholeTrace":true,"log2mqtt":false},"static_ip":{"ip":"","mask":"","dns1":"","dns2":"","gateway":""},"display":{"disp_typ":0,"disp_pwr":false,"disp_screensaver":0,"disp_rot":0,"disp_cont":140,"disp_graph_ratio":0,"disp_graph_size":2,"disp_clk":255,"disp_data":255,"disp_cs":255,"disp_dc":255,"disp_rst":255,"disp_bsy":255,"pir_pin":255}}

0.8.83 - 5ebfe5a
---------------
12:39:05.000 I: (#0) Radio infos: -2 -4 -4 -2 -2 | t: 7, s: 0, f: 0, n: 7 | p: 0
12:39:05.003 I: (#0) TX 27 CH61, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 0f d9 00 00 00 00 00 00 00 00 64 b2 57 
12:39:05.357 I: (#0) request timeout: 350ms
12:39:05.358 -----
12:39:05.358 I: com loop duration: 358ms
12:39:05.358 -----
12:39:35.000 I: (#0) Radio infos: -2 -4 -4 -4 -2 | t: 8, s: 0, f: 0, n: 8 | p: 0
12:39:35.003 I: (#0) TX 27 CH75, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 0f f7 00 00 00 00 00 00 00 00 c5 67 0d 
12:39:35.357 I: (#0) request timeout: 350ms
12:39:35.358 -----
12:39:35.358 I: com loop duration: 358ms
12:39:35.358 -----
12:40:05.000 I: (#0) Radio infos: -2 -4 -4 -4 -4 | t: 9, s: 0, f: 0, n: 9 | p: 0
12:40:05.003 I: (#0) TX 27 CH03, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 15 00 00 00 00 00 00 00 00 c4 86 10 
12:40:05.357 I: (#0) request timeout: 350ms
12:40:05.358 -----
12:40:05.358 I: com loop duration: 358ms
12:40:05.358 -----
12:40:35.000 I: (#0) Radio infos: -4 -4 -4 -4 -4 | t: 10, s: 0, f: 0, n: 10 | p: 0
12:40:35.003 I: (#0) TX 27 CH23, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 33 00 00 00 00 00 00 00 00 a5 34 e5 
12:40:35.357 I: (#0) request timeout: 350ms
12:40:35.358 -----
12:40:35.358 I: com loop duration: 358ms
12:40:35.358 -----
12:41:05.000 I: (#0) Radio infos: -4 -6 -4 -4 -4 | t: 11, s: 0, f: 0, n: 11 | p: 0
12:41:05.003 I: (#0) TX 27 CH40, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 51 00 00 00 00 00 00 00 00 c7 85 54 
12:41:05.357 I: (#0) request timeout: 350ms
12:41:05.358 -----
12:41:05.358 I: com loop duration: 358ms
12:41:05.358 -----
12:41:35.000 I: (#0) Radio infos: -4 -6 -6 -4 -4 | t: 12, s: 0, f: 0, n: 12 | p: 0
12:41:35.003 I: (#0) TX 27 CH61, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 6f 00 00 00 00 00 00 00 00 a6 9d 13 
12:41:35.357 I: (#0) request timeout: 350ms
12:41:35.358 -----
12:41:35.358 I: com loop duration: 358ms
12:41:35.358 -----
12:42:05.000 I: (#0) Radio infos: -4 -6 -6 -6 -4 | t: 13, s: 0, f: 0, n: 13 | p: 0
12:42:05.003 I: (#0) TX 27 CH75, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 8d 00 00 00 00 00 00 00 00 02 4d 85 
12:42:05.357 I: (#0) request timeout: 350ms
12:42:05.358 -----
12:42:05.358 I: com loop duration: 358ms
12:42:05.358 -----
12:42:35.000 I: (#0) Radio infos: -4 -6 -6 -6 -6 | t: 14, s: 0, f: 0, n: 14 | p: 0
12:42:35.003 I: (#0) TX 27 CH03, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 ab 00 00 00 00 00 00 00 00 63 ff 70 
12:42:35.357 I: (#0) request timeout: 350ms
12:42:35.358 -----
12:42:35.358 I: com loop duration: 358ms
12:42:35.358 -----
12:43:05.000 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 15, s: 0, f: 0, n: 15 | p: 0
12:43:05.003 I: (#0) TX 27 CH23, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 c9 00 00 00 00 00 00 00 00 01 4e c1 
12:43:05.357 I: (#0) request timeout: 350ms
12:43:05.358 -----
12:43:05.358 I: com loop duration: 358ms
12:43:05.358 -----
12:43:35.000 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 16, s: 0, f: 0, n: 16 | p: 0
12:43:35.003 I: (#0) TX 27 CH40, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 10 e7 00 00 00 00 00 00 00 00 a0 9b 9b 
12:43:35.357 I: (#0) request timeout: 350ms
12:43:35.358 -----
12:43:35.358 I: com loop duration: 358ms
12:43:35.358 -----
12:44:05.000 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 17, s: 0, f: 0, n: 17 | p: 0
12:44:05.003 I: (#0) TX 27 CH61, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 11 05 00 00 00 00 00 00 00 00 c1 1a 98 
12:44:05.357 I: (#0) request timeout: 350ms
12:44:05.358 -----
12:44:05.358 I: com loop duration: 358ms
12:44:05.358 -----
12:44:35.000 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 18, s: 0, f: 0, n: 18 | p: 0
12:44:35.003 I: (#0) TX 27 CH75, 15 ret. | 15 82 99 00 78 80 73 98 47 80 0b 00 65 e3 11 23 00 00 00 00 00 00 00 00 a0 a8 6d 
12:44:35.357 I: (#0) request timeout: 350ms
12:44:35.358 -----
12:44:35.358 I: com loop duration: 358ms
12:44:35.358 -----
```

### Error description

Ich musste unser Wlan neu aufsetzen und habe wohl in dem Zuge auch die DTU einmal aktualisiert. Verbindung und Einrichtung hat wieder gut funktioniert, nur leider findet die DTU den Inverter nicht mehr. Serial wurde mehrere Male kontrolliert und ich habe auch schon alle noch zur Verfügung stehenden Software Versionen installiert und probiert. Alles leider ohne Erfolg. Die Position der DTU habe ich nicht verändert. Die Sendeleistung habe ich per Standard auf "MIN" stehen, habe aber auch schon Low, High, Max ausprobiert. 

Der Wechselrichter an sich funktioniert, da ich die Produktion auch noch über einen Shelly verfolgen kann. 

Anbei noch ein paar Bilder von meiner DTU. 

Über jegliche Hilfestellung wäre ich sehr dankbar. 
[image]
[image]
[image]
[image]
[image]

tictrick, 2024-03-02:
Hallo,

auf dem letzten Bild sieht es so aus, als ob die Lötstellen zum nRF-Modul nicht oder nicht richtig verlötet wären. Kontrolliere doch bitte mal alle Lötstellen.

MetaChuh, 2024-03-02:
@Jakmanny 

danke dir für die fotos, super hilfreich für's fehlersuchen/finden 👍

wenn du zb. von ahoydtu v0.6.9 upgegradet hast:

kontrolliere deine pin settings.
bei deinem nrf24+ modul muss eingestellt sein:
CE		D4
IRQ		D3

seit v0.7.36 wurden diese beiden pins in den default einstellungen getauscht.

Jakmanny, 2024-03-02:
Hey zusammen,

vielen Dank für euren super schnellen Einsatz und Hilfestellung bei meinem, für mich undurchsichtigen, Problem.

@MetaChuh du hast den Fehler gefunden! Einstellungen angepasst und schon läuft die Anlage wieder :-)

Ich weiß gar nicht wie lange ich daran gefummelt habe und Ihr habt es so schnell gelöst. 

Vielen Vielen Dank !!!!
[image]

closed 2024-03-02 by Jakmanny