lumapu/ahoy #1470: [Bug]Anzeige der Alarme so korrekt?

Gubi2023, 2024-03-02:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.88

### Github Hash

2bc6610

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

V08.88

### Error description

Ist die Alarmanzeige korrekt?
bei mir werden die Alarme doppelt und 3-fach angezeigt und die Reihenfolge ist verwirrend (nicht chronologisch)

[image]

knickohr, 2024-03-03:
Das ist weder ein Bug noch ein Feature, das kommt leider genau so aus dem WR raus. Wenn man das zusammen ziehen wollte, müßte man alle Meldungen speichern und dann die Frage wie lange ?

Zordandi, 2024-03-03:
> Das ist weder ein Bug noch ein Feature, das kommt leider genau so aus dem WR raus. Wenn man das zusammen ziehen wollte, müßte man alle Meldungen speichern und dann die Frage wie lange ?

Werden im WR eigentlich nur 10 Fehlermeldungen gespeichert oder wurde die Begrenzung auf 10 wegen dem Speicherverbrauch im ESP gemacht?

knickohr, 2024-03-03:
Nein, im WR können mehr sein. Wieviele weiß ich nicht.

lumapu, 2024-03-03:
die Grenze in Wechselrichter kennen wir bis dato nicht, ist aber auch nicht wichtig, da wird zyklisch die neuesten Meldungen abholen und das sich in Grenzen hält.
Die Überlegung wäre die Limitierung von 10 Meldungen beim ESP32 zu erhöhen.

Gubi2023, 2024-03-03:
meine Frage wäre trotzdem, wieso die Reihenfolge in der Darstellung nicht chronologisch ist, ist das ein FIFO oder wie werden sie angezeigt?

lumapu, 2024-03-03:
frage bei Hoymiles nach, wie geschrieben geben wir nur das wieder was der Wechselrichter uns an Infos bereitstellt.

Zordandi, 2024-03-03:
> meine Frage wäre trotzdem, wieso die Reihenfolge in der Darstellung nicht chronologisch ist, ist das ein FIFO oder wie werden sie angezeigt?

Die 10 Einträge sind hardcoded und aktuell als Ringpuffer implementiert. Ja, wäre durchaus überlegenswert das zu ändern und chronologisch anzuzeigen.

lumapu, 2024-03-03:
ich prüfe was möglich ist, schön wäre es, wenn dadruch die Code-Size nicht unverhältmismäßig steigt.

Zordandi, 2024-03-03:
> ich prüfe was möglich ist, schön wäre es, wenn dadruch die Code-Size nicht unverhältmismäßig steigt.

Meine letzte Programmiererfahrung ist lange her, aber ich glaube es ist alles dafür bereits vorhanden ohne extra sortieren zu müssen. Mit mAlarmNxtWrPos und  alarmCnt und den fixen 10 (ich würde hier eine Konstante einführen statt hardcoden) sollte man in der Webausgabe die richtige Startposition im Array berechnen können. Quasi ein FifoBuffer realisiert mit Ringpuffer und Zeiger.

lumapu, 2024-03-03:
die Frage ist, ob der Wechselrichter die Reihenfolge einhält, aber ja den Zeiger haben wir schon, den müsste man nur noch über die API verfügbar machen - danke für den Hinweis

Gubi2023, 2024-08-14:
Alarmanzeige jetzt ok, über eine Erweiterung  mit mehr Einträgen würde ich mich freuen! (in Whishlist aufnehmen?)

knickohr, 2024-08-14:
Das Problem ist das diese Alarme zwischengespeichert werden müssen um sie zeitlich zu sortieren und auf Anfang und Endezeit zu warte.

lumapu, 2024-08-14:
ich glaube er meint diesen Zwischenspeicher, den können wir auf alle Fälle anheben

Gubi2023, 2024-09-10:
noch nicht getestet, da z.Zt. nicht so viele Meldungen auflaufen 😉

Gubi2023, 2024-09-23:
habe jetzt 15 Meldungen angezeigt, allerdings andere als in alter Version mit 10 Meld. Was stimmt jetzt?

[image]
[image]

lumapu, 2024-09-24:
die Meldungen starten jeden Tag aufs neue

Gubi2023, 2024-09-24:
weiss ich doch, sind aber vom gleichen Tag!

lumapu, 2024-09-24:
hab gerade nochmal deine Screenshots angeschaut. Die Alarme sind doch die gleichen, nur in der neueren Version sind es mehr (alle) und sortiert.

Gubi2023, 2024-09-24:
einer fehlt, das wäre der 16. Deswegen meine Frage  
[image]

knickohr, 2024-09-24:
Na schieb das Ganze doch mal nach oben.

Gubi2023, 2024-09-24:
hab ich versucht, ging aber nicht...🤷‍♂️

stefan123t, 2024-10-30:
@Zordandi ich glaube der WR kann zwei Mal zehn Events speichern. Wir fragen aber m.W. nur die ersten zehn ab.
Man kann glaube auch ein paar AlarmIDs als gelesen markieren bzw. erst ab einer bestimmten ID die Alarme auslesen.

Das ist aber m.W. bisher noch nicht implementiert, schau mal hier: 
* [DEVINFO_ALL (0x15)](https://github.com/lumapu/ahoy/wiki/Protocol#welche-device-info-req_arw_dat_all-0x15-sub-kommandos-gibt-es) [Alarm Data (0x11) / Alarm Update (0x12)](https://github.com/lumapu/ahoy/wiki/Protocol#wie-sieht-das-device-info-kommando-alarm-data-0x11--alarm-update-0x12-aus-)
* [DEVCONTROL_ALL (0x51)](https://github.com/lumapu/ahoy/wiki/Protocol#welche-device-control-devcontrol_all-0x51-sub-kommandos-gibt-es) CleanState_LockAndAlarm (0x14)

Wir könnten auch @broth-itk bitten ob er die Events mal mit der DTU auslesen und die Kommandos tracen kann ?
Dann hätten wir aktuelle Traces mit den ganzen Befehlen wie sie die Hoymiles DTU verwendet.

knickohr, 2024-10-30:
Ahoy sammelt jetzt bis zu 50 Events. Was und wieviel allerdings im Inverter „drin“ ist, weiß ich nicht.

Gubi2023, 2024-11-26:
soeben getestet mit V0.8.152:   21 Ereignisse; wurden alle richtig angezeigt. Issue wird geschlossen

closed 2024-11-26 by Gubi2023