lumapu/ahoy #1471: Fehlermeldungen falls sinnvoll zusammenfassen

Zordandi, 2024-03-03:
## Hardware
 * [X] ESP8266

### nRF24L01+ Module
* [X] nRF24L01+ 

### Antenna:
* [X] circuit board

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 

### Version / Git SHA: 
Version: 0.8.83
Github Hash: [5ebfe5a](https://github.com/lumapu/ahoy/commit/5ebfe5a8fb5bcc833b64a0dd083686031808cec8)

### Build & Flash Method:
* [X] AhoyDTU Webinstaller

### Debugging:
* [X] Setup settings (use our templates ... to be added)

[image]

Mein Aufbau: 4x385WP an einem HM-1500, wobei 2 Module direkt am MPP-A und 2 Module über einen Anker Solix e1600 am MPP-B hängen. Das führt dazu, dass in der Früh der MPP-A früher startet und der MPP-B zuerst auf den Anker Solix "warten" muss.

Bei den Fehlermeldungen führt das in der Anzeige dazu, dass zuerst eine Start Fehlermeldung mit der Ende Zeit 01:00:00 kommt und dann später irgendwann eine zweite Fehlermeldung, wo der Start Wert identisch und der Ende Wert dann korrigiert angezeigt wird.

Macht es eventuell Sinn, wenn man die beiden Fehlermeldungen zusammenfassen und nur eine Zeile darstellen würde? z.B. indem man die Zeile mit dem "falschen" Ende Wert 01:00:00 aktualisiert?

Gubi2023, 2024-03-03:
meine Rede.... #1470

knickohr, 2024-03-03:
Das ist weder ein Bug noch ein Feature, das kommt leider genau so aus dem WR raus. Wenn man das zusammen ziehen wollte, müßte man alle Meldungen speichern und dann die Frage wie lange ?

DanielR92, 2024-06-01:
Machbar ist hier alles. Keine Frage...
ein kleiner Workaround, die Daten in eine Tabelle ersetzen und diese mit einem Filter/Sort versehen. 
Je nach Filter/Sort Einstellungen kann man diese Speichern und beim nächsten mal wieder anwenden.
ZB: https://www.w3schools.com/howto/howto_js_sort_table.asp

lumapu, 2024-07-11:
seit gestern ist in der dev `0.8.128` eine Zusammenfassung der Alarme enthalten

Zordandi, 2024-07-12:
> seit gestern ist in der dev `0.8.128` eine Zusammenfassung der Alarme enthalten

Danke! Hab es gleich heute in der Früh ausprobiert. Statt der sonst "üblichen" 12 Meldungen kamen sie jetzt in den 7 sinnvollen zusammengefasst. Sieht schonmal sehr gut aus! Werde weiter berichten.

[image]

stefan123t, 2024-10-30:
@Zordandi kann der Feature Request jetzt geschlossen werden ?