lumapu/ahoy #1472: [Bug]

c-graeser, 2024-03-03:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Settings/Protection

### Error description

The settings under  "Settings -> Protection" are restored to be protected after reboot.
Changing all settings to "unprotected" (no checkmark) and storing the values without reboot is OK. The values are stored.
After reboot the checkmarks are back and the protection is active again.

lumapu, 2024-03-03:
the protection is only verified by password. Remove / clear password to disable protection

stefan123t, 2024-10-30:
@lumapu this is similar to the discussion we had on the DHCP vs. Static IP settings.
Maybe it would be advisable from UX perspective to add a boolean toggle switch / button to `enable` / `disable` each of these two features ?