lumapu/ahoy #1473: REST API control error "ERR_PROTECTED"

c-graeser, 2024-03-03:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

4 Inverter in operation with older versions of AhoyDTU sucessfully

### Error description

The release notes of 0.8.83 mention:
"added API-Token authentification for external scripts"
1. It seems to be not possible to deactivate authentication for the REST-API
Trying to change the power limits of an inverter with the curl sentence:
curl -i -H "Accept: application/json" -H "Content-Type:application/json" -X POST --data '{"id": 0, "cmd": "limit_nonpersistent_absolute", "val": 1500}' http://192.168.2.12/api/ctrl
 results to the curl answer: {"success":false,"error":"ERR_PROTECTED"}
2. There seem to be no documentation regarding acessing the REST-API by curl

MetaChuh, 2024-03-03:
@c-graeser 

please try: `curl --header "Content-Type:application/json" --request POST --data '{"auth": "your_password", "cmd": "limit_nonpersistent_absolute", "id": 0, "val": 1500}' "http://192.168.2.12/api/ctrl"`

to point 2.:
sorry about that, it's posted within the issues, but as it is a private, non funded project, there's not a lot of manpower left, between job, kids, friends and github.

i agree with you, it would have been better to add usage examples to the release notes of v0.8.83, retrospectively seen.

greetings,
metachuh

c-graeser, 2024-03-04:
Hello Metachuh,

 

thank you very, very much for your quick response!

Your solution worked perfectly.

Spending a sunny afternoon in front of my computer trying to reverse engineer the authentication lead me to the need to contact you.

 

I am sure a bit more documentation will take load away from you. You then don’t have to anwer silly questions like mine 😉

 

Best regards

Christian

MetaChuh, 2024-03-04:
@c-graeser 

i'm glad that it works 👍
ps: no silly question at all.
this is the base of re-documenting later,
and helps others for now, if they search for `ERR_PROTECTED`, `auth`, `authentification` or similar.

if you're satisfied, please leave a thumbs up and close this issue.

greetings,
metachuh

closed 2024-03-04 by c-graeser