lumapu/ahoy #1476: [Bug]MQTT not working, interruptpin "false" and no connection to HM-800

americanium, 2024-03-04:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36 and 0.8.83

### Github Hash

Whats that?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Darkmode selected
Fixed IP Adress: 192.168.1.8
Subnetmask: 255.255.255.0
SSID and Password entered.

Inverter serial added and Max Module Power 500 both + Modulname added

MQTT server information added.

All settings were automatically applied from the working Firmwareversion 0.5.66 before!
(Unfortunately I can't find the old firmware to flash it back)

### Debug Serial Log output

```shell
13:02:18 I: (#0) resetPayload
13:02:18 I: (#0) enqueCommand: 0x01
13:02:18 I: (#0) enqueCommand: 0x0b
13:02:18 I: (#0) prepareDevInformCmd 0x01
13:02:18 I: TX 27B Ch40 | 15 80 11 40 17 81 54 27 70 80 01 00 65 e5 b8 4a 00 00 00 00 00 00 00 00 dd a8 d7 
13:02:48 I: (#0) resetPayload
13:02:48 I: (#0) prepareDevInformCmd 0x0b
13:02:48 I: TX 27B Ch61 | 15 80 11 40 17 81 54 27 70 80 0b 00 65 e5 b8 68 00 00 00 00 00 00 00 00 76 22 de
```

### Error description

1.) First of all I used the 0.5.66 version for about 1 year without any problems, after that time i wanted to get the newest firmware
2.) When installing 0.7.36 MQTT works but connection to inverter fails
3.) When installing 0.8.83 (or one of the developer files) neither MQTT works nor connection to HM-800. Interrupt pin "false". 
4.) With 0.5.66 the Send/Receive LED was blinking, now it doesn't blink any more.

americanium, 2024-03-04:
Additional information:
Webserial Output with 0.8.89:

13:29:42.352 -----
13:29:42.353 I: com loop duration: 353ms
13:29:42.353 -----
13:29:57.352 -----
13:29:57.353 I: com loop duration: 353ms
13:29:57.353 -----
13:30:12.352 -----
13:30:12.353 I: com loop duration: 353ms
13:30:12.353 -----
13:30:27.352 -----
13:30:27.353 I: com loop duration: 353ms
13:30:27.353 -----
13:30:42.352 -----
13:30:42.353 I: com loop duration: 353ms
13:30:42.353 -----
13:30:57.352 -----
13:30:57.353 I: com loop duration: 353ms
13:30:57.353 -----
13:31:12.352 -----
13:31:12.353 I: com loop duration: 353ms
13:31:12.353 -----

MetaChuh, 2024-03-04:
@americanium 

please check your nrf24+ pin settings.

as you came from v0.5.66, your nrf24+ module settings for CE and IRQ have to be set to:
CE D4
IRQ D3

info about why: since v0.7.36 those pins were swapped within the default settings.

Peertje-62, 2024-03-04:
Hi,  if you want to go back
5.66 stil exist to download
https://github.com/lumapu/ahoy/releases
[image]

americanium, 2024-03-04:
Thank you very much!
Connection to HM-800 works now - then the MQTT problem still remains ...

There's no kind of FAQ where this "change" is described for other people who change to the newest firmware ?

Thanks @Peertje-62 : Now the connection works so i would like to stay with the newest firmware as long as I can fix the MQTT issue.

Peertje-62, 2024-03-04:
Yes 
there is a change log
(Rather Big)
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

Peertje-62, 2024-03-04:
- Never change a running system (Deutsche Bahn win 3.11)
- if it is not broken you cannot fix it
- if you can fix it whit a hammer is not electrical
have a nice day

Peertje-62, 2024-03-04:
and,
I would advise to upgrade your hardware ESP 32
runs smoother & better

MetaChuh, 2024-03-04:
@americanium 

regarding mqtt: please download and update to the latest developer firmware v0.8.89 for esp8266 from here: [`ahoy_v0.8.89`](https://fw.ahoydtu.de/fw/dev%2Fahoy_v0.8.89/)

the mqtt issues you encounter will most likely dissapear.
if not, post all your findings here, before you revert to v0.5.66 with the link given by @Peertje-62 

good luck and keep us informed.

americanium, 2024-03-04:
So just did the update to 0.8.89 (again, already tried this version) and also deleted all configuration settings to reconfigure it.

When using the newest dev version 
NRF24L01 = connected
Interrupt Pin working = true
MqTT connected = false

No more connection to HM-800

CS = D8
CE= D4
IRQ = D3

WebSerial Output:
15:06:28.352 -----
15:06:28.353 I: com loop duration: 353ms
15:06:28.353 -----
15:06:43.352 -----
15:06:43.353 I: com loop duration: 353ms
15:06:43.353 -----

MetaChuh, 2024-03-04:
at least you got it half ways working for now ... big step from 0.5.66 to 0.8.83/89 with nothing in between👍

i guess many do not notice all the ammount of major changes in between, as they either update and adapt regularly, or don't update at all.

nice if you could help us out with a how to of all steps necessary to get your specific setup working after the leap update.
might help others.
please keep us informed.

thx & greetings

americanium, 2024-03-04:
Sure.

So due to the "try and error" situation I think the final solutions were:
**1.) To get MQTT working: Define a DNS1 and DNS2, but maybe that's depending on your network situation. So I have only a standard 5G router with DHCP - nothing special. 
2.) To get connection to Hoymiles working: Set the Radio Power Level to "high". All other settings didn't work. Also swap pinout CE to D4 and IRQ to D3.** 

You're right, when I did my initial setup of ahoy-dtu 1.5years ago it worked for me, not 100% stable but it was "ok". Now I saw the there are many many updates and new revisions so I wanted to upgrade and also just for interest what's new or for finding new future projects ^^.

Overall - thank you very much - hope others who get into the same situation will have a benefit of this thread :-)

lumapu, 2024-03-04:
I really appreciate how fast and good the answers were to resolve this issue / questions. 😊
I'm really happy to have such a good team here.

Yesterday I started creating a documentation, for the beginning in German only: https://ahoydtu.readthedocs.io

cassidea, 2024-03-09:
> Thanks - I know the changelog but i honestly didn't ready the information of EVERY change since 0.5.66 :-) but it seems that I'm the only one with this issue until now or most other didn't upgrade :D
You are not the only one. I am still on 0.6.9 because I am not able to get newer versions running.

For example I just* learned, that connection to D3 (GPIO0) changes the mode on reset on ESP8266 NodeMCU. In my case this leads to a bootloop. I need to disconnect D3 to be able to configure everything.

*https://github.com/espressif/esptool/issues/719#issuecomment-1032309752

MetaChuh, 2024-03-09:
@americanium 

if this issue has been solved for you, please close it.

thx and greetings,
metachuh

closed 2024-03-11 by americanium