lumapu/ahoy #1477: [Bug]

grericht, 2024-03-04:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

none

### Error description

i have this board: https://forum.arduino.cc/t/esp32-c3-supermini-pinout/1189850

I can flash it with ESPHome without any problems. From Home Assistant and from the WebInstaller.
When I try to flash AhoyDTU I can access the console. When I click on install, only the message "Preparing installation" appears for about 60 seconds, then "Failed to initialize." Pressing the buttons doesn't help here either.

With ESP Home I don't have to press anything. it simply starts the installation.

When I got AhoyDTU installed (I think via ota from esp home web server) the board doesn't start properly. It only starts when I connect it to the laptop via USB and press console in the web installer. Sometimes I have to press reset in the console

lumapu, 2024-03-04:
I think our websinstaller was not tested with this kind of ESP. Thank you for reporting that. Did I understand correctly that you facing the issues on https://ahoydtu.de/web_install/

grericht, 2024-03-05:
yes. im facing on https://ahoydtu.de/web_install/

lumapu, 2024-03-06:
let me first try to do some things, after that it would be cool if you can test again

grericht, 2024-03-10:
I don't want to push! I just want to know what/how long should I wait? for the next release? I just want to avoid missing anything

lumapu, 2024-03-10:
thank you for your reminder - I will try now

lumapu, 2024-03-10:
added the ESP32-C3 in a few minutes it can be tested 😊

grericht, 2024-03-10:
up to now web installer still not working

lumapu, 2024-03-10:
same error message? Can you try to clear browser cache before testing? You mentioned that there is another page which supports this ESP. Can you share a link to that page?

grericht, 2024-03-10:
i tested again with cleared browser cache in chrome. dont work.
https://esphome.github.io/esp-web-tools/ works perfect. i dont have to hit buttons or something else. connect. choose com port. install -> erase -> install -> install complete

at https://ahoydtu.de/web_install/
connect. choose com port. install. choose erase or not. next -> prepairing to install (spinning circle but nothing else happens) -> 1min later: "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."

grericht, 2024-03-13:
again: i don't want to push!
is there a chance for a solution? can i help?

lumapu, 2024-03-14:
Let me check again, I had no time to look into that.
I'll let you know once you can test again

grericht, 2024-03-18:
thank u. then i will wait for ur comment

grericht, 2024-04-04:
hey. is there something new? can i hope that i can use my esp32c3supermini?

grericht, 2024-08-19:
there will be now support for this chip comming?

stefan123t, 2024-10-30:
@MetaChuh can you possibly look into this, according to lumapu you have your own ESP WebTools online ?

@grericht can you compile / install using esptools from development03 branch using Version 0.8.152 ?

grericht, 2024-11-09:
https://web.esphome.io/

esp32c3 with usb connected

241007_ahoy_0.8.152_78cbd77_esp32-c3-mini-de.bin selected

install successfully

it reboots all the time