lumapu/ahoy #1479: language

jacques3629, 2024-03-05:
Good morning

Sorry I'm going through a translator

  Thanks for the software I use it every day
I find it exceptional however
Just a question
When I download the version
Install on my esp32
The version is in English
When I did an update
The version is in German
How to keep the English version
Do you plan in the future to change the language directly selectable on the configuration page

THANKS

Peertje-62, 2024-03-05:
Hi, 
F240304_ahoy_0.8.90_e5c0e8e_esp32-wroom32.bin = English
de is German
[image]
have a nice day

Peertje-62, 2024-03-05:
Don't install DEV ahoy_0.8.90 has a few bugs

Gubi2023, 2024-03-05:
see also #1480

jacques3629, 2024-03-05:
Merci à tous 

j'ai réinstallé la version  0.8.83 qui fonctionne très bien 
pour garder la version english  je dois mettre à jour le firmware que sur les versions realease 

Thank you all

I reinstalled version 0.8.83 which works very well
to keep the english version I have to update the firmware only on the realease versions

Peertje-62, 2024-03-05:
Hi, (I am Dutch)
The Development variant is also in English available.

Salut, (je suis néerlandais)
La variante Développement est également disponible en anglais

https://fw.ahoydtu.de/fw/
regards

jacques3629, 2024-03-06:
Hi, I'm French, I tried opendtu I didn't really like it, a lot of problems with my official dtu-pro gateway
on the other hand, all in French
Ahoy dtu is very good even better when it is multilingual

Sincerely

Peertje-62, 2024-03-06:
Hi,
You should use only one DTU at the time (not two together at the same time)
regards

lumapu, 2024-03-06:
hi you're welcome to translate Ahoy to French, there are only two files which need to be translated..
For now we only have English and German versions.

jacques3629, 2024-03-07:
@ lumapu   thank you what file, I don't know if I'm capable of it, I have no knowledge of programming
Thank you again for the essential software to fill the gaps in the official gateway which refreshes the data every 15 minutes

@Peertje_62  
ahoyddo you do 0 injection? the official gateway yes
using both I have the best of all worlds
monitoring every 5 s of 0 injection with the gateway hoymile

Peertje-62, 2024-03-07:
Hello Jacques,
I have 3 panels 385 watts, east, south and west, each with a 300 watt inverter, this way I cover my average power consumption (during the day) & Open Fusion DTU
regards

You69Man, 2024-03-08:
> hi you're welcome to translate Ahoy to French, there are only two files which need to be translated..
> For now we only have English and German versions.

Hi, I've already tried to do this with GPT, it's a task for a second, and the results looked very promising. Of course a review by a native french speaker would be sensitive afterwards just to be sure of the quality.

jacques3629, 2024-03-08:
@Peertje-62 hi, I have 21 PV 15 inverter, I manage to charge the EV
ahoy dtu manages well, open dtu is limited with so many inverters although it is also excellent
Sincerely

@You69Man gpt=chat gpt not available for me
gemini from Google tells me that the program is also available in French

Additional Information :

     AhoyDTU is manufactured by the Danish company Ahoy ApS.
     AhoyDTU is available in several languages, including French.   
     You can contact AhoyDTU support by email or telephone.

I hope this translation was useful to you.

lumapu, 2024-03-08:
@jacques3629 no Ahoy isn't available in French for now.
This is the main language file I mentioned before:
https://github.com/lumapu/ahoy/blob/main/src/web/lang.json

The second one can be found here:
https://github.com/lumapu/ahoy/blob/main/src/web/lang.h

You69Man, 2024-03-08:
Ceci est une première tentative de traduction d'Ahoy en français (site web et affichage). La traduction a été réalisée par GPT et n'est certainement pas parfaite. Elle devrait donc être contrôlée par un locuteur natif:
https://github.com/You69Man/ahoy/tree/feature/PR_display_i18n_new 

Dies ist ein erster Versuch Ahoy auf französisch zu übersetzen (Webseite sowie Display). Die Übersetzung stammt von GPT und ist sicher nicht perfekt. Sie sollte daher von einem Native Speaker kontrolliert werden:
https://github.com/You69Man/ahoy/tree/feature/PR_display_i18n_new

lumapu, 2024-03-09:
Datei auswählen kommt vom deinem Browser, das mit den Buttons muss ich mir mal näher anschauen

jacques3629, 2024-03-15:
hello 

hi, I added 4 panels + hms2000 inverter
with AHYO-DTU the whole works very well with NRF24L01 + CMT2300A

with the official dtu-pro gateway the HM inverter works very well in parallel with AHYO-DTU
I could do 0 injection while using AHYO-DTU + official DTU pro gateway with HM inverters

I received an official DTU WLITE S gateway for the 4 panels with the hms2000
I cannot get everything to work in parallel to make 0 injection

Does Ahoy dtu allow zero injection or will it allow it in the near future?

thank you for your work

lumapu, 2024-03-16:
The HMS series are supported in the same way as the HM series. The HMS series has one crucial difference:
Only one DTU can communicate at a time with the inverter. Maybe you get it working with two DTUs if the adjust the communication frequencies (can be done in inverter settings).

jacques3629, 2024-03-16:
thank you for the answer
how to know the frequency of the official hoymilles DTU WLITE S gateway which communicates with the HMS series
ps; I sent you a small contribution to your project, not much, just to encourage development 
software

jean-Jacques

lumapu, 2024-03-16:
As I remember correctly the original DTU communicates on 863.0 Mhz. If you first start the original DTU and then Ahoy you will only get data in Ahoy if you found the correct frequency.
(I never tried that and won't do 😊)

jacques3629, 2024-03-17:
> Si je me souviens bien, le DTU d'origine communique sur 863,0 Mhz. Si vous démarrez d'abord le DTU d'origine, puis Ahoy, vous n'obtiendrez des données dans Ahoy que si vous avez trouvé la bonne fréquence.. (Je n'ai jamais essayé ça et je ne le ferai pas 😊)

Thank you for this information, I will be able to do some tests
at hoymiles it becomes a bit random, I bought an official dtu-pro gateway for the HMs after new inverters for the hms
not compatible, I buy a dtu-s withe gateway for 4 hms panels, but the gateway does not manage power management and not compatible with HM inverters
fortunately software like Ahyo dtu, OPENDTU exists :)

stefan123t, 2024-10-30:
@jacques3629 did you check the latest version 0.8.152 and the translation to french locale as in #1255 and #1778 ?
There is a [zero-export branch](https://github.com/lumapu/ahoy/tree/zero-export) in the making which currently undergoes testing.

jacques3629, 2024-10-30:
Thanks for the software, I use the version
GIT SHA: 78cbd77 :: 0.8.152 <https://github.com/lumapu/ahoy/commits/78cbd77>

which gives me satisfaction, but things are changing, I am in the process
of changing equipment with a Deye hybrid inverter
I wish you much success in the software