lumapu/ahoy #1480: [Bug]0.8.90 minimal don´t work correctly

Gubi2023, 2024-03-05:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.90

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

installed 0.8.90-minimal

### Error description

latest version don`t work correctly at me. No settings possible

mellow5, 2024-03-05:
I can confirm on the ESP32 neither the minimal nor the standard build worked. Even had to flash via serial because update section wasn't working.

Gubi2023, 2024-03-06:
kaum gefunden, schon gefixt! Super!

closed 2024-03-06 by Gubi2023