lumapu/ahoy #1481: [Feture Request / Idea] Uptime and other debug data as entity

DB1BMN-collection, 2024-03-05:
Hi,
since we are still observing sporadic restarts I thought it would be a good idea to have the Uptime and other data as entity which could easilier be implemented in Home Assistant etc.

A collegue recommended me the program MQTT Explorer [1] which shows that already uptime, wifi_rssi and heap info are already sent via MQTT. That's fine!
[image]

Never the less Home Assistant by default can not display theese data, as they are not defined as an entity I guess.

I was fiddeling yesterday night in the configuration.yaml (which I am a completely noob at) and ended up with an diagram showing Uptime vs. RSSI.
[image]

Just paste the following code into the configuration.yaml:
`mqtt:
    #AhoyDTU
    sensor:
        - name: "AHOY Uptime"
            unique_id: "ahoy_uptime"
            state_topic: inverter/uptime
            unit_of_measurement: "s"
            icon: "mdi:clock"

        - name: "AHOY RSSI"
            unique_id: "ahoy_rssi"
            state_topic: inverter/wifi_rssi
            unit_of_measurement: "dBm"
            icon: "mdi:wifi"`

Take care of the indents and don't forget the dashes ("-") before the "name"-tags!

I think it would be much easier for debugging if anybody could track uptime, rssi and heap-info when they are available as entities and it would help cathc rare restart events.

What do you think?

[1] http://mqtt-explorer.com/

P.S. It seems the "code"-tag seems not to work correctly, WTF!?

MetaChuh, 2024-03-05:
hi @DB1BMN-collection 

uptime and reboot events are already available at ahoy dtu via api and mqtt.
could you help us out with pull requests for your desired changes that are missing for you from your forks, or possibly make pull requests or suggestion at home assistant ?
we'll review them and give you our feedback.

many thanks for your help.

stefan123t, 2024-10-30:
@DB1BMN-collection regarding the issue I would assume that the Home Assistant Auto-Discovery message either does not contain the two topics `inverter/uptime` and `inverter/wifi_rssi` yet or you would need to start HASS Auto-Discovery once more under **Settings > MQTT** Discovery Config (homeassistant) SEND.

@MetaChuh I do not know if he would be able to provide us a PR for the necessary changes.