lumapu/ahoy #1482: Bitte kurze Info zu Zero. Ich finde keine Anleitung und Funktionsbeschreibung

firedp, 2024-03-06:
Hallo,
leider stehe ich mit github immer auf Kriegsfuß.
Bitte kurze Info zur Zero-Version. Ich finde keine Anleitung und Funktionsbeschreibung oder Diskussion.
Wie funktioniert es, wie gut klappt es, was benötigt man an Hardware?
Oder ist das noch nichts für normale Ahoy-Benutzer?

Gruß

Peertje-62, 2024-03-06:
Hi, 
I am Dutch, 
I can read German,
Would like to help you if I can, but:
I don't understand your question
regards

tictrick, 2024-03-06:
Bitte noch etwas Geduld,

die -zero Variante braucht noch etwas Zeit.

Peertje-62, 2024-03-06:
Hi,
''Zero export'' found it on GitHub, nice!
Bye

firedp, 2024-03-06:
Kann mir jemand erklären, was es genau wird?
Ahoy benutze ich länger und auch meist in der neusten Version.
OpenDTU on Battery wird wohl auch für eine regulierbare Einspeisung benutzt, aber mit z.B. Hichi und Datenaustausch über das Wlan-Netz.
Was ist bei Ahoy Zero geplant?

lumapu, 2024-03-10:
naja ganz stimmts nicht: es gibt das opendtufusion board, das hat auch auf dem Ethernet header einen Modbus PHY. Es könnte jederzeit in Betrieb gehen - es fehlt noch der Code 😅

stefan123t, 2024-10-30:
@firedp @Peertje-62 the zero-export branch can be found in the github repository.

Short explanation for building your own firmware:
```bash
git clone https://github.com/lumapu/ahoy
git checkout zero-export
cd src
code .
```

@lumapu there is a Modbus TCP Implementation for OpenDTU by b0661 here: https://github.com/tbnobody/OpenDTU/pull/1893

finezzo, 2025-03-15:
Auf der Website steht ja etwas von einer Zero Export Funktion. Sie soll wohl in den Einstellungen verfügbar sein. Bei mir fehlt sie aber. Habe auch extra dafür einen ESP-32 S3 gekauft. Taucht trotzdem nicht auf.

https://docs.ahoydtu.de/de/latest/zeroExport.html

Gubi2023, 2025-03-16:
kommt auch erst mit V0.9.xxx!!!

finezzo, 2025-03-16:
> kommt auch erst mit V0.9.xxx!!!

Alles Klar. Danke für die Info. Das war auf der Seite nirgendwo ersichtlich. Aufgrund der detaillierten Beschreibung könnte man ja davon ausgehen, es sei bereits verfügbar.

Gubi2023, 2025-03-16:
https://discord.com/channels/984173303147155506/1029761098381017098/1331271981529825282