lumapu/ahoy #1484: Wochentag im Display auf Englisch

dixy245, 2024-03-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.91

### Github Hash

7b030a3

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Der Wochentag auf dem externen Display wird in Englisch angezeigt. Habe die deutsche Version installiert.

You69Man, 2024-03-07:
Siehe #1255

You69Man, 2024-03-08:
@lumapu
Ich weiß nicht, ob du's gesehen hast, aber ich habe in https://github.com/lumapu/ahoy/pull/1255 schon länger einen Link eingefügt für eine Übersetzung des Displays, nach der neuen build methode:
https://github.com/You69Man/ahoy/tree/feature/PR_display_i18n_new

MetaChuh, 2024-03-08:
@You69Man 
es war in punkto reliability abschreckend, dass du damals chatGPT erwähnt hast 😉

You69Man, 2024-03-08:
> @You69Man 
> es war in punkto reliability abschreckend, dass du damals chatGPT erwähnt hast 😉
> 

Warum? Ist die Übersetzung wirklich gar so schlimm? Ich hätte gehofft dass zumindest an die 90% passen sollte. Mein eigenes Französisch ist aber nur sehr rudimentär, ich kann zwar das meiste lesen und verstehen, aber im schreiben und sprechen bin ich nicht geübt.

MetaChuh, 2024-03-09:
> Warum? Ist die Übersetzung wirklich gar so schlimm? Ich hätte gehofft dass zumindest an die 90% passen sollte. Mein eigenes Französisch ist aber nur sehr rudimentär, ich kann zwar das meiste lesen und verstehen, aber im schreiben und sprechen bin ich nicht geübt.

hahaha, nö nö,
nur scherz, wegen manchen gpt basierten commits in anderen projekten, die zwar plausibel aussahen, jedoch beim screening eher in die kategorie fielen: "klingt gut, is' aber utterly bullshit in real life"

trifft jedoch für dein pr imho nicht zu, da das human sprachmodell gut funktioniert.
ist nur untergegangen, rest war typischer metachuh pseudo scherz

closed 2024-08-16 by lumapu