lumapu/ahoy #1485: Idee: automatische Berechnung des Zeitintervalls zur ganztäglichen Anzeige auf dem Display

dixy245, 2024-03-07:
Wäre es nicht möglich das Zeitintervall für die Aktualisierung der Werte aus der Differenz zwischen der Sonnenuntergangszeit und Sonbenaufganszeit zu berechnen, damit auf dem Display automatisch der ganze Tag angezeigt wird?

(SONNENUNTERGANSZEIT - SONNENAUFGANGSZEIT) : 255 = Zeitinterval  für die Aktualisuerung

ggf. könnte man hierzu für das Zeitintervall die "Auto(matik)" Einstellung für die Berechnung nach o.a. Formel hinzufügen.

You69Man, 2024-03-07:
>auf dem Display ganzjährig automatisch der ganze Tag angezeigt wird

Was meinst damit? Was genau fehlt dir am Display?
Die Zeitberechnung zwischen Sonnenaufgang und Untergang wird bereits gemacht weil sue z. B.  für die Skalierung des Diagramms benötigt wird, welches ja auch von Sommer bis Winter optimal dargestellt werden soll.

You69Man, 2024-03-07:
> D.h. auf dem OLED Display wird der Zeitrange den Sonnenstunden bereits angepasst?
Richtig.

Im Webdiagramm wird nicht skaliert. Dort ist meines Wissens aktuell noch ein Ringpuffer implementiert, der die Daten hinten wieder rausschiebt, wenn das Abtastintervall hoch ist, und die Daten nicht mehr im Puffer Platz haben.

lumapu, 2024-03-07:
Das Webdiagramm ist noch total Basic, ich hatte leider noch keine Zeit mich näher damit zu befassen. Die Grafik auf dem ext. Display ist Dank @You69Man tip top 👍

closed 2024-08-16 by lumapu