lumapu/ahoy #1486: Inverter #0: AhoyDTU is just not available

jtrouvain, 2024-03-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

was already installed

### Setup

No change on setup

### Debug Serial Log output

```shell
16:09:20.352 -----
16:09:20.353 I: com loop duration: 353ms
16:09:20.353 -----
16:09:35.352 -----
16:09:35.353 I: com loop duration: 353ms
16:09:35.353 -----
16:09:50.352 -----
16:09:50.353 I: com loop duration: 353ms
16:09:50.353 -----
```

### Error description

1. before with release 0.6.0 everything worked well 
2. Installed latest stable release
3. no data from inverter are received now
4. added inverter again, no change
5. erased all settings and did them new, no chnage

jtrouvain, 2024-03-07:
https://github.com/lumapu/ahoy/issues/1459#issuecomment-1974740839

seems to be the same issue, but I don't know how to work with Tasmotizer

MetaChuh, 2024-03-07:
@jtrouvain 

please check your ahoydtu pin settings.
as you are comming from v0.6.0, your nrf24+ module will need following settings, if wired the old way:

```
CE:  D4
IRQ: D3
```

the default value of those 2 pins have been swapped since v0.7.36

greetings
metachuh

jtrouvain, 2024-03-07:
@MetaChuh 
changes settings based on your recommendation. Interrupt PIN is now working, before it wasn't. But still no data from inverter.

lumapu, 2024-03-07:
as you already wrote: the old version was running fine. So I imagine that you know that the inverter only communicates if the inverter is producing (= day).

jtrouvain, 2024-03-08:
I know that the inverter only communicates during the day, but I was worried about the message that it is not available.
I checked this morning and it works now.
Thank you @MetaChuh @lumapu  for your support.
One final question: does that mean that I have to do this pin change everytime if I do an update in the future?

flevo1880, 2024-03-08:
Hi,
nach Erstinstallation und aktuelle FW  loop duration: 353ms

MetaChuh, 2024-03-08:
@jtrouvain 

we are glad you got the irq working 👍

> Thank you @MetaChuh @lumapu for your support.
One final question: does that mean that I have to do this pin change everytime if I do an update in the future?

you are safe if you make the next ota update.
you can, now make any ota update (or downgrade to any version from v0.7.36 or higher, if needed for testing) without having to do any further config changes.

relief note: there is no further pinout change planned or known as necessary for the esp8266 in conjunction with the nrf24+ module, so probably this will not change from now on for this platform.

closed 2024-08-16 by lumapu