lumapu/ahoy #1488: 5 Platinen zu verschenken

kiu77, 2024-03-07:
Ich hätte noch fünf Platinen für ESP32-38-Pin und NRF24L01+ zu verschenken.
Statt den auf dem ESP32-Board integrierten 3,3-V-Regler zu verwenden, ist
hier ein kleiner dreipolier SMD-Regler für 3,3V vorgesehen (1=GND, 2=Out, 3=In),
wodurch das Funkmodul optimal entkoppelt wird. Statt eines Elkos sind 3 kleine
SMD-Vielschicht-Kondensatoren mit 22µF/10V vorgesehen (SMD 1210 oder 1206),
was ebenfalls positiv für die Stabilität ist.
Bestückt werden muss die Platine mit einem 38-poligen ESP32-BOB (mit aufgedruckter
Antenne oder SMA-Anschluss) und einem Funkmodul. Hier gehen Versionen mit
aufgedruckter Antenne oder das von eByte mit Schraubanschluss für eine 2,4GHz-Antenne.
Die Schaltung ist simpel und nachfolgend abgebildet.
Wer zuerst kommt, mahlt zuerst. Anfragen also per PM.
Es wäre nett, wenn man dafür eine kleine Spende an lumapu vornimmt:
[https://www.paypal.com/paypalme/lupusch](url)
Die Briefmarke und den Briefumschlag spendiere ich.

[image]

[image]

kiu77, 2024-03-07:
Uff, jetzt merke ich grade, dass ich garnicht weiß, wie man in Github eine private Massage verschickt...

dknubben, 2024-03-07:
Ich benötige keine Platinen. Allerdings ist der Spendenaufruf (danke dafür) von @kiu77 für dieses Projekt und den Hauptentwicklicher ein Anliegen, das ich ausdrücklich unterstützen möchte und weitere Nutzer von Ahoy dazu aufrufe.

dixy245, 2024-03-07:
> So eine Möglichkeit @dixy245 : Ich habe temporär auf meinem NAS einen Ordner zur Ablage einer Textdatei freigegeben. Also Format = .txt Da bitte eine Datei mit Nickname und kompletter Anschrift ablegen.
> 
> Link: [https://quickconnect.to/tomesc](url) User: PCB PW: AhoyDTU

erledigt, vielen Dank. Der ESP32 liegt hier schon für seinen Einsatz bereit :-)

hartloeter, 2024-03-07:
Hallo, hast Du noch eine Platine. Ich würde eine nehmen
Gruß harti

kiu77, 2024-03-07:
3 sind weg, zwei noch da.

hartloeter, 2024-03-07:
Super, eine würde ich nehmen. Meine Daten habe ich hochgeladen

kiu77, 2024-03-07:
Du bist schon mitgerechnet @hartloeter

hartloeter, 2024-03-08:
Spende ist raus

VitBeck, 2024-03-09:
Hi. Dann nehme ich die letzte, falls noch verfügbar.
Spende geht ebenfalls raus. 
Danke

lumapu, 2024-03-09:
vielen Dank für die nette Idee, meine Frau hat heute Blumen bekommen 💐

kiu77, 2024-03-09:
@VitBeck 
Sodele, die letzte Platine ist raus...

@lumapu 
You're welcome

Der Thread kann jetzt zu.

closed 2024-03-09 by kiu77

VitBeck, 2024-03-10:
Die Spende ist raus. Danke

dixy245, 2024-03-12:
Angekommen, Spende geht heute raus, lieben Dank

kiu77, 2024-03-12:
Bei Fragen dann einfach melden...
Kontaktdaten hast Du ja.

hartloeter, 2024-03-12:
Brief ist angekommen, leider ohne Inhalt. Umschlag hat einen Riss-Schade.

[image]

kiu77, 2024-03-12:
Shit, da hat jemand geklaut...
Hast Du schon die Bauteile?

hartloeter, 2024-03-12:
ja habe ich schon
Pech gehabt

hartloeter, 2024-03-12:
Ich nehme halt eine Lochrasterplatine

kiu77, 2024-03-12:
Nein, ich würde Dir noch eine schicken...
Also 38-polig, nicht 30-polig?

kiu77, 2024-03-12:
Okay. Morgen geht nochmal ein Brief raus. Ich klebe die Platine dann innen fest.

hartloeter, 2024-03-12:
vielleicht leicht einwickeln

hartloeter, 2024-03-12:
Vielen Dank

hartloeter, 2024-03-12:
kann auch in  der Sortiermaschine passiert sein

kiu77, 2024-03-12:
Sortiermaschine glaub ich weniger. Die Platine wiegt 12g und war auf ne Visitenkarte von mir geklebt, damit Du bei Problemen meine Kontaktdaten hast. Jetzt hat sie jemand anderes ;-)
Aber ich machs morgen etwas anders...

hartloeter, 2024-03-12:
ok

kiu77, 2024-03-13:
Ich hatte beide. In der Praxis merke ich keinen Unterschied. Die machen ja bloß die Umwandlung USB/seriell und sind daher nur fürs flashen nötig.

kiu77, 2024-03-13:
@hartloeter 
So, die Ersatz-Platine ist eingetütet und in den Umschlag geklebt. Melde Dich mal kurz telefonisch oder per Mail, wenn sie angekommen ist.

hartloeter, 2024-03-13:
Danke mache ich klar

hartloeter, 2024-03-21:
@kiu77  Hallo, die Platine ist eingetroffen und funktioniert. Vielen Dank noch einmal für Deine Mühe

kiu77, 2024-03-21:
Super. Bei Fragen einfach anrufen...