lumapu/ahoy #1489: Nach update von 0.7.36 auf 0.8.83 keine Kommunikation mit Wechselrichter

MatthiasFrenzel, 2024-03-08:
Nach Update von 0.7.36 auf 0.8.83 funktioniert die Kommunikation zwischen DTU und WR nicht mehr.

An den Einstellungen habe ich nichts verändert. Nach dem letzten Update auf 0.7.36 musste ich die PIN Belegung ändern. Gab es hier wieder Änderungen?

Vielen Dank für jede Art von Hilfe
Matthias

## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

mroenne2022, 2024-03-08:
Die Belegung am NRF hat sich zwischenzeitlich geändert, deshalb muss man D3 und D4 tauschen (CE und IRQ) ! Kabel tauschen und in der Ahoy Software ebenfalls:
CE D4
IRQ D3
Dann läuft es wieder.

MatthiasFrenzel, 2024-03-08:
Leider nicht.

mroenne2022, 2024-03-08:
@MatthiasFrenzel 
Wie ist denn der Status des NRF Moduls , Konfiguriert und Connected ?

mellow5, 2024-03-08:
Ich hatte mal Probleme und glaube es war auch bei diesem Update, dass ich die Staus LED ausschalten musste, da diese auf einem verwendeten PIN lagen. Nur eine Vermutung.

[image]

MatthiasFrenzel, 2024-03-08:
Das Funkmodul scheint korrekt zu laufen.
[image]

LED Status ist aus.

WR habe ich auch schon neu gestartet. Aber keine Verbidung.

[image]

Gubi2023, 2024-03-08:
für mich sieht es aus, als würde die DTU den WR nicht sehen. Wie ist deine Sendeleistung eingestellt?
Siehe Screenshot

[image]

MatthiasFrenzel, 2024-03-08:
ich habe alle Einstellungen probiert.
Mit LOW hat das seit Montagen problemos funktioniert. Die Entfernung zwischen DTU und WR beträgt ca 10 m mit nur einer Wand dazwischen.

MetaChuh, 2024-03-08:
@MatthiasFrenzel 

probier mal ein ota oder reflash downgrade auf v0.7.36 und gib' uns morgen (bei sonne oder anderen dc input) bescheid.

wenn nach dem ursprungszustand wiederherstellen, bei dem alles funktioniert hatte, der kommunikationsfehler behoben ist, können wir uns an die mögliche fehlerursache in deinem spezifischen setup wenden.

wenn du, trotz ursprungszustand wiederherstellen, die gleichen kommunikationsprobleme zwischen dtu und wr hast, liegt evtl. das seltene szenario vor, dass update und hardware failure gleichzeitig aufgetreten sind.

bitte halte uns auf dem laufenden.

MatthiasFrenzel, 2024-03-09:
Ich habe jetzt die 0.7.36 wieder eingespielt. Die Kommunikation funktioniert auf Anhieb. Dann wieder die 0.8.83 --> keine Kommunikation.
Jetzt läuft erst mal wieder die alte Version.

MetaChuh, 2024-03-09:
@MatthiasFrenzel 

danke für dein feedback, und gottseidank funktioniert es bei dir wieder 👍

@lumapu : hast du ne' idee ?
das problem dieses issues, konnte ich mit einem frisch geflashten esp8266, per down und upgrade zwischen v0.7.36 und v0.8.83 bei mir nicht nachproduzieren.

lumapu, 2024-03-10:
nein, ich habe keine Idee. Interessant wäre ein ausführlicheres Log. Manchmal sind die Einstellungen auch verkorkst durch das fortfürhren von Verison zu Version - evtl. hilft hier ein kompletter Reset aller Einstellungen und anschließendem manuellen "Neuanlegen" aller Einstellungen. Man kann die Einstellungen aus der export-JSON einfach manuell wieder in das Webinterface übernehmen.

Gubi2023, 2024-03-13:
ich habe auch den Eindruck, dass sich die DTU (ESP8266) sporadisch aufhängt. Gestern war sie bei mir auch nicht mehr erreichbar, erst nach Spannungswiederkehr kamen wieder Daten. Seither läuft sie aber problemlos (V0.8.92)

Gubi2023, 2024-03-13:
@MetaChuh
[image]

serabe6240, 2024-03-21:
habe auch teilweise Probleme seit dem Update auf 0.8.83 mit der Kommunikation zu den WR. Erst ein Factory Reset und alles neu conf hat die Probleme behoben. Keine Ahnung warum. Bei den letzten Updates hatte ich keinerlei Probleme.

MatthiasFrenzel, 2024-03-29:
Ich habe jetzt noch mal das update auf die 0.8.83 durchgeführt. Dann alle Einstellungen gelöscht (factory reset) und alles neu eingetragen.
Leider ohne Erfolg. Es gibt keine Kommunikation zum WR und auch nicht zum MqTT Server. Wahrscheinlich weil es nichts zum Senden gibt?!

Auch NTP funktioniert nicht. 
Konsole: 09:12:50.530 I: [NTP]: getNtpTime failed

Die Zeit bekomme ich nur über den Browser synchronisiert.

[image]

MatthiasFrenzel, 2024-03-30:
Auch über Nacht durchlaufen zu lassen hat nichts ergeben. Die Konsole meldet:

 09:27:29.117 -----
09:27:37.743 I: [NTP]: getNtpTime failed
09:27:37.744 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1771, s: 0, f: 0, n: 1771 | p: 2
09:27:44.340 I: [NTP]: getNtpTime failed
09:27:44.342 I: (#0) TX 27 CH40, 15 ret. | 15  *  *  *  * 80 61 44 59 80 0b 00 66 07 cd 00 00 00 00 00 00 00 00 00 f0 a7 ca 
09:27:52.917 I: [NTP]: getNtpTime failed
09:27:59.537 I: [NTP]: getNtpTime failed
09:27:59.539 I: (#0) request timeout: 15193ms
09:27:59.539 -----

Kann damit jemand etwas anfangen, um der Ursache auf die Spur zu kommen?

Ich werde jetzt erst mal wieder die alte Version installieren.
Frohe Ostern :-)

DerBorzi, 2024-06-26:
Bei mir genau das gleiche Problem, gibts es hier schon eine Lösung?

MatthiasFrenzel, 2024-06-26:
leider nein

pwsurf, 2024-06-27:
Bei mir grundsätzlich das gleiche Problem nach upgrade auf die deutsche esp8266 Version. 
Allerdings kein ntp Problem und auch kein mqtt Problem. Lediglich mein "Wechselrichter HM1500 ist gerade nicht verfügbar".
Nach downgrade geht alles wieder.

DerBorzi, 2024-06-27:
Bei mir funktioniert es jetzt, der Hinweis von Knickohr im oben verlinkten Thread war hilfreich!

pwsurf, 2024-06-28:
> Bei mir funktioniert es jetzt, der Hinweis von Knickohr im oben verlinkten Thread war hilfreich!

Danke für den Hinweis. Hilft bei mir leider nicht. Muss auch mit 0.7.36 ein paar Minuten warten bis es los geht. Mit 0.8.83 ging auch nach 3h noch nichts. Netzteil tauschen auch ohne Erfolg. Kann aber auch mit 0.7.36 gut weiterleben.

pwsurf, 2024-09-13:
Nur zur Info: Auch mit 0.8.140 das gleiche Spiel: "Wechselrichter HM1500 ist gerade nicht verfügbar", egal ob -all- oder normale deutsche Version aufgespielt wird. Bei -all- Version zusätzlich alle paar Minuten ein reboot. --> Bin wieder etwas ratlos zurück auf 0.7.36.

Gubi2023, 2024-09-14:
siehe https://github.com/lumapu/ahoy/issues/1691#issuecomment-2195173971  
Vllcht doch auf ESP32 umsteigen?