lumapu/ahoy #1494: Setting Inverter HM-1500 Interval can not change in newer Versions [Bug]

Andy811963, 2024-03-10:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

board

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

08.91

### Github Hash

?

### Build & Flash Method

ESP Tools (flash)

### Setup

Inverter Intervall can not chance more then 5 Sek.

In older Version it work. 

### Error description

Inverter Intervall HM-1500 
can not chance more then 5 Sek.
In older Version it work. 
[image]

lumapu, 2024-03-10:
Can you please be more detailed regarding the issue with interval?
If I understand correctly you can set every value up to 5s but no higher values?

You also can answer in German if you like.

Andy811963, 2024-03-10:
Ja genau so ist es.
Alles was ich einstelle funktioniert nicht.
Das Abfrage-Intervall ist immer 5. Egal was ich einstelle.
Angezeigt wird aber immer 30. Das habe ich vor Wochen mit einer alten Version mal so eingestellt.
Ich möchte gerne das Intervall auf 60 Sekunden setzen.
Wenn das gesperrt ist wozu ist dann das Eingabefeld siehe Bild oben.

knickohr, 2024-03-10:
Na dann schau mal in den Inverter-Settings

[image]

lumapu, 2024-03-10:
verstehe nicht warum es gesperrt sein sollte, komisch was hier wieder los ist ...

knickohr, 2024-03-10:
Ich glaube hier wird was durcheinander geworfen. Einmal das Abfrageintervall der Inverter und das andere Mal das Anzeigeintervall im Live-View ?

lumapu, 2024-03-10:
per default steht aber 15s drin, woher kommen dann die 5s? Kommt auch sicher "successfully saved"? Habe gerade bei einem ESP8266 das Interval erfolgreich von 15s auf 34s geändert

Andy811963, 2024-03-11:
Hallo zusammen,
habe den Fehler gefunden.
Ich benutze zum Anzeigen und ändern den Brave-Browser. Scheinbar überträgt der nicht alle HTML Daten.
Habe jetzt mal mit dem Edge das verstellt damit geht es.
FireFox sollte wohl auch gehen.
Der Brave Browser blockiert ja einiges.
Vielen Dank an alle die sich dem Problem oben angenommen haben.
Gruß Andy B.

closed 2024-03-11 by Andy811963