lumapu/ahoy #1495: [Bug] 0.8.92-all Display missing

DB1BMN-collection, 2024-03-10:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.92

### Github Hash

96eb787

### Build & Flash Method

was already installed

### Setup

Upgraded from 0.8.91 via bin-Upload.

### Error description

When I understand https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md#0892---2024-03-10 correctly the "-all.bin" should have support for Display as usual.

Unfortunately in all Versions "-all", "normal" and "minimal" the display don't work and there is no Menu in the Settings to switch it on (ore I can't find it)

BTW: What is the "minimal" good for, when there is neither Display, nor MQTT support. How to get the data out?

lumapu, 2024-03-10:
8266.bin
* only MqTT
* **this version is the prefered 8266 version, most stability**

8266-minimal.bin
* nothing except basic functions
* **use this version for debugging purposes**

8266-all.bin
* MqTT
* Display
* History
* **everything included without looking to stability** 

8266-prometheus.bin
* MqTT
* Display
* History
* Prometheus
* **everything + Prometheus**

### Important for testing different versions
If you want to test all version in a row make sure to reload the HTML pages directly from ESP and not from you browser cache. You can do that by pressing CTRL + F5 or clean your browser cache.

DB1BMN-collection, 2024-03-10:
> If you want to test all version in a row make sure to reload the HTML pages directly from ESP and not from you browser cache. You can do that by pressing CTRL + F5 or clean your browser cache.

OMG! Thanks.
Installed now again the normal *.bin with MQTT but no display and will observe Uptime.

closed 2024-03-10 by lumapu