lumapu/ahoy #1496: Keine grafische Darstellung im Display

khmaiterth, 2024-03-13:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.92

### Github Hash

?????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT aktiviert, IP Adressen gesetzt

### Debug Serial Log output

```shell
0.8.92 - 96eb787
---------------
18:44:47.354 -----
18:44:47.757 -----
18:44:47.757 I: com loop duration: 756ms
18:44:47.758 -----
18:45:02.354 -----
18:45:02.757 -----
18:45:02.757 I: com loop duration: 756ms
18:45:02.758 -----
```

### Error description

Im Reiter Verlauf sehe ich die Grafik von der Produktion der WR (HM600 (2x175 Wp)+HM1500(1x 255 Wp)) 
Auf dem angeschlossenen Display (SH1106 128x64 1,3") sehe ich keinen Verlauf 
bei folgender Einstellung:  Graph Anzeigeverhältnis 50 % + Zeile 1-2. 
Display schaltet um auf Grafik, aber die Grafik wird nicht angezeigt.

khmaiterth, 2024-03-13:
> Von "Dirk" ***@***.***>
> An "lumapu/ahoy" ***@***.***>
> Cc "khmaiterth" ***@***.***>; "Author"
> Datum 13.03.2024 19:12:40
> Betreff Re: [lumapu/ahoy] Keine grafische Darstellung im Display (Issue
> #1496)

Danke. Habe ich gerade gemacht.
Werde morgen das Resultat sehen.
Ich dachte ich hätte es schon gemacht.
Nochmal Danke an alle.

Karl-Heinrich

khmaiterth, 2024-03-14:
> Von "Karl-Heinrich Maiterth" ***@***.***>
> An "lumapu/ahoy"
> "Lukas Pusch" ***@***.***>
> Datum 13.03.2024 20:00:40
> Betreff Re[2]: [lumapu/ahoy] Keine grafische Darstellung im Display
> (Issue #1496)

Es klappt. Danke

Habe mir auch die Grafik aus  0.8.93 angesehen tolle Arbeit.
0.8.93
kh

lumapu, 2024-03-14:
zur Dokumentation habe ich die Problemlösung in unsere Doku aufgenommen:
https://docs.ahoydtu.de/de/latest/faq.html#fehlersuche

closed 2024-03-14 by lumapu