lumapu/ahoy #1497: keine MQTT Verbindung 0.8.64 und neuer

Krausbirn, 2024-03-13:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.64

### Github Hash

7f386f0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT Broker added

### Debug Serial Log output

```shell
Serial Monitor output: 240122_ahoy_0.8.64_7f386f0_esp32-wroom32-de.bin

19:50:49.721 -> --------------------------------
19:50:49.721 -> Welcome to AHOY!
19:50:49.721 -> 
19:50:49.721 -> point your browser to http://192.168.0.11 (Station)
19:50:49.721 -> to configure your device
19:50:49.721 -> --------------------------------
19:50:49.753 -> 
19:50:49.753 -> Wifi event: 11
19:50:49.753 -> Wifi event: 10
19:50:49.753 -> [WiFi] AP disabled
19:50:49.753 -> Wifi event: 11
19:50:49.881 -> [WiFi] mDNS established: AHOY-DTU.local
19:50:51.713 -> [ 14137][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for pool.ntp.org
19:50:55.634 -> 1710355855
19:50:56.728 -> [ 19137][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for pool.ntp.org
19:51:01.720 -> [ 24137][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for pool.ntp.org
19:51:03.715 -> I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
19:51:03.715 -> I: last tx setup: 0ms
19:51:03.747 -> I: (#0) TX 27 CH23 | 15 81 81 10 87 89 64 17 65 80 0B 00 65 F1 F5 98 00 00 00 00 00 00 00 00 AC F4 37 
19:51:04.101 -> I: (#0) request timeout: 350ms, ARC 15, PLOS 15
19:51:04.101 -> -----
19:51:04.101 -> I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
19:51:04.101 -> I: last tx setup: 37ms
19:51:04.101 -> I: (#1) TX 27 CH23 | 15 81 01 36 79 89 64 17 65 80 0B 00 65 F1 F5 98 00 00 00 00 00 00 00 00 AC F4 6F

Serial Monitor output: 240122_ahoy_0.8.63_77b1f96_esp32-wroom32-de.bin

19:45:05.828 -> to configure your device
19:45:05.828 -> --------------------------------
19:45:05.828 -> 
19:45:05.828 -> Wifi event: 11
19:45:05.860 -> Wifi event: 10
19:45:05.860 -> [WiFi] AP disabled
19:45:05.860 -> Wifi event: 11
19:45:05.957 -> [WiFi] mDNS established: AHOY-DTU.local
19:45:07.823 -> [ 13140][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for pool.ntp.org
19:45:07.951 -> I: MQTT connected
19:45:18.178 -> 1710355518
19:45:19.818 -> I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
19:45:19.818 -> I: last tx setup: 0ms
19:45:19.818 -> I: (#0) TX 27 CH23 | 15 81 81 10 87 89 64 17 65 80 0B 00 65 F1 F4 40 00 00 00 00 00 00 00 00 6C 5F 85 
19:45:20.171 -> I: (#0) request timeout: 350ms
19:45:20.171 -> -----
19:45:20.171 -> I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
19:45:20.203 -> I: last tx setup: 37ms
19:45:20.203 -> I: (#1) TX 27 CH23 | 15 81 01 36 79 89 64 17 65 80 0B 00 65 F1 F4 40 00 00 00 00 00 00 00 00 6C 5F DD 
19:45:20.557 -> I: (#1) request timeout: 350ms
19:45:20.557 -> -----
19:45:20.557 -> I: com loop duration: 731ms
19:45:20.557 -> -----
```

### Error description

Bis Version 0.8.63 keine Probleme mit MQTT Verbindung
ab Version 0.8.64 kann keine MQTT Verbindung hergestellt werden (System / Mqtt / verbunden / nein)

DTU wird in einem Netzwerk ohne Internetverbindung, ohne NTP Server betrieben
Einstellung: Kommunikation starten ohne gültige Zeit (sinnvoll im AP Modus)

Vermutung: Könnte es sein, dass durch den nicht möglichen NTP syc keine MQTT Verbindung hergestellt werden kann ?
Habe bereits einige FW Versionen ohne Erfolg durchprobiert 0.8.7x,  0.8.83, 08.92 
Vorschlag: Möglichkeit den NTP ServerSyc per check box zu deaktivieren integrieren

tictrick, 2024-03-14:
Ich habe mal auf die schnelle die beiden Versionen 0.8.63 und 0.8.64 miteinander verglichen ... da wurde am MQTT nichts verändert.

Hier ein Diff:
https://github.com/lumapu/ahoy/compare/77b1f96c03332f245d146a4d90c9e5264c054b07...7f386f07df602e71d0d66ed7e0ddc51c4630aa05

Gibts zu dem ESP32 MQTT Broker mehr Infos? So dass man das vielleicht nachstellen kann?

Hier noch eine Einladung auf den Discord Supportserver
https://discord.gg/WzhxEY62mB

Krausbirn, 2024-03-14:
ich hab auch den change log rauf und runter gelesen und gesehen, dass es am MQTT nix geändert wurde.
hab aber auch gesehen dass es schon einige anpassungen in bezug auf ntp gegeben hat, ob die da jetzt mit reinspielen weiss ich nicht. 

am esp32 läuft der sMQTTBroker von terrorsl ver 0.1.7

in beiden logs wird die zeit vom browser übernommen:
19:50:55.634 -> 1710355855
19:45:18.178 -> 1710355518

in der 64er version wird danach aber weiterhin versucht den ntp server zu erreichen:
DNS Failed for pool.ntp.org
in der 63er nicht.

@rori1977: ich hab schon mehrfach überlegt wie ich ohne inet zugang einen ntp server emulieren oder zeitabgleich realisieren könnte, der eingesetzte router kanns leider nicht. deshalb war ich ja hellauf begeistert als die option: Kommunikation starten ohne gültige Zeit (sinnvoll im AP Modus) auftauchte.

lumapu, 2024-03-14:
ich schaue da die Tage mal rein, da wir jetzt schon mehrere Issues zum NTP haben .. das ist komisch. Danke für die genauen Versionsangaben und @tictrick danke für den Diff.
Simulieren kann man das sehr einfach, daher denke kann man das Problem auch recht gut in den Griff bekommen. 😊

Krausbirn, 2024-03-17:
👍

lumapu, 2024-03-17:
heißt das, es funktioniert bei dir?

Krausbirn, 2024-03-17:
ja danke für die schnelle behebung, die 08.95 macht sich bis jetzt echt gut

closed 2024-03-17 by Krausbirn